esphome/feature-requests #2456: Support for A2DP Bluetooth profile (mainly audio sender)

spudje, 2023-11-02:
**Describe the problem you have/What new integration you would like**

No support for A2DP in ESPHome, I want it mainly as audio sender.

**Please describe your use case for this integration and alternatives you've tried:**

With the Bluetooth (proxy) support in ESPHome this ideally opens up the room for cheap multi-room audio streaming when ESP32s  (or RP2040, I don't care which) can be connected to Bluetooth speakers (or headphones) via the A2DP profile.

I actually think the bluetooth proxy does not really add something. It's basically the ESP32 acting as mediaplayer for Hass and connecting over Bluetooth (A2DP) with speakers/headphones as "audio sink".

When I dream out loud, it would be extremely awesome to roam with a bluetooth headphone from room to room and have handovers between the ESPs... But I guess that's a few bridges too far :)

**Additional context**

I could find one github with an A2DP implementation for ESP32, maybe that can be used as a starting point? It both supports audio receiving & audio sending.
https://github.com/pschatzmann/ESP32-A2DP

orgebar, 2024-01-05:
I don‘t quite understand why bt topics in general isnt more in focus, Especially bt speakers. Its still a big disappointment that there is no (simple) solution.

Andrew49, 2024-07-20:
+1 for me

bpavlov, 2024-09-10:
+1. My use case is to create an esp32 satellite connected to a Bluetooth speaker.

tbrasser, 2024-10-20:
+1 as receiver it would be awesome for Raspi Muse Luxe

mikulik86, 2026-10-09:
Also great a a fallback if HA goes down, one could stream directly to a ESPhome speaker over BT.

jpfaff, 2026-10-09:
https://github.com/PxPert/a2dp_sink

schmidttrick, 2026-10-29:
+1 for me. Would be great to use my Bluetooth speakers as MA player in the garden.