esphome/feature-requests #2457: Dashboard Editor right-click paste button missing

NonaSuomy, 2023-11-02:
**Describe the problem you have/What new integration you would like**

Add a clickable button to paste in the right-click menu of the editor.

**Please describe your use case for this integration and alternatives you've tried:**

You can see in the image Paste is not available.
[image]

**Additional context**

NonaSuomy, 2023-11-04:
Tested Chromium, Chrome, Firefox, Opera, Safari / Linux, Windows, MacOS all the same.

tiimsvk, 2023-12-31:
I would have more reservations when editing the text.
On mobile, it is not possible to mark more text at once, select, copy and paste.

NonaSuomy, 2024-05-31:
I was nudged in the right direction by balloob on discord he said "you can't access the clipboard when not browsing over HTTPS, it's a browser limitation"

I then looked at the site settings and sure enough

[image]
Seems you can write (cut, copy) to the clipboard without HTTPS, just not read (paste) from the clipboard without HTTPS 😅 

So I added the site to the `Insecure origins treated as secure` list in `chrome://flags/`
[image]

Boom it was there now after the browser restarted:

[image]

[image]

Thank you!

closed 2024-05-31 by NonaSuomy

tiimsvk, 2024-06-01:
How to enable in firefox?

NonaSuomy, 2024-06-02:
There currently isn't a way for firefox however you could probably use some scripting skills to inject a right click option that presses the ctrl+v keyboard shortcut for you.