esphome/feature-requests #2458: INA226 configure the "Operating mode" to save power

gargamelonly, 2023-11-03:
**Describe the problem you have/What new integration you would like**

The device has a power-down or triggered mode setting in the Configuration Register (00h) (Read/Write)
MODE: Operating Mode
Bits 0-2 Selects continuous, triggered, or power-down mode of operation. 
These bits default to continuous shunt and bus measurement mode.
Currently there is no option in ESPHome to change this mode.
I would like to see an option to write to the MODE register.

**Please describe your use case for this integration and alternatives you've tried:**

My setup is battery operated with the INA226 used to monitor battery voltage and current consumption to calculate remaining battery charge and circuit consumption. While the INA226 is not being read it has a current draw of 220uA while it supports a power-down mode when the power consumption of the device drops to about 2uA considerably saving more power.
When requesting to do a measurement the device mode needs to change to "triggered" then back to "power-down".
There is no alternate way to reduce the power consumption of the device.

**Additional context**

latonita, 2024-03-17:
@gargamelonly is it still needed?

gargamelonly, 2024-03-17:
Yes please.
Appreciate it.

latonita, 2024-03-18:
@gargamelonly try https://github.com/esphome/esphome/pull/6394
please try with usual config (continuous mode)
and with "operating_mode: low_power"

gargamelonly, 2024-04-27:
I know my next question is not in the scope of this issue...

I get an error in ESPHOME saying that the "operating_mode" is an error. 
Running ESPHOME 2024.4.1 on docker.

latonita, 2024-04-27:
Please attach your config file

gargamelonly, 2024-04-27:
```
esphome:
  name: sensor-test
  on_boot:
    then:
      - script.execute: consider_deep_sleep

esp8266:
  board: nodemcuv2
...
i2c:
  sda: 12
  scl: 14
  scan: False

status_led:
  pin: GPIO2

sensor:
  - platform: bmp280
    temperature:
      name: "BoxTemperature"
    pressure:
      name: "BoxPressure"
    address: 0x76
    update_interval: 60s
  - platform: sht3xd
    temperature:
      name: "AirTemperature"
    humidity:
      name: "AirHumidity"
    address: 0x44
    update_interval: 60s
  - platform: ina226
    address: 0x40
    #operating_mode: low_power  # continuous (default) or low_power
    shunt_resistance: 0.1 ohm
    current:
      name: "Current"
    power:
      name: "Power"
    bus_voltage:
      name: "Bus_Voltage"
    shunt_voltage:
      name: "Shunt_Voltage"
    max_current: 3.2A
    update_interval: 60s
...
```

latonita, 2024-04-27:
you shall use external component, since this is not yet merged into main branch.
make sure you have this section in yaml:
```
external_components:
  - source: github://pr#6394
    refresh: 10s
    components: [ina226]

```