esphome/feature-requests #2465: DSMR Compatibility - Water Readout

Vato9394, 2023-11-07:
**Describe the problem you have/What new integration you would like**
Include the DSMR water readout into the official release in esphome.
Consolidating:
https://github.com/esphome/feature-requests/issues/1873
https://github.com/esphome/feature-requests/issues/1731

**Please describe your use case for this integration and alternatives you've tried:**
We actually have the documentation to implement this. it just have to be approved into the official integration.
https://www.cdem.be/13_technical/#what-information-is-provided-with-the-p1-port

**Additional context**

OBIS reference | Value | Example
0-n:24.2.1 | Last 5-minute Meter reading in 0,001m3 and capture time | 0-1:24.2.1(12785.123*m3)
suggest the following name for water belgium:
water_delivered_be: Water Consumed Belgium.

To prove this is working, someone implemented this under the luxembourg value: https://github.com/peno64/slimmemeter/wiki/Water-consumed

this should be nice if this is implemented under its own sensor so that we don't have to override other sensors.

And afterwards, update the DSMR ESPHome documentation here: https://esphome.io/components/sensor/dsmr.html

Coollision, 2023-11-18:
@Vato9394 looks like there is already something there:
https://github.com/esphome/esphome/pull/4237

will test it out tomorrow when I setup mine dsmr

Vato9394, 2023-11-20:
@Coollision Did it work?

Coollision, 2023-11-20:
Yes it did! You can add it without any issues or special things, we might want to update the docs, but it wirks like a charm (in Belgium)

Vato9394, 2023-11-20:
So no defining "uart_id" and "water_mbus_id"?

`dsmr:
  uart_id: p1_uart
  water_mbus_id: 2`

So the configuration for the "Slimmelezer" in Belgium will be like this:

`---
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
     
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.0"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"

wifi:
  networks:
    - ssid: "networkname"
      password: "networkpassword"
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
  baud_rate: 0
 
# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
    then:
      - logger.log:
          format: Setting private key %s. Set to empty string to disable
          args: [private_key.c_str()]
      - globals.set:
          id: has_key
          value: !lambda "return private_key.length() == 32;"
      - lambda: |-
          if (private_key.length() == 32)
            private_key.copy(id(stored_decryption_key), 32);
          id(dsmr_instance).set_decryption_key(private_key);
 
ota:
 
web_server:
  port: 80
 
uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1500
 
globals:
  - id: has_key
    type: bool
    restore_value: yes
    initial_value: "false"
  - id: stored_decryption_key
    type: char[32]
    restore_value: yes
 
dsmr:
  id: dsmr_instance
  # For Luxembourg users set here your decryption key. 
  # Enable this when using decryption for Luxembourg;
  # key looks like '00112233445566778899AABBCCDDEEFF'
  #decryption_key: !secret decryption_key 
 
sensor:
  - platform: dsmr
    water_delivered:
      name: Water delivered
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
      state_class: total_increasing
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
      state_class: total_increasing
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
      state_class: total_increasing
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
      state_class: total_increasing
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered_be:
      name: "Gas Consumed Belgium"
      state_class: total_increasing
    active_energy_import_current_average_demand:
      name: "Current Average Quarterly Peak Demand"
    active_energy_import_maximum_demand_running_month:
      name: "Current Month Average Quarterly Peak Demand"
    active_energy_import_maximum_demand_last_13_months:
      name: "Last 13 months Average Quarterly Peak Demand"
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s
 
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp:  #true`

Coollision, 2023-11-20:
that looks an awful lot like what i did 😄

Coollision, 2023-11-20:
so we can close this feature request 🥳

Vato9394, 2023-11-20:
Issue resolved. Documentation still needs t be updated.

Working Code:

substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
     
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.0"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"

wifi:
  networks:
    - ssid: "networkname"
      password: "networkpassword"
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
  baud_rate: 0
 
# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
    then:
      - logger.log:
          format: Setting private key %s. Set to empty string to disable
          args: [private_key.c_str()]
      - globals.set:
          id: has_key
          value: !lambda "return private_key.length() == 32;"
      - lambda: |-
          if (private_key.length() == 32)
            private_key.copy(id(stored_decryption_key), 32);
          id(dsmr_instance).set_decryption_key(private_key);
 
ota:
 
web_server:
  port: 80
 
uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1500
 
globals:
  - id: has_key
    type: bool
    restore_value: yes
    initial_value: "false"
  - id: stored_decryption_key
    type: char[32]
    restore_value: yes
 
dsmr:
  id: dsmr_instance
  # For Luxembourg users set here your decryption key. 
  # Enable this when using decryption for Luxembourg;
  # key looks like '00112233445566778899AABBCCDDEEFF'
  #decryption_key: !secret decryption_key 
 
sensor:
  - platform: dsmr
    water_delivered:
      name: "water delivered"
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
      state_class: total_increasing
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
      state_class: total_increasing
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
      state_class: total_increasing
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
      state_class: total_increasing
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered_be:
      name: "Gas Consumed Belgium"
      state_class: total_increasing
    active_energy_import_current_average_demand:
      name: "Current Average Quarterly Peak Demand"
    active_energy_import_maximum_demand_running_month:
      name: "Current Month Average Quarterly Peak Demand"
    active_energy_import_maximum_demand_last_13_months:
      name: "Last 13 months Average Quarterly Peak Demand"
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s
 
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp:  #true`

``` #

closed 2023-11-20 by Vato9394