esphome/feature-requests #2466: Update NeoPixelBus to latest version and add support for SM16825 LEDs

charlievogt, 2023-11-07:
**Describe the problem you have/What new integration you would like**

I'd like to use LEDs with the SM16825 chip.  NeoPixelBus has a new release (2.7.7) that supports this chip.  https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.7

**Please describe your use case for this integration and alternatives you've tried:**

When setting the variant to "SM16825", the YAML fails to validate.

**Additional context**

This chip runs the new Govee Outdoor Permanent Lights Pro.  Adding support for the new release of NeoPixelBus should allow for support of the new lights from Govee.

tungmeister, 2024-06-08:
I'd like to see this as well for the same use case.

edwardtfn, 2025-06-01:
> When setting the variant to "SM16825", the YAML fails to validate.

Have you ever tried PR#7980? It should support `SM168XX`.
I've worked with this a while ago but never got a change to test with many different models.

- esphome/esphome#7980

syssi, 2026-10-28:
Somebody likes to play around with this custom component?

https://github.com/syssi/esphome-yeelight-ceiling-light/pull/168