esphome/feature-requests #2467: Door sensor deep sleep for bk7231n board

yaminahacs, 2023-11-08:
Door sensor deep sleep for bk7231n board

**Describe the problem you have/What new integration you would like**
  Add door sensor deep sleep for bk7231n board

**Please describe your use case for this integration and alternatives you've tried:**
 ESPHOME deep sleep not working for bk72xx platform

**Additional context**

imorlxs, 2023-12-12:
I didn't know how to enable deep sleep too

Mostalk, 2023-12-24:
My temporary solution

```
script:
    - id: scr
      then:
        - wifi.enable:
        - delay: 30s
        - wifi.disable:
      mode: restart
      
binary_sensor:
  - platform: gpio
    device_class: door
    name: Door
    pin: P22
    on_state:
      then:
        - script.execute: scr
```

yaminahacs, 2023-12-27:
ThanksMy iPad will find you

AzSaSiN, 2024-01-21:
Having functional deep sleep will help in a long run.

in mean time, I've been trying to get up to speed with the project

grego1981, 2024-01-24:
I've implemented your solution as I'm trying to find the deep sleep code for the exact same type of sensor... Below my running ESPHome config as it works now, quite well. Please comments are needed as I'm not really sure it is efficent enough!

`esphome:
  name: basement-door-sensor
  friendly_name: basement_door_sensor
  on_boot:
    priority: 10
    then:
      - switch.turn_on: batt_dv
      - script.execute: scr
bk72xx:
  board: cb3s
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Basement-Door-Sensor"
    password: "1234567890"

web_server:
  port: 80
  version: 2
  include_internal: true
  ota: true

captive_portal:

script:
    - id: scr
      then:
        - wifi.enable:
        - delay: 30s
        - wifi.disable:
      mode: restart    
binary_sensor:
  - platform: gpio
    device_class: door
    name: Door
    pin: P8
    on_state:
      then:
        - output.turn_on: light_output
        - delay: 500ms
        - output.turn_off: light_output 
        - script.execute: scr    
switch:
  - platform: gpio
    id: batt_dv
    pin: P14
sensor:
  - platform: adc
    pin: ADC3
    id: "VCC"
    name: "Battery Level"
    filters:
      - multiply: 2.4
  - platform: template
    name: "Battery Level (%)"
    unit_of_measurement: '%'
    update_interval: 240s
    lambda: |-
      return ((id(VCC).state /3.0) * 100.00);
light:
  - platform: binary
    name: "Activity LED"
    output: light_output

output:
  - id: light_output
    platform: gpio
    pin: P26`

0x3333, 2024-02-22:
There is some work going on the LibreTiny to support it in ESPHome, but I believe it will take some time until we have Deep Sleep for these chips.

jon-daemon, 2024-03-09:
Does the above script work? What is the power consumption? I have two water sensors operating with 2xAAA batteries and I'd really like to replace the tuya firmware with something else.

afarago, 2024-06-01:
I have found a nice deep_sleep component for bk72xx / cb3s processor.

Additionally added code for delayed deep sleep and forced manual mode as well.
I would love to see the deep_sleep code from @Xmister merged if someone has the right knowledge.

It took 2-3 days for me to fine tune this, pasting it here to share the info and love of bkn :D hope it helps someone.

```yaml
# PINOUT: CB3S Module Datasheet - https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076
# UART flash with: https://github.com/libretiny-eu/ltchiptool
# UART flashing: GND(pin9),VCC3V(pin8),TX(pin16),RX(pin15) / connect CEN(pin3) to GND(pin9)
# CB3S / tuya chip - https://community.home-assistant.io/t/tuya-door-sensor-configuration/673484
# MORE INFO on the board: https://www.elektroda.com/rtvforum/topic3960149.html
# MORE INFO on the board: https://www.elektroda.com/rtvforum/topic3914412-30.html#gallery-2

### TODO & FURTHER IDEAS
# - additional pins?
# - explore manual "stored" server side update switch
# - switch off w restore LED on activate
# - support Bluetooth LE.

bk72xx:
  # board: generic-bk7231n-qfn32-tuya
  board: cb3s # FAMILY_BK7231N
  framework:
    version: dev

esphome:
  name: z-test-door
  friendly_name: z-test-door
  on_boot:
    # https://esphome.io/components/esphome.html#on-boot
    - priority: 600.0 # 600.0: This is where most sensors are set up.
      then:
        - component.update: VCC
        - switch.turn_on: batt_dv # start measuring voltage
    - priority: 100.0
      then:
        - output.turn_on: activity_led_output
    - priority: -100 # 200.0: Network connections like MQTT/native API are set up at this priority.
      then:
        - component.update: VCC
        - component.update: homeassistant_time
        - component.update: update_time
        - script.execute: scr

logger:
  # level: info

api:

globals:
  - id: first_connect_ms_int
    type: uint32_t
    initial_value: "0"

ota:
  on_begin:
    - output.turn_on: activity_led_output
  # on_progress:
  #   - output.toggle: light_output

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # fast_connect: True
  # manual_ip:
  #   gateway: 192.168.0.1
  #   static_ip: 192.168.0.212
  #   subnet: 255.255.255.0
[... 177 lines cut ...]

external_components:
  - source: github://Xmister/libretuya-esphome@deep-sleep
    components: [deep_sleep]
  # - source: github://libretiny-eu/libretiny-esphome@dev
  #   components: [ deep_sleep ]
  # - source:
  #     type: git
  #     url: https://github.com/bkaufx/esphome
  #   components: [ web_server_base ]

deep_sleep:
  id: deep_sleep_control
  sleep_duration: 24h
  run_duration: 10s
  wakeup_pins:
    - pin:
        number: P7 # pin-14 bottom button
        allow_other_uses: true
      wakeup_pin_mode: IGNORE
    - pin:
        number: P8 # pin-13 hall sensor
        allow_other_uses: true
      wakeup_pin_mode: INVERT_WAKEUP
```

Xmister, 2024-06-01:
@yaminahacs Welcome to the club :smiling_face_with_tear: This was my initial PR to prepare the deep sleep for BKN: https://github.com/esphome/esphome/pull/5739

afarago, 2024-06-02:
> @afarago  Welcome to the club :smiling_face_with_tear: This was my initial PR to prepare the deep sleep for BKN: https://github.com/esphome/esphome/pull/5739

Thanks for your awesome work! Is there anything missing for the PR to be merged?
I love it and it seems to work flawlessly ony device after testing it for a few days.

AlenHodnik, 2024-06-02:
How fast does it use batteries with this code?
All seem to work good, showing battery, deep_sleep checking, manual override, ... but it doesn't change state of the binary_sensor. Wrong pin maybe?

afarago, 2024-06-02:
I have not measured the deep_sleep current, yet it should be the same as the original tuya firmware.
Power consumption seems to be almost zero after a few days.

PIN depends on your board, mine is as above in the links. You could check the integrated circuit by eyes or by a manual multimeter.

AlenHodnik, 2024-06-03:
Found right pin. But seems smth wrong in my case. Doesnt change to open at all, battery -30% over night.

[image]

0x3333, 2024-06-06:
> ...battery -30% over night.

It seems that deep sleep is not working.

starbrights, 2024-06-21:
Maybe my measurement is to bad, but I don't see much of an effect for disabling wifi for some time 1..2min. 
Any experience how much energy is needed when Wifi is  enabled again? I mean is there a certain period to be off required that it is worth (in case reconnect takes more energy than keep connection). 
What else can be done to reduce consumption?
I am already using this:
(About the measurements results in comment: Not sure this is serious, I do see mixed results).
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.3.250
    gateway: 192.168.3.1
    subnet: 255.255.255.0
  ap:  
    password: !secret wifi_ap_password
  fast_connect: 'true'
  power_save_mode: 'high'    # Optionen: none 1.5W, light 1.43W, high 1.28W
  output_power: 10dB         # reduce power (default is 20db)  10/15 ->  0,92W
  reboot_timeout: '0s'
```

AlenHodnik, 2024-06-23:
> > ...battery -30% over night.
> 
> It seems that deep sleep is not working.

Deep sleep is working, measured current consumption. The problem is that device keeps getting out of deep sleep for no reason (see led light). It should only wake every 24h and on pin change, but it isnt. Any ideas?

V.

afarago, 2024-06-23:
We might spot the error seeing the complete yaml file. Would you add it to this discussion?

Changing Wifi power save and transmit power does not make a sense imo, as ideally the device is like 10s awake every now and then, so if deep_sleep is operational you practically have zero wake time.

AlenHodnik, 2024-06-23:
Don't think I changed much really. Here's yaml:

bk72xx:
  board: cb3se # FAMILY_BK7231N
  framework:
    version: dev

esphome:
  name: kopalnica-vrata-contact-sensor
  friendly_name: kopalnica-vrata-contact-sensor
  on_boot:
    - priority: 600.0 # 600.0: This is where most sensors are set up.
      then:
 #       - lambda: |-
 #           id(my_ota).set_auth_password("");
        - component.update: VCC
        - switch.turn_on: batt_dv # start measuring voltage
    - priority: 100.0
      then:
        - output.turn_on: activity_led_output
    - priority: -100 # 200.0: Network connections like MQTT/native API are set up at this priority.
      then:
        - component.update: VCC
#        - component.update: homeassistant_time
#        - component.update: update_time
        - script.execute: scr

logger:
  # level: info

api:

globals:
  - id: first_connect_ms_int
    type: uint32_t
    initial_value: "0"

ota:
  on_begin:
    - output.turn_on: activity_led_output
#  password: "xxxxxxxx"
#  id: my_ota
  # on_progress:
  #   - output.toggle: light_output

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  on_connect:
    - lambda: |-
        if (id(first_connect_ms_int) == 0) id(first_connect_ms_int) = esphome::millis();
    - component.update: first_connect_ms

web_server:
  version: 3

script:
  - id: scr
    then:
      - deep_sleep.prevent
      # - wifi.enable:
      - delay: 10s
      # - wifi.disable:
      - script.execute: check_deep_sleep
    mode: restart
  - id: check_deep_sleep
    then:
      - deep_sleep.prevent
      - if:
          condition:
            - switch.is_off: manual_mode
          then:
            - output.turn_off: activity_led_output
            - deep_sleep.allow
    mode: restart

binary_sensor:
  - platform: gpio
    device_class: door
    name: Door
    pin:
      number: P16 # pin-13 hall sensor, GPIOP_8, which corresponds to P8 of the IC, PWM 2
      allow_other_uses: true
      inverted: true
    on_state:
      - if:
          condition:
            - switch.is_off: manual_mode
          then:
            - output.turn_on: activity_led_output
            - delay: 1000ms
            - output.turn_off: activity_led_output
          else:
            - light.turn_on:
                id: activity_led
                effect: pulse
      - script.execute: scr
    icon: mdi:door

  - platform: gpio
    id: bottom_button
    pin:
      number: P7 # pin-14 bottom button, GPIOP_7, which corresponds to P7 of the IC, PWM 1
      inverted: true
      allow_other_uses: true
    name: "Bottom button"
    icon: mdi:button-pointer
    on_state:
      - script.execute: check_deep_sleep
    on_multi_click:
      - timing:
          - on for at least 1s
        then:
          - switch.toggle: manual_mode

switch:
  - platform: gpio
    id: batt_dv
    pin: P14

  - platform: template
    # manual mode for OTA
    optimistic: True
    name: Manual mode
    id: manual_mode
    on_turn_off:
      - script.execute: check_deep_sleep
    on_turn_on:
      - script.execute: check_deep_sleep
      - light.turn_on:
          id: activity_led
          effect: pulse
    entity_category: config
    icon: mdi:sleep-off

light:
  # - platform: binary
  #   name: "Activity LED"
  #   output: light_output
  #   id: activity_led_output
  #   icon: mdi:lamp
  #   effects:
  #     - strobe:
  - platform: monochromatic
    name: "Activity LED"
    output: activity_led_output
    id: activity_led
    icon: mdi:lamp
    effects:
      - strobe:
      - pulse:
          transition_length:
            on_length: 1s
            off_length: 2s
          update_interval: 3s

output:
  # - id: light_output
  #   platform: gpio
  #   pin:
  #     number: P26 # P26=pin-5
  #     inverted: true
  - id: activity_led_output
    platform: libretiny_pwm
    frequency: 1000 Hz
    pin:
      number: P26 # P26=pin-5
      inverted: true

#time:
#  - platform: homeassistant
#    id: homeassistant_time
#    on_time_sync:
#      then:
#        - component.update: update_time

sensor:
  - platform: adc
    pin: ADC3 # ADC3 = pin-2 (internally connnected), gives valid only if "batt_dv" is enabled (see more info there)
    id: "VCC"
    name: "Battery Level"
    filters:
      - multiply: 2.4
      - clamp:
          min_value: 0.0
          max_value: 3.0
          ignore_out_of_range: true
    on_value:
      then:
        - component.update: battery_level
    icon: mdi:battery

  - platform: template
    name: "Battery Level (%)"
    unit_of_measurement: "%"
    update_interval: 86400s
    id: battery_level
    lambda: |-
      return ((id(VCC).state /3.0) * 100.00);
    icon: mdi:battery

  - platform: template
    # temporary solution to measure startup time
    name: first_connect_ms
    id: first_connect_ms
    lambda: |-
      return id(first_connect_ms_int);

  - platform: internal_temperature
    name: Internal Temperature
    icon: mdi:thermometer

#text_sensor:
#  - platform: template
#    id: update_time
#    name: "Current time"
#    lambda: |-
#      char str[17];
#      time_t currTime = id(homeassistant_time).now().timestamp + 7200;
#      strftime(str, sizeof(str), "%Y-%m-%d %H:%M", localtime(&currTime));
#      return  { str };
#    icon: "mdi:clock"

external_components:
  - source: github://Xmister/libretuya-esphome@deep-sleep
    components: [deep_sleep]
  # - source: github://libretiny-eu/libretiny-esphome@dev
  #   components: [ deep_sleep ]
  # - source:
  #     type: git
  #     url: https://github.com/bkaufx/esphome
  #   components: [ web_server_base ]

deep_sleep:
  id: deep_sleep_control
  sleep_duration: 24h
  run_duration: 10s
  wakeup_pins:
    - pin:
        number: P7 # pin-14 bottom button
        allow_other_uses: true
      wakeup_pin_mode: IGNORE
    - pin:
        number: P16 # pin-13 hall sensor
        allow_other_uses: true
      wakeup_pin_mode: INVERT_WAKEUP

AlenHodnik, 2024-06-23:
After some testing I have a feeling that sensors (temperature and startup time might be the issue in my case. If anyone stumbles upon this, try removing these sensors or set their update_interval. They seem to wake from deep sleep for some reason. I don't need them, so I removed them and so far it works good.

afarago, 2024-06-26:
I have perfomed two test scenarios with the deep_sleep component above.

1. wake up every 6 hours (or on hall sensor / bottom button press)
battery performance: started with two rechargeable IKEA batteries (worst choice) at 2.64V
after cc two days and 9 wake cycles => 2.62V

2. wake up every minute (endurance test)
after cc 1.5 days and 1843 wake cycles battery is depleted
(I checked with a multimeter and change the multiplier factor afterwards -> 2.64V->2.00V)

[image]
[image]

Hope this helps, certainly impressive data for me that predicts >1y runtime.
I am not sure that an I-am-alive is neccessary every 6h, could easily be every day or so.

starbrights, 2024-07-01:
> I have found a nice deep_sleep component for bk72xx / cb3s processor.
>
> Additionally added code for delayed deep sleep and forced manual mode as well. I would love to see the deep_sleep code from [...]

Can you explain what I am missing? I get: Unable to find action with the name 'deep_sleep.prevent'
And - isn't it like that if WiFi is not working it starts to reboot without  that reboot_timeout: '0s' ?

starbrights, 2024-07-07:
@afarago  @Xmister :
Can you explain how to use deep sleep properly, or what I am doing wrong.
Intention of the above is to sleep until it wakes up by pressing up/down/stop key and than perform the action.

It seems not doing anything but wake up and reboot. It is difficult to follow with no (serial log) attached.

starbrights, 2024-08-06:
No updates here?

toughvj, 2024-08-09:
it looks like there is an never ending deep_sleep loop. 

this is my yaml:
`substitutions:    device_name: pokoj_helenki_czujnik_okna  upper_devicename: Pok%C3%B3j Helenki  czujnik okna  esphome:  name: ${device_name}  friendly_name: ${upper_devicename}bk72xx:  board: cbu  logger:api:  encryption:    key: "bMQI1GY+PxUJ7BoL4MxBOiT4mw/Ie04pJ6QIdL30ktU="  reboot_timeout: 0sota:  - platform: esphome    password: "d8f6b1151718556c88576ec6e059ac07"wifi:  ssid: !secret wifi_ssid  password: !secret wifi_password  fast_connect: True  manual_ip:    static_ip: 10.0.40.219    gateway: 10.0.40.1    subnet: 255.255.255.0  ap:    ssid: ${upper_devicename}    password: ""captive_portal:web_server:  port: 80light:- platform: status_led  id: activity_led  pin:     number: P15    inverted: True     text_sensor:  - platform: wifi_info    ip_address:      name: Adres IPbutton:  platform: restart  name: Restart  id: button_restart  binary_sensor:  - platform: gpio    pin:      number: P22      allow_other_uses: True    device_class: window    name: Okno    on_state:      then:        - light.turn_on: activity_led        - delay: 250ms        - light.turn_off: activity_led  - platform: gpio    pin:      number: P14      allow_other_uses: True    id: btn_rst  - platform: gpio    id: voltage_relay    internal: True    pin:       number: P20      inverted: False      mode: INPUT_PULLUPsensor:  - platform: adc    pin: ADC3    id: VCC    filters:      - multiply: 2.4      - clamp:          min_value: 0.0          max_value: 3.0          ignore_out_of_range: true  - platform: template    id: battery_level    name: "Bateria"    unit_of_measurement: '%'    update_interval: 240s    lambda: |-      return ((id(VCC).state /3.0) * 100.00);external_components:  - source: github://Xmister/libretuya-esphome@deep-sleep    components: [deep_sleep]deep_sleep:  id: deep_sleep_control  sleep_duration: 24h  run_duration: 10s  wakeup_pins:    - pin:        number: P14        allow_other_uses: True      wakeup_pin_mode: IGNORE    - pin:        number: P22        allow_other_uses: True      wakeup_pin_mode: INVERT_WAKEUP`

and log:
[logs_pokoj-helenki-czujnik-okna_logs.txt](https://github.com/user-attachments/files/16566370/logs_pokoj-helenki-czujnik-okna_logs.txt)

szupi-ipuzs, 2024-08-13:
Hi, if anyone is interested I have a [somewhat working external component](https://github.com/szupi-ipuzs/esphome-external-lt-deepsleep) that implements deep sleep for bk7231 boards.
I do plan to create a PR out of it, but seeing how @Xmister 's PR for deep sleep is treated by the maintainers, I don't get my hopes up...

szupi-ipuzs, 2024-08-13:
@Xmister , I can now see that you were the original poster of the PR I based my external component on. So, the cycle loops again :)

toughvj, 2024-08-15:
> Hi, if anyone is interested I have a [somewhat working external component](https://github.com/szupi-ipuzs/esphome-external-lt-deepsleep) that implements deep sleep for bk7231 boards. I do plan to [...]

Sorry for my begginers question but how to use it?

szupi-ipuzs, 2024-08-15:
> Sorry for my begginers question but how to use it?

Just follow [the official guide for using external components](https://esphome.io/components/external_components.html).
You need to specify the whole set of my external components, which is [here](https://github.com/szupi-ipuzs/esphome-external-components), and then tell esphome which of them you import.

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/szupi-ipuzs/esphome-external-components
      ref: main
    components: [ deep_sleep_libretiny ]
```
And then follow [the description](https://github.com/szupi-ipuzs/esphome-external-lt-deepsleep?tab=readme-ov-file#usage).

toughvj, 2024-08-16:
I did as you wrote, but in the esphome editor I get a message like in the screenshot. I added the component locally, then everything works, the device goes into deep sleep, but now I have a problem with the unavailable state. I edited the file you indicated in the esphome container, compiled, added the device again, but after going into sleep mode, it still appears as unavailable. Am I doing something wrong?

Are you sure this part of the code is correct?

#ifdef USE_DEEP_SLEEP
namespace esphome::deep_sleep {
extern bool global_has_deep_sleep; // NOLINT(cppcoreguidelines-avoid-non-const-global-variables)
#endif
} 

Shouldn't it be:

 #ifdef USE_DEEP_SLEEP namespace esphome::deep_sleep { extern bool global_has_deep_sleep; // NOLINT(cppcoreguidelines-avoid-non-const-global-variables)
}
[image]
#endif

szupi-ipuzs, 2024-08-18:
> Are you sure this part of the code is correct?

Yes, I'm sure. This should work.
How long is the sleep time you have setup in the yaml? I had the same problem when I set it to 24h. When I changed it to 1h, HA started to show the correct state. I haven't yet investigated why.
Also, what are your esphome and Home Assistant versions?

szupi-ipuzs, 2024-08-18:
> Yes, I'm sure. This should work.

Ok, I can now see that you're right, the last curly brace is out of place. But your code snippet is also not correct (maybe due to white chars?).
The correct code is:
```yaml
#ifdef USE_DEEP_SLEEP
namespace esphome::deep_sleep {
extern bool global_has_deep_sleep;  // NOLINT(cppcoreguidelines-avoid-non-const-global-variables)
}
#endif
```
I will change it in the readme of my external component.

szupi-ipuzs, 2024-08-18:
@toughvj I've also noticed I haven't checked in all the needed changes before. I've just fixed it. Please download the external component locally again. It should work fine now.
Also, for any future problems with my external component, please open an issue in my repository, It will be easier to track the problems there.

AlenHodnik, 2024-08-21:
Having same error as @toughvj, and I have no clue how to make local external component. I have esphome in docker.
Any hints?
@szupi-ipuzs

AlenHodnik, 2024-08-21:
I must say everything works somewhat good, except it takes ~35s after open/close sensor for it to change status. Any ideas what could be wrong?

szupi-ipuzs, 2024-08-21:
I also use this with a door sensor and sadly I observe similar behavior. The device actually wakes up immediately, but it takes veery long time to connect to wifi. I know this because I told it to start blinking at boot and then turn on the light completely on HA connection.
I will fiddle with my wifi settings, maybe something will help, but I think that to really improve the connection time - some changes need to be done in libretiny itself. Eg, [this PR](https://github.com/libretiny-eu/libretiny/pull/276) might help.

dskw, 2024-12-04:
So what is the current state of deep sleep support on BK7231N ?
To this day I am still using the deepsleep component from @Xmister:
`github://Xmister/libretuya-esphome@deep-sleep`
which works well for my window/door sensors.

Yes, they do take much longer to connect to wifi compared to the stock firmware, but with the 2 aaa batteries in there it lasts me for 6-12 months, so I'm not complaining.
I would really like to see this getting integrated though, to make use of the pure esphome repository.

wrouesnel, 2025-02-05:
> I also use this with a door sensor and sadly I observe similar behavior. The device actually wakes up immediately, but it takes veery long time to connect to wifi. I know this because I told it to [...]

Is it connect to wifi or connect to Home Assistant? (i.e. are you using MQTT?)

The issue with using the API with deep sleep is it depends on Home Assistant detecting the device and connecting to it, rather then the device connecting to home assistant as soon as it can (which I'd really like to see fixed - it seems like a no brainer to offer an endpoint in HA that devices can hit to say "I'm awake").

szupi-ipuzs, 2025-02-11:
> Is it connect to wifi or connect to Home Assistant? (i.e. are you using MQTT?)
>
> The issue with using the API with deep sleep is it depends on Home Assistant detecting the device and connecting to [...]

@wrouesnel, you might be right, I assumed that it's esphome that reports to HA, and then HA connects immediately.
From what you say it seems HA does a polling, which means the sleeping devices would be better off reporting their state via MQTT...

szupi-ipuzs, 2025-02-11:
@jon-daemon this is because my repo uses git submodules and esphome currently cannot handle this, at least until [my PR](https://github.com/esphome/esphome/pull/7319) is merged. See the [this](https://github.com/szupi-ipuzs/esphome-external-components/blob/main/README.md#troubleshooting).

deeteeakl, 2025-02-12:
> So what is the current state of deep sleep support on BK7231N ? To this day I am still using the deepsleep component from [@Xmister](https://github.com/Xmister): [...]

I dream of getting 6-12 months life.  I am using the same deepsleep component and can't get the door sensor to last more than 24/36 hours.  I based my config on the example from this thread and have also disabled web server, reducing how frequently the battery voltage is checked but I can't seem to make an improvement.  I assume it's just not going into deep sleep but not sure how to confirm that.

dskw, 2025-02-16:
> I dream of getting 6-12 months life. I am using the same deepsleep component and can't get the door sensor to last more than 24/36 hours. I based my config on the example from this thread and have [...]

You can check my config which works pretty well.
Although I'm using the devices with windows that get opened 1-2 times per day the 2x rechargable AAA's I've in them last for the metioned 6-12 months. (depends on the use frequency)

I've attached my config for reference, without sensible data of course:
[bk_window.yml](https://github.com/user-attachments/files/18814637/bk_window.yml.txt)

I'm using static IPs where possible and MQTT to transmit the state and battery data ASAP.
The config includes some specifics about my device for the voltage divider and forced updates at boot for the components.
Works well enough for me that I don't touch it, out of fear an update might break their current performance.

I'll update a test device I have left once the deep sleep component was integrated

jon-daemon, 2025-02-17:
@szupi-ipuzs I tried to use it as a local external component but I still have an issue:
```
Compiling .pioenvs/test-bk7231n/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test-bk7231n/src/esphome/components/deep_sleep_libretiny/deep_sleep_libretiny.cpp.o
src/esphome/components/api/api_connection.cpp:13:10: fatal error: esphome/components/deep_sleep/deep_sleep_component.h: No such file or directory
   13 | #include "esphome/components/deep_sleep/deep_sleep_component.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/test-bk7231n/src/main.cpp.o
*** [.pioenvs/test-bk7231n/src/esphome/components/api/api_connection.cpp.o] Error 1
========================== [FAILED] Took 1.67 seconds ==========================
```

@dskw I used a config based on yours and I was able to put the bk7231n in deep sleep. My multimeter shows about 7.5μA.
Could you explain why did you use these options?
```
power_save_mode: LIGHT
passive_scan: true
```
I tried them on another device that's using `api` and deep sleep and I  had connectivity issues. On bk7231 it's fine but I'm not sure if they are needed.

szupi-ipuzs, 2025-02-17:
@jon-daemon follow [this](https://github.com/szupi-ipuzs/esphome-external-lt-deepsleep#fixing-the-unavailable-state-in-home-assistant) guide. I think originally it was only needed to fix the availability state in HA, but apparently they have changed something recently and it now is required for compilation.

dskw, 2025-02-21:
> [@szupi-ipuzs]
> [@dskw](https://github.com/dskw) I used a config based on yours and I was able to put the bk7231n in deep sleep. My multimeter shows about 7.5μA. Could you explain why did you use these options?
> 
> ```
> power_save_mode: LIGHT
> passive_scan: true
> ```
> 
> I tried them on another device that's using `api` and deep sleep and I had connectivity issues. On bk7231 it's fine but I'm not sure if they are needed.

I have not made use of the HA api because mqtt was processing way faster to push out the state of the sensor.

Honestly I can hardly remember why I've set them. It might be left-overs from trying different options to speed up the initial wifi connection and also preserve power.

The problem is that I initially went with "fast_connect: true" but that just made the connection for the beken chip take up to 5 minutes! So I was looking for alternatives.
But try without them. Looking at the docs at least passive_scan might not be needed and I'm not sure if libretiny is applying the power save mode at all. (I do hope so)

deeteeakl, 2025-02-22:
> You can check my config which works pretty well. Although I'm using the devices with windows that get opened 1-2 times per day the 2x rechargable AAA's I've in them last for the metioned 6-12 [...]

Thank you for this.  I have had this going for a week now and the battery level has not dropped at all.  I see what you mean about the time it takes to wake up and report status when the door/window is opened.  Its not a major issue, just not ideal.

dskw, 2025-02-26:
Yeah, the original firmware was able to wake up, send battery, state and firmware info in a maximum of one second. I was always surprised how fast they made it work.

Compared to this my config for esp-home takes 3 seconds at best and up to ~20 seconds at worst for a single update.
But I never managed to make it run any faster than this.

Btw, the config makes use of the pushbutton on my device. Clicking it makes the device turn on like with the reed switch.
Keep pressing the button will make the device stay awake, so it's easier to upgrade the firmware.