esphome/feature-requests #2468: IRremoteESP8266 libraries doesn't include bk7231T and bk7231N boards

yaminahacs, 2023-11-08:
IRremoteESP8266 libraries doesn't include bk7231T and bk7231N boards

**Describe the problem you have/What new integration you would like**
 Please add bk7231T and bk7231N boards in  IRremoteESP8266 libraries

**Please describe your use case for this integration and alternatives you've tried:**
 The firmware fails to compile on bk7231T and bk7231N boards if you include IRremoteESP8266 libraries because they are not supported

**Additional context**
 none

ssieb, 2023-11-08:
esphome doesn't have any control over the libraries.  I guess it should fail the config instead failing the compile though.

yaminahacs, 2023-11-08:
Config is fine and valid, but only fail to compile do to missing bk72xxx
components

ssieb, 2023-11-08:
You didn't provide any config or logs as requested by the template that you deleted...