esphome/feature-requests #2470: Support for M5STACK's ATOM-S3 C123 device with 0.85 OLED screen

den-mac, 2023-11-08:
**Describe the problem you have/What new integration you would like**
Hi, I am trying to get the screen to work on this device and am not having much luck. You can see my work to try and get it going in this thread:
(https://community.home-assistant.io/t/support-for-atom-s3-c123-device-with-0-85-oled-screen/637958)
I feel like I'm pretty close but so far I am not getting anything from the display to show up.

**Please describe your use case for this integration and alternatives you've tried:**
I found that the M5STACK SPI display is mentioned in [this article](https://esphome.io/components/display/ili9xxx), so I've been trying to use it but it's just not working. I think a new config/module/model for the SPI display might need to be added, but I'm not sure.

**Additional context**
My goal is to get this working with Home Assistant to use as a tiny display for data output.

randybb, 2023-11-08:
just add your wireless related stuff, api and ota
```yaml
esphome:
  name: atoms3
  on_boot:
    - priority: 800
      then:
        - lambda: |-
            id(display_tft).enable();
            id(display_tft).transfer_byte(0x11);
            id(display_tft).disable();

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

spi:
  clk_pin: 17
  mosi_pin: 21

display:
  - platform: st7789v
    id: display_tft
    model: Custom
    height: 128
    width: 128
    offset_height: 2
    offset_width: 1
    backlight_pin: 16
    cs_pin: 15
    dc_pin: 33
    reset_pin: 34
    eightbitcolor: true
    update_interval: 1s
    lambda: |-
      it.print(0, 0, id(font_default), "Hello World!");
      it.strftime(0, 0, id(font_default), TextAlign::TOP_LEFT, "%H:%M", id(ha_time).now());

time:
  platform: homeassistant
  id: ha_time
```

den-mac, 2023-11-08:
OK, initial test fails (no display) so I'll try and do some debugging. One thing I noticed though is with this new config esphome in HA sees the device as offline, like it's crashed internally or something.

Before at least it was online and HA showed a button device that worked when pressing the button on the unit.

den-mac, 2023-11-08:
OK, took a peek, wanted to ensure that I didn't inadvertently "burn up" my screen by picking the wrong pins for voltage, etc. ... and is that even possible?

Anyway, flashed it back to stock fw and it's back online with a working screen with the defaults. So the device is still good :-)

So I'll try tweaking your yaml file to see if I can get it to work. Am I really the first person to try this? Seems like it would be appealing to folks as a cheap, but small simple display for HA.

den-mac, 2023-11-08:
Thanks .. here's my complete config that doesn't do anything on the display. One question - where do you define `font_default` on your setup? As you can see below I am pointing to a custom font but it doesn't seem to make any difference. When I try your yaml I get an error about unknown `font_default`

```

esphome:
  name: atoms3
  on_boot:
    - priority: 800
      then:
        - lambda: |-
            id(display_tft).enable();
            id(display_tft).transfer_byte(0x11);
            id(display_tft).disable();

wifi:
  ssid: "******"
  password: "******"

# Enable Home Assistant API
api:
  encryption:
    key: "******"

ota:
  password: "******"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

font:
  - file: "comic.ttf"
    id: my_font
    size: 12

spi:
  clk_pin: GPIO17
  mosi_pin: GPIO21

display:
  - platform: st7789v
    id: display_tft
    model: Custom
    height: 128
    width: 128
    offset_height: 2
    offset_width: 1
    backlight_pin: GPIO16
    cs_pin: GPIO15
    dc_pin: GPIO33
    reset_pin: GPIO34
    eightbitcolor: true
    update_interval: 1s
    lambda: |-
      it.print(0, 0, id(my_font), "Hello World!");
      it.strftime(0, 0, id(my_font), TextAlign::TOP_LEFT, "%H:%M", id(ha_time).now());

time:
  platform: homeassistant
  id: ha_time

```

randybb, 2023-11-08:
Yea, I noticed that later so I updated my comment :) Is that comic.ttf in your config dir? The config with roboto will download it from google fonts repo.

den-mac, 2023-11-08:
Thanks! I'll try your updated code - I didn't notice the change in your original post so let's see how it goes. And yes, I have comic.ttf in my config directory as I've used it for other ESP8266/SPI display setups.

den-mac, 2023-11-08:
Hmph... oh well. Guess it's not in the cards for me. Nothing at all on the display, unfortunately. I'm sure there's something else outside of the scope of this device that needs to be configured or something.

The thing is, it stays offline the whole time after restarting the device with the new bin file. Meaning HA never sees it back online. And the wifi stuff is correct as I've been able to get this to connect to HA and expose a button device with a previous config.

den-mac, 2023-11-08:
Unfortunately, I'm not sure how to debug this. I'll look around and see if there's a process for troubleshooting these types of scenarios.

den-mac, 2023-11-08:
I will add one thing: When I make this change the device appears ONLINE in esphome in HA:

From:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
```

To:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
```

den-mac, 2023-11-08:
OMG, playing around worked. This works for me:

```

esphome:
  name: atoms3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

wifi:
  ssid: "******"
  password: "******"

api:
  encryption:
    key: "******"

ota:
  password: "******"

#G4=IR
remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 50%

#G41=Button
binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed

font:
  - file: "gfonts://Roboto"
    id: font_default
    size: 32

color:
  - id: my_red
    red: 100%
    green: 3%
    blue: 5%

time:
  platform: homeassistant
  id: ha_time

spi:
  clk_pin: 17
  mosi_pin: 21

display:
  - platform: st7789v
    id: display_tft
    model: Custom
    height: 128
    width: 128
    rotation: 180
    offset_height: 2
    offset_width: 1
    backlight_pin: 16
    cs_pin: 15
    dc_pin: 33
    reset_pin: 34
    eightbitcolor: true
    update_interval: 30s
    lambda: |-
      it.strftime(64, 0, id(font_default), TextAlign::TOP_CENTER, "%l:%M %p", id(ha_time).now());
```

den-mac, 2023-11-08:
[image]

randybb, 2023-11-09:
tested it again with latest dev and stable release, everything works with esp-idf. And yea, the on_boot part is not needed. No clue why I had it in my config. So no special settings are required and it is fully supported.

TiganSan, 2024-10-29:
I just came across this the YAML file worked on the first try.  Thank you very much for posting this info.

darquati, 2025-02-26:
My thanks for this too, got it working first go with this YAML and it was easy to then adjust it (I’m using it to read a thermometer probe and display the temperature)

Redspray00, 2025-03-04:
I'm baffled, neither one of the examples worked for me, just a blank screen. @darquati would you mind posting your code?
Throwing the webserver in the code I can see it's working. I'm at a loss...