esphome/feature-requests #2472: AHT-30 Temperature / Humidity Sensor Support (ESP32-S3-Box3 SENSOR-Dock)

ChristophCaina, 2023-11-08:
**Describe the problem you have/What new integration you would like**
The ESP32-S3-Box 3 comes with a Sensor-Docking Station.
Part of this Station is an AHT30 Temperature and Humidity Sensor.

Based on the information I found, it is connected to the I2C of the Chip.
[image]

SCL = GPIO40
SDA = GPIO41

**Please describe your use case for this integration and alternatives you've tried:**
I've tried to set the Sensor up with the platform AHT-10, but the Sensor value in HA remains UNKNOWN
It seems, that the AHT-30 is not yet supported by ESPHome, and it is not working with "AHT10" platform.
https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_3/hardware_overview_for_box_3.md

In the Documentation

**Additional context**

taylorfinnell, 2023-12-13:
With this https://github.com/esphome/esphome/pull/5198 I am getting correct values from my dock sensors (setting `variant: aht20`)

ChristophCaina, 2023-12-21:
I just wanted to link this PR since it is now in the official release... :)

closed 2023-12-21 by ChristophCaina

ChristophCaina, 2023-12-21:
@taylorfinnell: Are you using the S3-Box3 default VoiceAssistant Firmware?

taylorfinnell, 2023-12-21:
Yea, this works or seems to at first glance.

```
substitutions:
  name: esp32-s3-box-3-5ab3d8
  friendly_name: ESP32 S3 Box 3 5ab3d8
packages:
  esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  
api:
  encryption:
    key: "asdad"

ota:
  password: "asdasd"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-S3 Fallback Hotspot"
    password: "asda"

i2c:
  - id: bus_a
    sda: GPIO08
    scl: GPIO18
    scan: true

  - sda: GPIO41
    scl: GPIO40
    id: bus_b

sensor:
  - platform: aht10
    i2c_id: bus_b
    variant: AHT20
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 1s
```

ChristophCaina, 2023-12-21:
Thanks :)
Probably my issue was with the Bus then ^^
Will try it later