esphome/feature-requests #2473: Add conditional processing to the YAML file

rarroyo6, 2023-11-08:
**Describe the problem you have/What new integration you would like**

Add a way of conditionally processing parts of a YAML file based on defined variables, similarly to how C++ and some assembly language processors do it.
**Please describe your use case for this integration and alternatives you've tried:**

This will allow you to have generic YAML files, or "include" files, with multiple options, and the correct option will get processed based on a variable being defined in a "substitute" statement.
**Additional context**

For example:
You can have a main YAML file with:
```
substitutions:
  node_name: "my-node"
  node_display_name: "My node"
  enable_web: true

<<: !include common.yaml
```
Then in "common.yaml" you have the following:
```
esphome:
  name: $node_name
!ifdef node_display_name
  friendly_name: ${node_display_name}
!endif

!if $enable_web
web_server:
  port: 80
!endif
```
This way, if the "node_display_name" is not used in the substitutions, it won't give an error, and it will also allow to have additional sections included in the file, but only used when needed.

ssieb, 2023-11-08:
https://github.com/esphome/esphome/pull/3612

rarroyo6, 2023-11-15:
> [esphome/esphome#3612](https://github.com/esphome/esphome/pull/3612)

That issue seems to have been canceled because of no activity.

GraffJosh, 2023-12-13:
I would actually really like this functionality for multi-network behavior.

!if $network1_connected
MQTTBroker: server1
!else 
MQTTBroker: server2
!endif

Unfocused, 2024-04-11:
https://github.com/esphome/esphome/pull/3612 (Jinja2 templating) just ran out of steam - generally supported, but caught up on struggling to get agreement on the details. Much as I'd like full Jinja2 templating, just having some more basic building blocks (like `!extend` and `!remove`) would cover a LOT of use-cases without so much complexity.

I've been thinking about (but lacking time) cherry-picking just the `!if` stuff from the Jinja2 PR, since it looked like it could be easily added without needing everything else (the code for it in that PR doesn't necessarily need Jinja2), and it was the one thing that seemed to have unanimous support as-is.

And adding basic support for `!if` now wouldn't be incompatible with eventually adding full Jinja2 templating in the future.

Unfocused, 2024-04-11:
@GraffJosh wrote:
> I would actually really like this functionality for multi-network behavior.
> 
> !if $network1_connected MQTTBroker: server1 !else MQTTBroker: server2 !endif

Note that what's being proposed here wouldn't work like that. `!if` (like any `!` directive) is basically a pre-processor to build the YAML config - it doesn't work at run-time (on the device). What you're wanting to do in that example is already doable with a script/lambda.

rarroyo6, 2024-06-11:
What I talked about in the original post was a pre-processor directive, so the YAML could be written with all the different options, and only the parts that are needed would be compiled.
I was not talking about a run-time "if" statement, which can already be done with lambdas.

GraffJosh, 2024-06-11:
It's a fair point that conditionals in lambdas can be done with if statements generally. But those lambdas can't affect configurations (take for instance the mqtt server hostname). Having conditionals for the yaml at compile time also doesn't solve runtime issues such as the above.

rarroyo6, 2024-06-28:
Again, the problem is not with runtime, those are already available with if statements, what I want to do is have several compile options at "COMPILE" time, so we can have a single YAML with multiple options that can be COMPILED into the binary, based on options that are defined ahead of time. Much like is available on other programming languages like C or Assembly.

vvrein, 2024-09-26:
I think jinja may be used for yaml rendering, which should be familiar to the users. It is already used in Home Assistant for templating, and also it is widely used in Salt Stack for data structures rendering.

rarroyo6, 2024-09-27:
Once again, what I was requesting was a conditional processing at compile time NOT at runtime. Jinja processes conditions at RUNTIME, this is not what I am talking about. I'm talking about conditional inclusion of code at COMPILE time.
See [here](https://www.geeksforgeeks.org/cpp-preprocessors-and-directives/) for a sample of what I'm talking about for C++, look at the `#ifdef` and `#ifndef` directives, I want something similar for YAML.

ssieb, 2024-09-27:
This was about using jinja at "compile" time for processing the yaml file, so yes, what you're asking for.

vvrein, 2024-09-27:
@rarroyo6 in general terms jinja is a template engine and is used for rendering. Following that logic, it can be used anywhere when "rendering stage" occurs.  Consider this example:

Here `tmpl` variable just holds text, let's imagine that it is a yaml file contents.
In line `print(template.render(platform="esp8266"))` `platform` means variable, which is set to some content and will later be used at rendering stage in `{{ }}` or `{% %}` brackets.

```python
#!/usr/bin/env python3
from jinja2 import Template

tmpl = """
# This is example for {{ platform }}
{%- if platform == "esp8266" %}
preferences:
  flash_write_interval: 1min
{%- endif %}

logger:

ota:
  password: !secret ota_password
  platform: esphome

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password

web_server:
  port: 80
  ota: False
{%- if platform == "esp32" %}
  version: 3
{%- endif %}
"""

template = Template(tmpl)
print(template.render(platform="esp8266"))
print(template.render(platform="esp32"))
```

We will get different yaml contents in case of different `platform`s. 
(There is an issue with setting the "built-in" variables like `platform`, because the board variants are by itself defined in the yaml, and here we get chicken and egg problem. Also, which variables should be available beforehand is a separate question)

```yaml
# This is example for esp8266
preferences:
  flash_write_interval: 1min

logger:

ota:
  password: !secret ota_password
  platform: esphome

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password

web_server:
  port: 80
  ota: False
```
```yaml
# This is example for esp32

logger:

ota:
  password: !secret ota_password
  platform: esphome

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password

web_server:
  port: 80
  ota: False
  version: 3
```

vvrein, 2024-09-27:
So, yes, jinja may not be the great option in terms of rendering the config leaning on the earlier data in the config.

archef2000, 2024-09-29:
What would be if the variables from the substitutions are used and there is just an error when jinja is used bevor the substitutions?

bojanpotocnik, 2025-04-19:
Any updates on this? I found this while searching for exactly something like
```yaml
sensor:
  - platform: uptime
    name: "Uptime"

{%- if platform == "esp32" %}
  - platform: internal_temperature
    name: "Internal Temperature"
{%- endif %}
```
so it can be put into `device_base.yaml` which I include in all devices (configuring API, OTA, Time, ...).

jpeletier, 2025-11-27:
@bojanpotocnik For your example, you could do:

```yaml
esphome:
  name: test

esp32:
  board: esp32dev

substitutions:
  platform: esp32

sensor:
  - platform: uptime
    name: "Uptime"
  - platform: internal_temperature
    id: internal_temp_sensor
    name: "Internal Temperature"
  - id: !remove ${"internal_temp_sensor" if platform != "esp32" else "nothing"}
```

Regarding everyone else, stay tuned and support https://github.com/esphome/esphome/pull/12126 which will do what you want!

jpeletier, 2025-11-28:
> +1
> 
> I was looking for a way to conditionally include a remote package and ran into this issue. Isn’t this fundamental to reusable code? If it’s not supported yet, could we at least allow substitutions within packages (ref, file) as a workaround to control YAML contents?

@f-io Could you create a feature-request? I have something in the pipe that is near what you describe.

jpeletier, 2026-01-19:
@rarroyo6 you already have the functionality you describe in your OP, but with a different syntax. In your OP you have the following:

```yaml
esphome:
  name: $node_name
!ifdef node_display_name
  friendly_name: ${node_display_name}
!endif

!if $enable_web
web_server:
  port: 80
!endif
```

With the new features you'd write it as follows:

```yaml
substitutions:
  web_server_package:
    web_server:
      port: 80
  friendly_name_package:
    esphome:
      friendly_name: ${node_display_name}

esphome:
  name: $node_name

packages:
  - ${friendly_name_package if node_display_name else {} }
  - ${web_server_package if enable_web else {} }
```

The above defines two variables, `web_server_package` and `friendly_name_package` that contain an entire package. Then, under the `packages:` key, those entire packages are included or not, depending on the value of `node_display_name` or `enable_web`.

Alternatively, for modularity, those packages can be in a different file:

```yaml
substitutions:
  web_server_package: !include web_server.yaml
  friendly_name_package: !include friendly.yaml

esphome:
  name: $node_name

packages:
  - ${friendly_name_package if node_display_name else {} }
  - ${web_server_package if enable_web else {} }
```

f-io, 2026-01-19:
@jpeletier 

Maybe I just haven’t spent enough time on this yet, but what I’m trying to do is something simple as this:

```yaml
substitutions:
  model_name: "pure" # pure | combi | elite | combi_elite
  branch_ref: "main" # main | dev

....

external_components:
  - source:
      type: git
      url: https://github.com/...
      ref: ${branch_ref}
    refresh: 1s
    components: [sauna360]

packages:
 ....
  model:
    url: https://github.com/...
    ref: ${branch_ref}
    refresh: 1s
    files: ["models/${model_name}.yaml"]
```

Substitution in external_components.source.ref works, but it doesn’t work in packages—neither in ref nor in files.