esphome/feature-requests #2475: Support for Radar-Sensor of the S3-Box3 "Sensor-Dock"

ChristophCaina, 2023-11-09:
**Describe the problem you have/What new integration you would like**
The Espressif S3-Box3 comes with an additional Sensor-dock.
This Sensor-dock contains an IR-Transmitter, IR-Receiver, a [AHT-30 Temperature & Humidity Sensor](https://github.com/esphome/feature-requests/issues/2472), as well as a radar-based presence detector.

Based on the Schematics, the AHT-30 and the Radar Sensor are using the I2C-Bus, both should be available on the following ports:
```
i2c:
  sda: GPIO41
  scl: GPIO40
```

From the technical documentation, the Radar-Sensor should be an MS58-3909S68U4

**Please describe your use case for this integration and alternatives you've tried:**
Support components for the Espressif ESP32-S3-Box3

**Additional context**

thejevans, 2023-12-14:
It looks like Espressif recently released a driver for this: https://components.espressif.com/components/espressif/at581x

jaymunro, 2024-01-06:
I have the radar working. Here is a RADAR config that:

- adds an occupancy sensor to the **Sensors** section.
- adds a configurable occupancy duration to the **Config** section (default: 60s).
- adds a switch to the **Config** section that mutes the box and turns off the backlight when nobody is present (default: not enabled).

The last helps if on battery and also reduces network chatter.
The first can be used in HA automations to turn on lights, fans, etc.
```
switch:
  - platform: template
    name: "Mute when absent"
    id: mute_when_absent
    icon: mdi:account-right-arrow
    optimistic: true
    entity_category: config
    restore_mode: RESTORE_DEFAULT_OFF

number:
  - platform: template
    name: "Presence duration"
    id: radar_delayed_off
    icon: mdi:account-clock
    optimistic: true
    restore_value: true
    initial_value: 60
    min_value: 0
    step: 5
    max_value: 300
    unit_of_measurement: s
    entity_category: config
    mode: box

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO21
    name: "Presence detect"
    disabled_by_default: false
    device_class: "occupancy"
    filters:
      - delayed_off: !lambda return id(radar_delayed_off).state * 1000;
    on_release:
      then:
        - if:
            condition: 
              switch.is_on: mute_when_absent
            then:
              - switch.turn_on: mute
              - light.turn_off: led
    on_press:
      then:
        - if:
            condition: 
              switch.is_on: mute_when_absent
            then:
              - switch.turn_off: mute
              - light.turn_on: led
```
It is working well for me. I’d be interested to hear how well it works for others too. There was an initial issue for me with the device starting muted and not waking due to the absence switch defaulting to off, however I haven’t been able to replicate this.

Remember to clean your build files before install.

Enjoy.

lajoshanko, 2024-01-06:
Seems to be working but sometimes it got cleared even if I sit in front of it. Thanks for your work!

ChristophCaina, 2024-01-06:
> Seems to be working but sometimes it got cleared even if I sit in front of it. Thanks for your work!

That might be the Case, If you don't move enough for the Sensor to be recognized^^

jaymunro, 2024-01-07:
> > Seems to be working but sometimes it got cleared even if I sit in front of it. Thanks for your work!

As Chris mentioned, the radar can see what's in front of it, chair, wall, plant, person, but if none of them move, it doesn't trigger. Radar does not have the resolution to understand what is in front of it, only that something has changed position since the last scan.

If it turns off too much for you just increase the **Presence duration**.

Pat0856, 2024-01-12:
Dear Jaymunro, 
the radar configuration is also working for me. Thank you very much.
My ESP32 S3 Box3 is integrated in Home Assistant / Esphome/ Voice Assistant.
 Besides the radar, other items like Temperature, Humidity, Batterie Level an Batterie voltage are working very well.

jaymunro, 2024-01-13:
This seems to be working well for everyone (also [here](https://community.home-assistant.io/t/esp32-s3-box3/638287/181)). If everyone agrees I can submit a PR to get this out for everyone else going forward.

ScottG489, 2024-01-22:
I have it "working" as well using part of @jaymunro yaml config. However, it seems to behave more like a PIR than mmwave sensor. Wasn't able to find much info on "MS58-3909S68U4".

Is it just a really bad mmwave sensor or can improved drivers for it help?

And sure, increasing the timeout can help alleviate this problem, but yeah I'd like to use it as a presence sensor not PIR which is typically how you deal with those kinds of problems.

jaymunro, 2024-01-22:
Can you clarify what you mean by it behaving like a PIR sensor? For me it is way more sensitive than a PIR, a few centimetres of movement 3-4 metres away can trigger it. My experience with PIR often involves waving arms around to trigger them.

If you mean by the lack of ability to configure zones, sensitivity and such, then yes, to my understanding that is
1. the lack of drivers (i.e. a 'component' in esphome
2. I2C bus issues (https://github.com/esphome/issues/issues/5296)

The chip would need to be configured by sending parameters over the I2C bus but the I2C is presently conflicting with the audio (maybe whole I2S bus) in ESPHome and there is presently no component in esphome for that chip that I am aware of.

Once there is support, we should be able to configure the chip beyond its default configuration which seems what we are stuck with at present.

ScottG489, 2024-01-23:
Just playing around with it again, I still have to say its sensitivity feels about on par with PIR sensors I've used. It resets presence a lot faster obviously, but yeah it is very insensitive.

In any case, I hope it is just a case of needing drivers to increase the sensitivity so it'll be usable as a typical mmwave sensor in the future.

Also off topic, but does anyone have an ESPHome config for the temperature sensor?

Pat0856, 2024-01-30:
hello at all
I have no conflicts with VA when using I2C .
Besides the radar, other items like Temperature, Humidity, Batterie Level an Batterie voltage are working very well.

Vladx71, 2024-01-30:
Can you please share the whole config with the community? I have snippets
from several places, but it would be great to see a config already proved
to be stable :)

Thank you
Laszlo

jaymunro, 2024-01-30:
> I have no conflicts with VA when using I2C .

Did you power down and restart the box? I thought the same too until I did a cold restart.

jaymunro, 2024-01-30:
My current config is:
```yaml
substitutions:
 name: esp32-s3-box-3-5ad0fc
 friendly_name: Jarvis

packages:
 esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main

esphome:
 name: ${name}
 name_add_mac_suffix: false
 friendly_name: ${friendly_name}

api:
 encryption:
   key: rTJvoot5F8Z3gFEQHtaEFjXYV05tFMJ2TLON6zReaGc=

wifi:
 ssid: !secret wifi_ssid
 password: !secret wifi_password

sensor:
 - platform: adc
   pin: GPIO10
   name: "Battery voltage"
   id: battery_voltage
   unit_of_measurement: "V"
   accuracy_decimals: 3
   device_class: "voltage"
   entity_category: "diagnostic"
   disabled_by_default: true
   update_interval: 30s
   attenuation: auto
   filters:
       - multiply: 4.01

 - platform: copy
   source_id: battery_voltage
   name: "Battery level"
   unit_of_measurement: "%"
   accuracy_decimals: 1
   device_class: "battery"
   entity_category: "diagnostic"
   filters:
     - lambda: return (x - 3.1) / (4.14 - 3.1) * 100;
     - clamp:
         min_value: 0
         max_value: 100
         ignore_out_of_range: true

switch:
 - platform: template
   name: "Mute when absent"
   id: mute_when_absent
   icon: mdi:account-right-arrow
   optimistic: true
   entity_category: config
   restore_mode: RESTORE_DEFAULT_OFF

number:
 - platform: template
   name: "Presence duration"
   id: radar_delayed_off
   icon: mdi:account-clock
   optimistic: true
   restore_value: true
   initial_value: 60
   min_value: 0
   step: 5
   max_value: 300
   unit_of_measurement: s
   entity_category: config
   mode: box

binary_sensor:
 - platform: gpio
   pin:
     number: GPIO21
   name: "Presence detect"
   disabled_by_default: false
   device_class: "occupancy"
   filters:
     - delayed_off: !lambda return id(radar_delayed_off).state * 1000;
   on_release:
     then:
       - if:
           condition:
             switch.is_on: mute_when_absent
           then:
             - switch.turn_on: mute
             - light.turn_off: led
   on_press:
     then:
       - if:
           condition:
             switch.is_on: mute_when_absent
           then:
             - switch.turn_off: mute
             - light.turn_on: led
```

Pat0856, 2024-01-31:
Hello 
no need to write down my configuration. It's exactly the same as Jaymunro's

Unfocused, 2024-04-11:
This will be in ESPHome 2024.4.0, usable with the `at581x` component. The just released beta version (2024.4.0b1) includes it:
- https://github.com/esphome/esphome/pull/6297

Docs:
- https://deploy-preview-3646--esphome.netlify.app/components/at581x

ChristophCaina, 2024-06-15:
Hi,
I've implemented the at581x as explained in the documentation:

**Documentation:**
```
# Example configuration entry
at581x:
  id: "Radar"
  i2c_id: bus_a

binary_sensor:
  - platform: gpio
    pin: GPIO21
    name: "Radar motion"

switch:
  # Switch to turn on/off RF emission
  - platform: at581x
    at581x_id: "Radar"
    name: "Enable Radar"
```

**Implementation:**

```
at581x:
  id: "Radar"
  i2c_id: bus_a

i2c:
  - id: bus_a
    sda: GPIO8
    scl: GPIO18
    frequency: 100kHz
    scan: True

  - id: bus_b
    sda: GPIO41
    scl: GPIO40
    frequency: 100kHz
    
   - platform: at581x
    at581x_id: "Radar"
    name: "Enable Radar"

  - platform: template
    name: "Presence duration"
    id: radar_delayed_off
    icon: mdi:account-clock
    optimistic: true
    restore_value: true
    initial_value: 60
    min_value: 0
    step: 5
    max_value: 300
    unit_of_measurement: s
    entity_category: config
    mode: box

  - platform: gpio
    pin:
      number: GPIO21
    name: "Presence detect"
    disabled_by_default: false
    publish_initial_state: true
    device_class: "occupancy"
    filters:
      - delayed_off: !lambda return id(radar_delayed_off).state * 1000;
    on_release:
      then:
        - if:
            condition:
              switch.is_on: mute_when_absent
            then:
              - switch.turn_on: mute
              - light.turn_off: led
    on_press:
      then:
        - if:
            condition:
              switch.is_on: mute_when_absent
            then:
              - switch.turn_off: mute
              - light.turn_on: led
```

But I don't see any difference in the behave of the device between switching the "enable radar" on or off...
Any idea?
Does it use bus_a or bus_b on the Box3 ?

closed 2024-06-23 by ChristophCaina