esphome/feature-requests #2476: Add option to remove the prefixing of "friendly_name" since this affects the voice assistant

rarroyo6, 2023-11-10:
**Describe the problem you have/What new integration you would like**

The prefixing of "friendly_name" to entities should be optional, or have a way to disable it.
**Please describe your use case for this integration and alternatives you've tried:**

When the "friendly_name" gets prefixed to the name of an entity, especially when a ESP node handles multiple different type of entities, it can sometimes make the name incoherent. This can sometimes be acceptable to look at, but when the name is used by voice assistant, it can make the sentence completely awkward or too wordy.
I know that the name can be changed in the UI, or an alias can be created, but it should be possible to do this from the beginning.
**Additional context**

Here is an example of the problem:
I have a ESP node with a friendly name of "Pool chemical controller", then in that node I have entities "Chlorine pump", "Acid pump", plus a few others and a variety of sensors.
If I want to turn on the chlorine pump, right now I have to say "Turn on the pool chemical controller chlorine pump", I should be able to just say "Turn on the chlorine pump".

rarroyo6, 2023-11-26:
> as a workaround, you could use the Alias feature in HomeAssistant for changing the entity name... [image]

I mentioned in the original post, that I know that I could add an alias, but the idea is to make it simpler, and not to have to take extra steps in order to make it work right.

clinta, 2023-12-07:
The purpose is incredibly useful to me. I have several of the same device type, like an s31 relay for example, and I use templates to mange the configuration. So in my template I can have something simple like `name: Button`, then if the friendly name is "Living Room Lamp", in home assistant I get "switch.living_room_lamp_button".

I depend on this, and if it is optional it should be opt-out, except in one case. In the case of an empty name like `name: None` or `name: ""`, the documented behavior is that this should use the friendly_name, but because of the prefixing it is repeated. This should be fixed by default. I should not end up with an entity of `switch.living_room_lamp_living_room_lamp`, when I use `name: None` for the relay. It should just be `switch.living_room_lamp`.

Linwood-F, 2023-12-28:
First, let me say I hate the whole concept of the automatically prefixed friendly name, and it was absolutely the most disruptive rollout of anything I have encountered in HA in many years of using it. I think many of us use various tricks to work around it (e.g. using name with a single space as opposed to empty strings).

That said, I think some care has to be taken in changing it yet again. The other thing that has happened is 3rd parties have rolled out very useful (and widely used) git based implementations, which make it more complicated to change THEIR code (e.g. forking and merging updates -- and HA thinks YAML is hard to edit!). 

For example, I personally would like to see any such control over whether or not to prefix be locally controlled, and by that I mean in different parts of the same esphome program.  If I have includes from GIT, and local includes of my own standard code in the same file, being able to have some kind of pragma declaration might work, where one is #PREFIX OFF and #PREFIX ON (clearly not that syntax, just as an example) that has a respected scope. 

But please... stop breaking existing code for this.  It was a royal mess most have gotten past, I am sure it was well intentioned but it was a mess.  Don't make another with trying to add more lipstick to this pig.

rarroyo6, 2024-01-10:
This change has affected Voice Assistant, Automations, Scripts, and Dashboards.
I'm still finding some rarely used automations that when I realize they didn't work and check why, it turns out that they have the old name before it was prefixed.
I've also come across some old EspHome nodes that had not been updated, and were working fine. After updating them, the names changed and created more problems.

alexruffell, 2024-02-29:
I have this issue too. My most common use case is when I create devices in ESPHome that can be turned on or off, or opened and closed.

I am modifying the default YAML of the ratgdo and since my ratgo is called "Overhead Garage Door", I do not want anything added to the entity name... in other words it must match the device name. I have tried a few things:

name: None (does not work, I just get "none" as a name)
name: "" (does not work)
name: " " (works as desired but it seems more like exploiting a bug)

My end goal is for the cover entity that controls the "Overhead Garage Door" to be called `cover.overhead_garage_door` which is also the name of the device (ratgdo). Simply put, some way for an entity to get the device name and nothing else would be great. Before the automated prepending was done, I had full control and it worked perfectly... sadly the "facilitation" works in most cases but caused major annoyance in these particular cases (frequent in my case).

Summary: What @clinta said sounds perfect - https://github.com/esphome/feature-requests/issues/2476#issuecomment-1844034117

kooswillem, 2024-04-02:
You can overwrite (change) the entity ID in 'Developer Tools -> States page' by finding the entity, clicking on the 'i' icon (more information) and then in the pop-up window click the sprocket icon. Change the entity ID and confirm by clicking 'Update' in the left bottom corner of this pop-up window.

nagyrobi, 2024-04-02:
> You can overwrite (change) the entity ID in 'Developer Tools -> States page' by finding the entity, clicking on the 'i' icon (more information) and then in the pop-up window click the sprocket icon. [...]

You should avoid doing that on long term and with many entities. If for any reason in the future you need to remove and re-add the ESPHome node to HA, you'll have to do the renames again manually, for all your automations and data history to be functional again.

Unfocused, 2024-04-11:
@alexruffell wrote:
> name: None (does not work, I just get "none" as a name)

This should work - I'd suggest filing a separate bug report (not feature request) just for this.

Unfocused, 2024-04-11:
With specifying `None` as the name, some of the use-cases mentioned here would work well with the sub-device/multi-device config being added in this PR:
- https://github.com/esphome/esphome/pull/6484

For background discussion on that, see:
- https://github.com/esphome/feature-requests/issues/1335

DrBlokmeister, 2024-06-14:
I'm also really surprised that this change was suddenly rolled out and I find that all my ESPHome sensors now are named something like sensor.plug_nas_plug_nas_power, instead of sensor.plug_nas_power and I would need to manually modify around 20 sensors per device. 

Please give us the option to remove prefixing of friendly_name!

s-wachspress, 2024-09-01:
I found this thread while looking for a solution to this issue and discovered that if you use a space for the friendly name, the whitespace prefix will get trimmed automatically, avoiding any automated prefix being added to the other entity's name properties:
```
esphome:
  friendly_name: " "

button:
  - platform: template
    name: "TV Power"
    id: tv_power_on
```

Automatically shows up with the name "TV Power" in my list of Entities.
Notably, this won't work if you provide a completely empty string: ""

rarroyo6, 2024-09-18:
> I found this thread while looking for a solution to this issue and discovered that if you use a space for the friendly name, it get's recognized as a valid prefix, but the whitespace gets trimmed from the entity name automatically,
> 
> ```
> esphome:
>   friendly_name: " "
> 
> button:
>   - platform: template
>     name: "TV Power On"
>     id: tv_power_on
> ```
> 
> This config automatically generated an entity with the name "TV Power On" in my list of Entities.
> 
> Notably, this won't work if you provide a completely empty string: "" I'm pretty sure you also need to Clean Build Files before applying this config change.

The blank space works for the individual entity, but it defeats the purpose of the friendly name, which is to name the node.
You could have a node with a generic friendly name, to identify it in the ESPHome dashboard, but that node can have multiple unrelated entities, which won't make any sense when the friendly name gets appended.
For example: I have a node with the friendly name "Backyard Sonoff 4ch relay board", which makes it easy to identify on the ESPHome dashboard, but this node controls multiple unrelated entities, like Fence lights, Tree lights, Fairy lights, and Pond light, when the friendly name gets appended, the names become "Backyard Sonoff 4ch relay board Fence lights", etc...
This should be an option, but it should not be forced.
I have 62 nodes, and more planned, and most have multiple entities. It becomes a hassle to manually rename the entities after making a change, or after removing and adding it again.

clinta, 2024-09-18:
@rarroyo6 did you intend to close this?

DrBlokmeister, 2024-09-19:
@rarroyo6 please re-open the issue as it has not been solved yet AFAIK.

rarroyo6, 2024-09-21:
Sorry about that, I accidentally clicked on the submit and close button.

Naguz, 2024-12-05:
Is this on any roadmap? I have had to stop using friendly name altogether. This gives a slightly less appealing, but at least sensible entity names. For any device with more than one entity I find this to be an absolute necessity, as the hack with " " as entity name obviously does not work for these.

FreAcX, 2024-12-22:
Vote up

sumasage, 2024-12-23:
vote up +1

mainakae, 2025-04-03:
Vote up +1

SparkyDan555, 2025-05-18:
I agree the friendly name shouldn't be added to entity names. It is supposed to prevent _"less intuitive names and a less polished experience in Home Assistant"_ but in reality it makes things more messy:

[image]

In Apple Home:
[image]

juicejuice, 2025-06-09:
I have a similar issue to the above. I have an ESPHome device like "Hallway Switch". It controls four different lights that are in totally different parts of the house. Hallway Switch Side Gate Light doesn't really roll off the tongue...

Makr91, 2025-07-01:
This seems to have become a bigger problem starting in April this year

https://github.com/esphome/feature-requests/issues/3174
https://github.com/esphome/feature-requests/issues/2732

caco3, 2025-07-06:
Oh bugger, it even gets odder because it is not consistent.
I just started using some smart plugs driven by ESPHome. Points out the shown duplication depends on the used view:
### Masonry View
Entity card:

[image]

### Sections View
Entity card:

[image]

---

I am using
-  Core 2025.7.1
- Frontend 20250702.1

MarkProminic, 2025-10-05:
I agree with @DrBlokmeister

mateuszdrab, 2026-11-16:
I think a lot has changed since then, entity generation is almost perfect. 

You can create an ESPhome device with name Test Device, assign it to area Room and have a sensor called Temperature and HA will generate the right `room_test_device_temperature` as entity ID, but if you put the device friendly name in ESPhome as `Room Test Device`, then it gets messy because HA appends the area name as well as we get `room_room_test_device_temperature`. 

This can be changed so area is not appended automatically since recently, but I actually want it that way.

What I think we need, is the ability for HA to automatically strip area name from device name, so in ESPhome we can have `Room Test Device` and `Another Room Test Device` and HA would not duplicate the area name when generating entity IDs. Another way to do this is to name the friendly names as `Test Device` for both devices in ESPhome, but then the dashboard there looks ugly with two identically named devices, but at least HA doesn't double prefix the names out of box. 

Yet another option is to use room qualified name in ESPhome, manually rename in HA to remove it. This gets us the best of both. If it was automatically stripped, it would not only fix it for ESPhome but also every other integration that might have the same issue.

And lastly, the least useful fix (due to it being ESPhome only, leaving other integrations still unaddressed) would be to automatically append area name inside ESPhome's dashboard to the friendly name, so that the view makes it easy to distinguish devices in the build log; and HA can have clear not area qualified name for its use.

Also posted here https://github.com/esphome/feature-requests/issues/2732#issuecomment-6930114320

I also can't wait for it to get addressed, voice assistant suffers due to same and I have to create aliases also.