esphome/feature-requests #2479: Possibility to define pin aliases

alexiri, 2023-11-11:
**Describe the problem you have/What new integration you would like**

All of my devices include a basic yaml with general configuration, like `api`, `ota`, etc. This configuration also includes `status_led` with `pin: LED`. This works fine with commercial boards which have a LED alias defined, but it doesn't work for custom boards.

I'd like to be able to define custom pin aliases, something like this:

```
esp8266:
  board: esp8285
  pin_aliases:
    LED: 13
```

**Please describe your use case for this integration and alternatives you've tried:**

I can't reuse my common yaml templates unless the board has a defined LED alias already.

**Additional context**

This might also be solved by something like https://github.com/esphome/feature-requests/issues/2348, but this solution might be simpler.

ssieb, 2023-11-11:
Why not just use a substitution?

alexiri, 2023-11-11:
I guess I could, but this would force me to define it in all the other devices that already have a pin alias. Kind of defeats the purpose of aliases, then. I guess it wouldn't matter if I could define define the `status_led` pin as `pin: LED || $LED` or something like that.