esphome/feature-requests #2480: State based Icons

ChristophCaina, 2023-11-11:
**Describe the problem you have/What new integration you would like**
ESPhome allows to define Icons for Sensors as well as for binary Sensors.

This is great, because in ha this Icon will be shown in the ui.

It would be great, If you could also configure Icons based in their state - especially for binary Sensors...

**Please describe your use case for this integration and alternatives you've tried:**
There's probably a workaround with Template Sensors in ha, but IT would be great If ESP Home could Support this within the config

**Additional context**

Unfocused, 2024-05-05:
For reference, here's Home Assistant's documentation on this:

- https://developers.home-assistant.io/docs/core/entity/#icon-translations

ChristophCaina, 2024-05-24:
lol...
I didn't remember that I already raised this...
Just found my own questions after searching again for a solution for this :)

nagyrobi, 2024-05-24:
At least, if `icon` would be templatable, we could probably set it through lambdas dynamically.
Not sure if it would go through the API though...

ChristophCaina, 2025-01-04:
Any Progress on this?
I know, it's only a minor Thing without big Impact ...

jorelvaccaro, 2025-02-18:
Hi Christoph - Finding your requests on Home Assistant community and here regarding state-based Material Design Icons (MDI icons), so I'm adding on to your feature request.

I'd like to do this as well, and more than just with binary sensors. For instance, as a use case: i'm using esphome to turn a knob with a programmed number slider and a servo. I'd like to depict ranges with the fan-speed-1, fan-speed-2, and fan-speed-3 mdi icons based on the knob setting.  With a slider number with a range of 1 to 100, the fan speed 1 icon would represent 1 to 33, and so on.

ChristophCaina, 2025-05-04:
The first thing - in my mind would add the support for state-based icons where they are already available in HA itself.
That are - binary sensors - and in addition to that, some other device / Sensor classes... but I am not sure, which ones those are.

But since the initial request is now over a year old... I am having doupts that we might see this in the near future.
I guess, focus is more on voice-stuff atm