esphome/feature-requests #2481: add i2c support for esp32-c6

TRusselo, 2023-11-14:
**Describe the problem you have/What new integration you would like**

ESP32-c6-devkitm-1
ESP32-c6-devkitc-1

both have i2c controller, but editor says c6 is not supported

[image]

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

linkedupbits, 2023-11-14:
Looks like it needs the variant defined here: https://github.com/esphome/esphome/blob/dev/esphome/components/i2s_audio/__init__.py

TRusselo, 2023-11-14:
was chatting with jesserockz in discord

he had my add

external_components:
  - source: github://esphome/esphome@jesserockz-2023-409
    components: i2s_audio
    refresh: 0s

to my config

it failed compile

log :
[logs_esp-mic_run.txt](https://github.com/esphome/feature-requests/files/13344595/logs_esp-mic_run.txt)

mazdrab85, 2025-01-02:
I know this is an old thread, but I have the same issues after trying these steps as well.

trianglesis, 2025-03-22:
I'm bumping this request with my C6 variant too!

```log
INFO ESPHome 2025.2.2
INFO Reading configuration my_proj/esp32-c6-1_mic.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO8 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

i2s_audio: [source my_proj/esp32-c6-1_mic.yaml:73]
  
  Unsupported variant ESP32C6.
  - i2s_lrclk_pin: 18
    i2s_bclk_pin: 23
```

Also, it will be great to move sucha  project to a new board too:
- https://github.com/stas-sl/esphome-sound-level-meter

spires0, 2025-08-07:
Should we perhaps add this incompatibility to the [documentation page for the i2s_audio component](https://esphome.io/components/i2s_audio.html)? It says right now that i2s "only works on ESP32 based chips", and perhaps that is meant to exclude the RISC-V esp32 variants but perhaps it should be more explicit now that esp32-c6 is officially supported by esphome generally.

Rvh91, 2025-09-03:
Same situation here.. soldered over both my older d1_mini boards to these newer boards. Only to find out here they are unsupported.

kabakaev, 2025-09-04:
I've fixed esphome code to support i2s on esp32-c6. See [the PR description](https://github.com/esphome/esphome/pull/10576) on how to compile the feature branch.

> What changes would need to be done for a simple microphone?

@Rvh91, follow the PR description, but instead of `speaker` and `media_player` in the config.yaml, add [`microphone` component](https://esphome.io/components/microphone/i2s_audio/):

<details><summary>Click here to see the i2s-microphone example config.yaml</summary>

```yaml
esphome:  # https://esphome.io/components/esphome.html
  name: test-i2s-on-c6
  platformio_options:
    upload_port: /dev/ttyACM1
    framework: espidf
    board_build.flash_mode: dio

esp32:  # https://esphome.io/components/esp32.html
  variant: esp32c6
  board: esp32-c6-devkitc-1
  framework:
    advanced:
      compiler_optimization: PERF
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_DETECT: y

# Enable logging
logger:  # https://esphome.io/components/logger.html
  #level: NONE
  #level: INFO
  level: DEBUG
  hardware_uart: UART0

psram:
  disabled: true

debug:  # https://esphome.io/components/debug/
  update_interval: 2s

sensor:
  - platform: debug
    free:
      name: "Heap Free"

i2s_audio:
  id: i2s_audio_bus
  i2s_lrclk_pin: GPIO6
  i2s_bclk_pin: GPIO5
  # use_legacy: true

microphone:
  - platform: i2s_audio
    id: external_mic
    adc_type: external
    i2s_din_pin: GPIO4
    i2s_audio_id: i2s_audio_bus

```

</details>

Rvh91, 2025-09-04:
Hi @kabakaev , thank you for your effort.
I must admit that it is not directly clear from the description of the PR how I can use your branch in my esphome add-on in home assistant. I had expected to see some 'external component' somewhere in the yaml, pointing to your GitHub repo. Is there a way for me to do so?

kabakaev, 2025-09-04:
> I had expected to see some 'external component' somewhere in the yaml, pointing to your GitHub repo. Is there a way for me to do so?

Do you mean [something like this](https://github.com/esphome/esphome/pull/10576#issuecomment-3254658969)?

zastrixarundell, 2025-11-27:
I am having difficulties with this uController for voice assistant. I'm not exactly sure whether I made an issue somewhere but I'm not getting anything on home assistant regarding the voice. Just want to make sure that my yml configuration and that it might just be a hardware defect on the microphone.

<details>

<summary>My yml config</summary>

```yml
esphome:
  name: "jarvis-office"
  friendly_name: Jarvis-Office
  platformio_options: 
    board_build.flash_mode: dio

esp32:
  variant: esp32c6
  board: esp32-c6-devkitc-1
  framework:
    advanced:
      compiler_optimization: PERF
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_DETECT: y

# Enable Home Assistant API
api:
  encryption:
    key: "s+Nr8n1kKtDdSY79SN1oAjvCtwydwUoPjyv3HXwgX40="

ota:
  - platform: esphome
    password: "ed6c734a8d4ee7ff645d46bba49cdc8b"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jarvis-Office Fallback Hotspot"
    password: "aQUM2h9xk4bU"

# Enable logging
logger:  # https://esphome.io/components/logger.html
  #level: NONE
  #level: INFO
  level: DEBUG
  hardware_uart: UART0

debug:  # https://esphome.io/components/debug/
  update_interval: 2s

sensor:
  - platform: debug
    free:
      name: "Heap Free"

i2s_audio:
  id: i2s_audio_bus
  i2s_lrclk_pin: GPIO6
  i2s_bclk_pin: GPIO5

microphone:
  - platform: i2s_audio
    id: external_mic
    adc_type: external
    i2s_din_pin: GPIO4
    i2s_audio_id: i2s_audio_bus

micro_wake_word:
  id: va_wake_word
  microphone: external_mic
  models:
    - model: okay_nabu

voice_assistant:
  microphone: external_mic
  auto_gain: 31dBFS
  noise_suppression_level: 0
  volume_multiplier: 4.0
  micro_wake_word: va_wake_word
  use_wake_word: True
```
</details>