esphome/feature-requests #2482: Secondary mqtt settings for dockerised/isolated instances

MiggyMan, 2023-11-15:
**Describe the problem you have/What new integration you would like**
With esphome inside a docker container esphome itself is unable to connect to the same mqtt broker via the address devices use it would however be able to connect via the docker network, having the ability to provide a second set of mqtt settings explicitly for esphome to use when talking to devices for logs would solve this, this could simply be a setting for an alternate broker, ideally reusing authentication settings from the original setting unless overidden 

**Please describe your use case for this integration and alternatives you've tried:**
Access to mqtt logging from within esphome when dockerised.

nagyrobi, 2023-11-15:
Not quite clear what you mean, but perhaps specifying your MQTT broker by an internal DNS name, which could resolve different addresses from the devices and from behind your docker.

MiggyMan, 2023-11-15:
Possible but dns inside docker is a bit of a pain so id need to either use hosts or setup and internal resolver (or depending on how the source address presents itself bind with views) and by default docker pulls it's resolv.conf from the docker host itself.

But these are all somewhat merrier solutions, what i'm proposing is the addition of a secondary set of mqtt broker settings to be used just by esphome when defined.

nagyrobi, 2023-11-15:
See this comment on our Discord support server: https://discord.com/channels/429907082951524364/822162083452813312/1174361563713523763

MiggyMan, 2023-11-15:
Discord is a no for me, it's a chat system not a support mechanism, i'll figured something out.

closed 2023-11-15 by MiggyMan

nagyrobi, 2023-11-15:
The support mechanism is chosen by ESPHome team. If you don't want to take part of it, of course it's up to you, but keep in mind that you loose a lot of chance to get proper answers to your questions.