esphome/feature-requests #2483: Add support for Gree AC with 'YAY' remote

Brix1330, 2023-11-18:
**Describe the problem you have/What new integration you would like**

I would like to have Gree AC with YAY1F1 remote to be supported for IR Remote Climate.
None of these available models work.

- generic
-  yan
-  yaa
-  yac

**Please describe your use case for this integration and alternatives you've tried:**

I have also tried using HeatpumpIR platform with every available protocols but still not responding.
At the moment, I can control the AC with Tuya IR remote (manual button learning).

**Additional context**

[image]

marcinus66, 2023-11-26:
Same problem with YTF1 remote... anyone can help? Also tried with HeatpumpIR with no results.

pavleknezevic, 2023-12-06:
Also YAW1F remote.

636stefan, 2024-05-04:
I have the YAY1F1 as well. 
Few things I found out: 
1. esphome Integrations (Gree and IR-Heatpump) are not working 
2. [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) does support the remote and its working

revilo196, 2024-06-17:
arduino-heatpumpir added [greeyac and greeyt](https://github.com/ToniA/arduino-heatpumpir/commit/41e26721b08baa56d535a081ed5abfafc124bc3f) with version 1.0.25 and up. 

would it be possible to just update the dependencies?

nagyrobi, 2024-06-20:
@revilo196 try please https://github.com/esphome/esphome/pull/6948

nagyrobi, 2024-06-25:
Merged...

ppiwowar, 2024-07-07:
This also works with AlpicAir aircon system with YT1F remote and protocol: greeyan.
Questions:
1. How to make I-Feel button working ?
2. How to make ir-receiver mode working ? (reflect thermostat state in HA when original remote used)

nagyrobi, 2024-07-07:
These features are not entirely supported for Gree.

Einixx, 2024-10-26:
Hi, 
is there any solution yet? 
I also struggle to get the model yay1f1 working. I am using ESP32 (not ESP8266)  ESPHome 2024.9.0
Thanks

Brix1330, 2024-10-28:
> Hi, is there any solution yet? I also struggle to get the model yay1f1 working. I am using ESP32 (not ESP8266) ESPHome 2024.9.0 Thanks
> 
> I just tried all models and "yac1fb9" does the trick for YAY1F1

The yac1fb9 works but has limited features. The swing and fan modes don't work for me.

warehouseElc, 2026-08-17:
it has been 2y but no update at all ? got problem with gree & sharp product, please update it

indraperkasa, 2026-10-07:
just want to express got the same problem here. couldn't connect to my Gree AC with YAY1F1 remote. tried all the models available with no success. hope there's an update soon.