esphome/feature-requests #2485: Support for sensor: LTR-303

Ventilgummi, 2023-11-18:
**Describe the problem you have/What new integration you would like**
I would like the Adafruit Light Sensor "LTR-303" to be integrated to ESPHome. I suspect it would be similar to the integration that already exists named "LTR-390"

**Please describe your use case for this integration and alternatives you've tried:**
1. The integration does not exist with ESPHome
2. I've tried using Micropython, but there are no libraries for it.
3. I can't use CircuitPython, since ESP8266 is no longer supported

**Additional context**
N/A

Ventilgummi, 2023-11-19:
Perhaps possible to use this repository: https://github.com/adafruit/Adafruit_LTR329_LTR303
Instead of the current one being used for the LTR390? (https://github.com/adafruit/Adafruit_LTR390)

latonita, 2023-12-26:
@Ventilgummi do you still own this sensor and can test the component? If yes, I can try and make it

Ventilgummi, 2023-12-26:
Yes to both! I'll happily help with testing.

latonita, 2023-12-27:
please try https://github.com/latonita/esphome-ltr303 
you don't need to download code, use example config from there, it will download component itself

please send me results - all output from console

Ventilgummi, 2023-12-27:
I'm away for the Holidays, but will test during the week after new year!

Ventilgummi, 2024-01-05:
Hey, finally got time to test.

**Here are the full logs:**
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome-web-0e6e97.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing garageoppnare (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
RAM:   [====      ]  42.3% (used 34660 bytes from 81920 bytes)
Flash: [=====     ]  47.7% (used 498349 bytes from 1044464 bytes)
========================= [SUCCESS] Took 1.31 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.46
INFO Uploading /config/.esphome/build/garageoppnare/.pioenvs/garageoppnare/firmware.bin (502496 bytes)
INFO Compressed to 352114 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 5.74 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.46 using esphome API
INFO Successfully connected to garageoppnare @ 192.168.1.46 in 31.416s
INFO Successful handshake with garageoppnare @ 192.168.1.46 in 2.611s
[12:48:40][I][app:102]: ESPHome version 2023.12.5 compiled on Jan  5 2024, 12:38:37
[12:48:40][C][wifi:573]: WiFi:
[12:48:40][C][wifi:405]:   Local MAC: C8:C9:A3:0E:6E:97
[12:48:40][C][wifi:410]:   SSID: 'redacted'[redacted]
[12:48:40][C][wifi:411]:   IP Address: 192.168.1.46
[12:48:40][C][wifi:412]:   BSSID: redacted[redacted]
[12:48:40][C][wifi:414]:   Hostname: 'garageoppnare'
[12:48:40][C][wifi:416]:   Signal strength: -67 dB ▂▄▆█
[12:48:40][C][wifi:420]:   Channel: 11
[12:48:40][C][wifi:421]:   Subnet: 255.255.255.0
[12:48:40][C][wifi:422]:   Gateway: 192.168.1.1
[12:48:40][C][wifi:423]:   DNS1: 192.168.1.1
[12:48:40][C][wifi:424]:   DNS2: 0.0.0.0
[12:48:40][C][logger:443]: Logger:
[12:48:40][C][logger:444]:   Level: DEBUG
[12:48:40][C][logger:445]:   Log Baud Rate: 115200
[12:48:40][C][logger:447]:   Hardware UART: UART0
[12:48:40][C][i2c.arduino:053]: I2C Bus:
[12:48:40][C][i2c.arduino:054]:   SDA Pin: GPIO4
[12:48:40][C][i2c.arduino:055]:   SCL Pin: GPIO5
[12:48:40][C][i2c.arduino:056]:   Frequency: 50000 Hz
[12:48:40][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[12:48:40][I][i2c.arduino:069]: Results from i2c bus scan:
[12:48:40][I][i2c.arduino:075]: Found i2c device at address 0x29
[12:48:40][C][ltr303:218]:   Address: 0x29
[12:48:40][C][ltr303:219]:   Gain: x8
[12:48:40][C][ltr303:220]:   Integration time: 100 ms
[12:48:40][C][ltr303:221]:   Measurement repeat rate: 500 ms
[12:48:40][C][ltr303:222]:   Attenuation factor: 1.000000
[12:48:40][C][ltr303:223]:   Update Interval: 60.0s
[12:48:40][C][ltr303:226]:   CH1 Infrared 'Infrared'
[12:48:40][C][ltr303:226]:     Device Class: 'illuminance'
[12:48:40][C][ltr303:226]:     State Class: 'measurement'
[12:48:40][C][ltr303:226]:     Unit of Measurement: 'counts'
[12:48:40][C][ltr303:226]:     Accuracy Decimals: 0
[12:48:40][C][ltr303:226]:     Icon: 'mdi:brightness-6'
[12:48:40][C][ltr303:228]:   CH0 Visible + IR 'Visible and Infrared'
[... 65 lines cut ...]

external_components:
  - source: github://latonita/esphome-ltr303
    components: [ ltr303 ]

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

sensor:
  - platform: ltr303
    address: 0x29
    gain: X8
    integration_time: 100ms
    repeat: 500ms
    glass_attenuation_factor: 1.0
    infrared:
      name: "Infrared"
    full_spectrum:
      name: "Visible and Infrared"
    actual_gain:
      name: "Actual gain"
    calculated_lux:
      name: "Calculated lux"

Ventilgummi, 2024-01-05:
Also, I don't know how to copy paste logs in a fancy way... Feel free to let me know how to do that :)

latonita, 2024-01-05:
Please retry, i've made minor update.
As for logs - you can put them in code block. Put use triple backticks on one line, then insert log, then again use triple backticks.
Check here https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

Ventilgummi, 2024-01-05:
Had to use another board (ESPHUZZAH8266 this time). I just built a relay with the other one... 

Testing inserting code!
Log ouput:
```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome-web-9034c9.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphuzzah (board: huzzah; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
Compiling .pioenvs/esphuzzah/src/main.cpp.o
Linking .pioenvs/esphuzzah/firmware.elf
RAM:   [====      ]  42.3% (used 34644 bytes from 81920 bytes)
Flash: [=====     ]  47.7% (used 498413 bytes from 1044464 bytes)
Building .pioenvs/esphuzzah/firmware.bin
esp8266_copy_factory_bin([".pioenvs/esphuzzah/firmware.bin"], [".pioenvs/esphuzzah/firmware.elf"])
========================= [SUCCESS] Took 3.88 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.210
INFO Uploading /config/.esphome/build/esphuzzah/.pioenvs/esphuzzah/firmware.bin (502560 bytes)
INFO Compressed to 352133 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 3.81 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.210 using esphome API
INFO Successfully connected to esphuzzah @ 192.168.1.210 in 15.286s
INFO Successful handshake with esphuzzah @ 192.168.1.210 in 2.865s
[16:26:40][I][app:102]: ESPHome version 2023.12.5 compiled on Jan  5 2024, 16:26:13
[16:26:40][C][wifi:573]: WiFi:
[16:26:40][C][wifi:405]:   Local MAC: C8:C9:A3:90:34:C9
[16:26:40][C][wifi:410]:   SSID: 'redacted'[redacted]
[16:26:40][C][wifi:411]:   IP Address: 192.168.1.210
[16:26:40][C][wifi:412]:   BSSID: redacted[redacted]
[16:26:40][C][wifi:414]:   Hostname: 'esphuzzah'
[16:26:40][C][wifi:416]:   Signal strength: -63 dB ▂▄▆█
[16:26:40][C][wifi:420]:   Channel: 11
[16:26:40][C][wifi:421]:   Subnet: 255.255.255.0
[16:26:40][C][wifi:422]:   Gateway: 192.168.1.1
[16:26:40][C][wifi:423]:   DNS1: 0.0.0.0
[16:26:40][C][wifi:424]:   DNS2: 0.0.0.0
[16:26:40][C][logger:443]: Logger:
[16:26:40][C][logger:444]:   Level: DEBUG
[16:26:40][C][logger:445]:   Log Baud Rate: 115200
[16:26:40][C][logger:447]:   Hardware UART: UART0
[16:26:40][C][i2c.arduino:053]: I2C Bus:
[16:26:40][C][i2c.arduino:054]:   SDA Pin: GPIO4
[16:26:40][C][i2c.arduino:055]:   SCL Pin: GPIO5
[16:26:40][C][i2c.arduino:056]:   Frequency: 800000 Hz
[16:26:40][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[16:26:40][I][i2c.arduino:069]: Results from i2c bus scan:
[16:26:40][I][i2c.arduino:075]: Found i2c device at address 0x29
[16:26:40][C][ltr303:218]:   Address: 0x29
[16:26:40][C][ltr303:219]:   Gain: x8
[16:26:40][C][ltr303:220]:   Integration time: 100 ms
[16:26:40][C][ltr303:221]:   Measurement repeat rate: 500 ms
[16:26:40][C][ltr303:222]:   Attenuation factor: 1.000000
[16:26:40][C][ltr303:223]:   Update Interval: 60.0s
[16:26:40][C][ltr303:226]:   CH1 Infrared 'Infrared'
[... 74 lines cut ...]
  - id: bus_a
    sda: GPIO4
    scl: GPIO5
    scan: true
    frequency: 800kHz
external_components:
  - source: github://latonita/esphome-ltr303
    components: [ ltr303 ]

sensor:
  - platform: ltr303
    address: 0x29
    gain: X8
    integration_time: 100ms
    repeat: 500ms
    glass_attenuation_factor: 1.0
    infrared:
      name: "Infrared"
    full_spectrum:
      name: "Visible and Infrared"
    actual_gain:
      name: "Actual gain"
    calculated_lux:
      name: "Calculated lux"
```

latonita, 2024-01-05:
@Ventilgummi thanks. do you receive readings?
please use config which enables debug log with 
```
logger:
  level: DEBUG
```

Ventilgummi, 2024-01-05:
Sorry about that!
Here's the new log output with logger at DEBUG:

```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome-web-9034c9.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphuzzah (board: huzzah; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
Compiling .pioenvs/esphuzzah/src/main.cpp.o
Linking .pioenvs/esphuzzah/firmware.elf
RAM:   [====      ]  42.3% (used 34644 bytes from 81920 bytes)
Flash: [=====     ]  47.7% (used 498413 bytes from 1044464 bytes)
Building .pioenvs/esphuzzah/firmware.bin
esp8266_copy_factory_bin([".pioenvs/esphuzzah/firmware.bin"], [".pioenvs/esphuzzah/firmware.elf"])
========================= [SUCCESS] Took 3.93 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.210
INFO Uploading /config/.esphome/build/esphuzzah/.pioenvs/esphuzzah/firmware.bin (502560 bytes)
INFO Compressed to 352132 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 4.44 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.210 using esphome API
INFO Successfully connected to esphuzzah @ 192.168.1.210 in 15.261s
INFO Successful handshake with esphuzzah @ 192.168.1.210 in 2.882s
[16:34:09][I][app:102]: ESPHome version 2023.12.5 compiled on Jan  5 2024, 16:33:41
[16:34:09][C][wifi:573]: WiFi:
[16:34:09][C][wifi:405]:   Local MAC: C8:C9:A3:90:34:C9
[16:34:09][C][wifi:410]:   SSID: [redacted]
[16:34:09][C][wifi:411]:   IP Address: 192.168.1.210
[16:34:09][C][wifi:412]:   BSSID: [redacted]
[16:34:09][C][wifi:414]:   Hostname: 'esphuzzah'
[16:34:09][C][wifi:416]:   Signal strength: -72 dB ▂▄▆█
[16:34:09][C][wifi:420]:   Channel: 11
[16:34:09][C][wifi:421]:   Subnet: 255.255.255.0
[16:34:09][C][wifi:422]:   Gateway: 192.168.1.1
[16:34:09][C][wifi:423]:   DNS1: 0.0.0.0
[16:34:09][C][wifi:424]:   DNS2: 0.0.0.0
[16:34:09][C][logger:443]: Logger:
[16:34:09][C][logger:444]:   Level: DEBUG
[16:34:09][C][logger:445]:   Log Baud Rate: 115200
[16:34:09][C][logger:447]:   Hardware UART: UART0
[16:34:09][C][i2c.arduino:053]: I2C Bus:
[16:34:09][C][i2c.arduino:054]:   SDA Pin: GPIO4
[16:34:09][C][i2c.arduino:055]:   SCL Pin: GPIO5
[16:34:09][C][i2c.arduino:056]:   Frequency: 800000 Hz
[16:34:09][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[16:34:09][I][i2c.arduino:069]: Results from i2c bus scan:
[16:34:09][I][i2c.arduino:075]: Found i2c device at address 0x29
[16:34:09][C][ltr303:218]:   Address: 0x29
[16:34:09][C][ltr303:219]:   Gain: x8
[16:34:09][C][ltr303:220]:   Integration time: 100 ms
[16:34:09][C][ltr303:221]:   Measurement repeat rate: 500 ms
[16:34:09][C][ltr303:222]:   Attenuation factor: 1.000000
[16:34:09][C][ltr303:223]:   Update Interval: 60.0s
[16:34:09][C][ltr303:226]:   CH1 Infrared 'Infrared'
[16:34:09][C][ltr303:226]:     Device Class: 'illuminance'
[16:34:09][C][ltr303:226]:     State Class: 'measurement'
[16:34:09][C][ltr303:226]:     Unit of Measurement: 'counts'
[16:34:09][C][ltr303:226]:     Accuracy Decimals: 0
[16:34:09][C][ltr303:226]:     Icon: 'mdi:brightness-6'
[16:34:09][C][ltr303:228]:   CH0 Visible + IR 'Visible and Infrared'
[16:34:09][C][ltr303:228]:     Device Class: 'illuminance'
[16:34:09][C][ltr303:228]:     State Class: 'measurement'
[16:34:09][C][ltr303:228]:     Unit of Measurement: 'counts'
[16:34:09][C][ltr303:228]:     Accuracy Decimals: 0
[16:34:09][C][ltr303:228]:     Icon: 'mdi:brightness-6'
[16:34:09][C][ltr303:230]:   Actrual gain 'Actual gain'
[16:34:09][C][ltr303:230]:     Device Class: 'illuminance'
[16:34:09][C][ltr303:230]:     State Class: 'measurement'
[16:34:09][C][ltr303:230]:     Unit of Measurement: ''
[16:34:09][C][ltr303:230]:     Accuracy Decimals: 0
[16:34:09][C][ltr303:230]:     Icon: 'mdi:brightness-6'
[16:34:09][C][ltr303:232]:   CH0 Visible + IR 'Calculated lux'
[16:34:09][C][ltr303:232]:     Device Class: 'illuminance'
[16:34:09][C][ltr303:232]:     State Class: 'measurement'
[16:34:09][C][ltr303:232]:     Unit of Measurement: 'lx'
[16:34:09][C][ltr303:232]:     Accuracy Decimals: 1
[16:34:09][C][ltr303:232]:     Icon: 'mdi:brightness-6'
[16:34:09][C][captive_portal:088]: Captive Portal:
[16:34:09][C][mdns:115]: mDNS:
[16:34:09][C][mdns:116]:   Hostname: esphuzzah
[16:34:09][C][ota:097]: Over-The-Air Updates:
[16:34:09][C][ota:098]:   Address: 192.168.1.210:8266
[16:34:09][C][api:139]: API Server:
[16:34:09][C][api:140]:   Address: 192.168.1.210:6053
[16:34:09][C][api:142]:   Using noise encryption: YES
[16:34:12][D][api:102]: Accepted 192.168.1.125
[16:34:12][W][component:214]: Component api took a long time for an operation (0.29 s).
[16:34:12][W][component:215]: Components should block for at most 20-30ms.
[16:34:13][D][api.connection:1121]: Home Assistant 2024.1.0 (192.168.1.125): Connected successfully
[16:34:46][D][ltr303:241]: Updating
[16:34:46][D][ltr303:083]: Reading sensor data
[16:34:46][D][ltr303:090]: LTR303_ALS_STATUS = 0x34
[16:34:46][D][ltr303:093]: New data is ready
[16:34:46][W][ltr303:105]: Data available but not valid
```

Ventilgummi, 2024-01-05:
Ah, and no readings, sadly. Only "Data available but not valid"

latonita, 2024-01-05:
@Ventilgummi it looks like old version got cached  - please delete temp files in .esphome/build and .esphome/external_components
I have uploaded new one to github after checking your first log file.

Ventilgummi, 2024-01-05:
Nice job!
I cleared the cache.
New log output, with data:

```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome-web-9034c9.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphuzzah (board: huzzah; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
Compiling .pioenvs/esphuzzah/src/esphome/components/ltr303/ltr303.cpp.o
Compiling .pioenvs/esphuzzah/src/main.cpp.o
Linking .pioenvs/esphuzzah/firmware.elf
RAM:   [====      ]  42.3% (used 34644 bytes from 81920 bytes)
Flash: [=====     ]  47.7% (used 498505 bytes from 1044464 bytes)
Building .pioenvs/esphuzzah/firmware.bin
esp8266_copy_factory_bin([".pioenvs/esphuzzah/firmware.bin"], [".pioenvs/esphuzzah/firmware.elf"])
========================= [SUCCESS] Took 3.95 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.210
INFO Uploading /config/.esphome/build/esphuzzah/.pioenvs/esphuzzah/firmware.bin (502656 bytes)
INFO Compressed to 352201 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 4.49 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.210 using esphome API
INFO Successfully connected to esphuzzah @ 192.168.1.210 in 15.416s
INFO Successful handshake with esphuzzah @ 192.168.1.210 in 2.869s
[16:46:16][I][app:102]: ESPHome version 2023.12.5 compiled on Jan  5 2024, 16:45:48
[16:46:16][C][wifi:573]: WiFi:
[16:46:16][C][wifi:405]:   Local MAC: C8:C9:A3:90:34:C9
[16:46:16][C][wifi:410]:   SSID: 'CZ-Wifi'[redacted]
[16:46:16][C][wifi:411]:   IP Address: 192.168.1.210
[16:46:16][C][wifi:412]:   BSSID: 78:45:58:C5:AC:80[redacted]
[16:46:16][C][wifi:414]:   Hostname: 'esphuzzah'
[16:46:16][C][wifi:416]:   Signal strength: -61 dB ▂▄▆█
[16:46:16][C][wifi:420]:   Channel: 11
[16:46:16][C][wifi:421]:   Subnet: 255.255.255.0
[16:46:16][C][wifi:422]:   Gateway: 192.168.1.1
[16:46:16][C][wifi:423]:   DNS1: 0.0.0.0
[16:46:16][C][wifi:424]:   DNS2: 0.0.0.0
[16:46:16][C][logger:443]: Logger:
[16:46:16][C][logger:444]:   Level: DEBUG
[16:46:16][C][logger:445]:   Log Baud Rate: 115200
[16:46:16][C][logger:447]:   Hardware UART: UART0
[16:46:16][C][i2c.arduino:053]: I2C Bus:
[16:46:16][C][i2c.arduino:054]:   SDA Pin: GPIO4
[16:46:16][C][i2c.arduino:055]:   SCL Pin: GPIO5
[16:46:16][C][i2c.arduino:056]:   Frequency: 800000 Hz
[16:46:16][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[16:46:16][I][i2c.arduino:069]: Results from i2c bus scan:
[16:46:16][I][i2c.arduino:075]: Found i2c device at address 0x29
[16:46:16][C][ltr303:223]:   Address: 0x29
[16:46:16][C][ltr303:224]:   Gain: x8
[16:46:16][C][ltr303:225]:   Integration time: 100 ms
[16:46:16][C][ltr303:226]:   Measurement repeat rate: 500 ms
[16:46:16][C][ltr303:227]:   Attenuation factor: 1.000000
[16:46:16][C][ltr303:228]:   Update Interval: 60.0s
[... 25 lines cut ...]
[16:46:16][C][mdns:115]: mDNS:
[16:46:16][C][mdns:116]:   Hostname: esphuzzah
[16:46:16][C][ota:097]: Over-The-Air Updates:
[16:46:16][C][ota:098]:   Address: 192.168.1.210:8266
[16:46:16][C][api:139]: API Server:
[16:46:16][C][api:140]:   Address: 192.168.1.210:6053
[16:46:16][C][api:142]:   Using noise encryption: YES
[16:46:18][D][api:102]: Accepted 192.168.1.125
[16:46:18][W][component:214]: Component api took a long time for an operation (0.29 s).
[16:46:18][W][component:215]: Components should block for at most 20-30ms.
[16:46:19][D][api.connection:1121]: Home Assistant 2024.1.0 (192.168.1.125): Connected successfully
[16:46:40][D][ltr303:246]: Updating
[16:46:40][D][ltr303:083]: Reading sensor data
[16:46:40][D][ltr303:090]: LTR303_ALS_STATUS = 0x34
[16:46:40][D][ltr303:093]: New data is ready
[16:46:40][D][ltr303:119]: Channel data CH1 = 381, CH0 = 380, diff = 0
[16:46:40][D][sensor:093]: 'Infrared': Sending state 381.00000 # with 0 decimals of accuracy
[16:46:40][D][sensor:093]: 'Visible and Infrared': Sending state 380.00000 # with 0 decimals of accuracy
[16:46:40][D][sensor:093]: 'Actual gain': Sending state 8.00000  with 0 decimals of accuracy
[16:46:40][D][ltr303:067]: CALC LUX ratio 0.500657, gain 8.000000, int 1.000000, pfactor 1.000000
[16:46:40][D][ltr303:144]: Calculated lux: 110.131401
[16:46:40][D][sensor:093]: 'Calculated lux': Sending state 110.13140 lx with 1 decimals of accuracy
[16:46:40][W][component:214]: Component ltr303.sensor took a long time for an operation (0.06 s).
[16:46:40][W][component:215]: Components should block for at most 20-30ms.
```

latonita, 2024-01-05:
Great to hear. Test it a little and if it works okay I will create pull request with new component for the esphome main repository.

Ventilgummi, 2024-01-05:
I dont seem to be able to adjust the update frequence with :

```
sensor:
  - platform: ltr303
    address: 0x29
    gain: X8
    integration_time: 100ms
    repeat: 500ms
    glass_attenuation_factor: 1.0
    infrared:
      name: "Infrared"
    full_spectrum:
      name: "Visible and Infrared"
    actual_gain:
      name: "Actual gain"
    calculated_lux:
      name: "Calculated lux"
    update_interval: 30s
```

However I'm not the best at this, so I might be doing something wrong.
It remains at one minute interval

latonita, 2024-01-11:
> I dont seem to be able to adjust the update frequence with :
> 
> ```
>     update_interval: 30s
> ```
> 
`update_internal` shall work properly. looks like previous build files got cached. 

I made Pull Request. You may want to check if it works :) config a little bit reworked. And source code URL is different.
```

esphome:
  name: ltr303-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# esp8266:
#   board: nodemcuv2

logger:
  level: DEBUG

external_components:
   - source: github://pr#6067
     components: [ltr303]
#  - source: 
#      type: local
#      path: components

i2c:
  sda: GPIO25
  scl: GPIO32
  scan: true
  frequency: 5kHz

sensor:
  - platform: ltr303
    address: 0x29
    gain: 1x
    integration_time: 100ms
    ambient_light:
      name: "Ambient light"
# Following sensors are not really of a lot of use:
#    full_spectrum_counts:
#      name: "Full spectrum counts"
#    infrared_counts:
#      name: "Infrared counts"
#    actual_gain:
#      name: "Actual gain"
```

I might implement 'automatic gain' , but I don`t have a real device to test. Idea (and implementation) is very simple: use X1 gain first, then if counts are low go down gradually when counts somewhere in-between min and max.

Ventilgummi, 2024-01-20:
Hi!
Sorry for the extremely slow respons time....

I ran it just now with the new code - seems to be working!

Config:
```
esphome:
  name: esphuzzah
  friendly_name: ESP8266
  #on_boot:

esp8266:
  board: huzzah

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "[redacted]"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  manual_ip:
    static_ip: 192.168.1.210
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  ap:
    ssid: "Esphome-Web-9034C9"
    password: "---"

captive_portal:

i2c:
  - id: bus_a
    sda: GPIO4
    scl: GPIO5
    scan: true
    frequency: 400kHz

external_components:
   - source: github://latonita/esphome-ltr303
     components: [ltr303]
#  - source: 
#      type: local
#      path: components

sensor:
  - platform: ltr303
    address: 0x29
# auto mode - if true, then gain and integration time are not used
    auto_mode: true
    gain: 1x
    integration_time: 100ms
    ambient_light:
      name: "Ambient light"
# Following sensors are not really of a lot of use:
#    full_spectrum_counts:
#      name: "Full spectrum counts"
#    infrared_counts:
#      name: "Infrared counts"
    actual_gain:
      name: "Actual gain"
[... 119 lines cut ...]
[11:55:49][D][ltr303:076]: Auto - gain = 2x, time = 400 ms
[11:55:50][W][ltr303:164]: Waiting for data
[11:55:50][D][ltr303:146]: Got sensor data: CH1 = 588, CH0 = 503
[11:55:50][D][ltr303:105]: Low illuminance. Increasing gain
[11:55:50][D][ltr303:076]: Auto - gain = 4x, time = 400 ms
[11:55:50][W][ltr303:164]: Waiting for data
[11:55:50][W][ltr303:164]: Waiting for data
[11:55:50][D][ltr303:146]: Got sensor data: CH1 = 1182, CH0 = 978
[11:55:50][D][ltr303:105]: Low illuminance. Increasing gain
[11:55:50][D][ltr303:076]: Auto - gain = 8x, time = 400 ms
[11:55:51][W][ltr303:164]: Waiting for data
[11:55:51][D][ltr303:146]: Got sensor data: CH1 = 2383, CH0 = 1976
[11:55:51][D][ltr303:105]: Low illuminance. Increasing gain
[11:55:51][D][ltr303:076]: Auto - gain = 48x, time = 400 ms
[11:55:51][W][ltr303:164]: Waiting for data
[11:55:51][D][ltr303:146]: Got sensor data: CH1 = 14984, CH0 = 12337
[11:55:51][D][ltr303:124]: Illiminance is good enough
[11:55:51][D][sensor:093]: 'Actual gain': Sending state 48.00000  with 0 decimals of accuracy
[11:55:51][D][sensor:093]: 'Actual integration time': Sending state 400.00000 ms with 0 decimals of accuracy
[11:55:51][D][ltr303:033]: Lux calculation: ratio 0.548443, gain 48.000000, int time 4.000000, inv_pfactor 1.000000
[11:55:51][D][ltr303:207]: Calculated ALS: 122.360054 lx
[11:55:51][D][sensor:093]: 'Ambient light': Sending state 122.36005 lx with 1 decimals of accuracy
[11:55:51][W][component:214]: Component ltr303.sensor took a long time for an operation (6.54 s).
[11:55:51][W][component:215]: Components should block for at most 20-30ms.
```

latonita, 2024-01-20:
@Ventilgummi thanks a lot.
I would appreciate If you can test new version. I had to rework it a lot after esphome team review.
thanks!

abluesheep, 2024-02-26:
@latonita , thanks for adding support for LTR303

I tried to use it, but the values appear to periodically (every 5 mins) go super high (like 5k lux) before coming back to a "normal" reading. Please see the sensor reading chart below to illustrate the issue:

[image]

Config:
```
esphome:
  name: esphome-web-18b9f8
  friendly_name: Mini2

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

#section for LTR303 beta support
external_components:
  - source: github://latonita/esphome-ltr303
    components: [ ltr303 ]

# Enable Home Assistant API
api:
  encryption:
    key: [removed]

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-18B9F8"
    password: [removed]

captive_portal:

# I2C Bus
i2c:
  sda: 4
  scl: 5
  scan: true
  id: bus_a
  frequency: 100kHz

# Sensors
sensor:
  - platform: ltr303
    address: 0x29
    auto_mode: true
# gain and time ignored in auto mode
    gain: 1x
    integration_time: 100ms
    glass_attenuation_factor: 1.0
    ambient_light:
      name: "Ambient light"
# Following sensors are not really of a lot of use, to be honest :)
    full_spectrum_counts:
      name: "Full spectrum counts"
    infrared_counts:
      name: "Infrared counts"
    actual_gain:
      name: "Actual gain"
[... 623 lines cut ...]
[18:01:20][D][ltr303:127]: Reading sensor data having gain = 48x, time = 100 ms
[18:01:20][D][ltr303:256]: Got sensor data: CH1 = 1573, CH0 = 1900
[18:01:20][D][ltr303:300]: Illuminance is good enough.
[18:01:20][D][ltr303:329]: Lux calculation: ratio 0.452923, gain 48.000000, int time 1.000000, inv_pfactor 1.000000
[18:01:20][D][sensor:094]: 'Ambient light': Sending state 105.29687 lx with 1 decimals of accuracy
[18:01:20][D][sensor:094]: 'Infrared counts': Sending state 1573.00000 # with 0 decimals of accuracy
[18:01:20][D][sensor:094]: 'Full spectrum counts': Sending state 1900.00000 # with 0 decimals of accuracy
[18:01:20][D][sensor:094]: 'Actual gain': Sending state 48.00000  with 0 decimals of accuracy
[18:01:20][D][sensor:094]: 'Actual integration time': Sending state 100.00000 ms with 0 decimals of accuracy
[18:02:17][D][ltr303:086]: Updating
[18:02:17][D][ltr303:088]: Initiating new data collection
[18:02:17][D][ltr303:145]: Reconfiguring sensitivity
[18:02:18][D][ltr303:244]: Data ready, reported gain is 48
[18:02:18][D][ltr303:127]: Reading sensor data having gain = 1x, time = 100 ms
[18:02:18][D][ltr303:256]: Got sensor data: CH1 = 1576, CH0 = 1910
[18:02:18][D][ltr303:300]: Illuminance is good enough.
[18:02:18][D][ltr303:329]: Lux calculation: ratio 0.452094, gain 1.000000, int time 1.000000, inv_pfactor 1.000000
[18:02:18][D][sensor:094]: 'Ambient light': Sending state 5091.17041 lx with 1 decimals of accuracy
[18:02:18][D][sensor:094]: 'Infrared counts': Sending state 1576.00000 # with 0 decimals of accuracy
[18:02:18][D][sensor:094]: 'Full spectrum counts': Sending state 1910.00000 # with 0 decimals of accuracy
[18:02:18][D][sensor:094]: 'Actual gain': Sending state 1.00000  with 0 decimals of accuracy
[18:02:18][D][sensor:094]: 'Actual integration time': Sending state 100.00000 ms with 0 decimals of accuracy
```

Thanks again and let me know what I can do to help.

latonita, 2024-02-26:
@abluesheep thanks for the report! log is useful, i will take a look. For unknown reason sensor didn't accept new gain setting and reported data using old one.
I have updated repo and uploaded latest code from PR.
please try.

note, the component name is different

abluesheep, 2024-02-27:
Thanks for the quick edits @latonita

Unfortunately the problem seems to be still occurring, as per the chart below.
[image]

Config:
```
esphome:
  name: esphome-web-18b9f8
  friendly_name: Mini2

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

#section for LTR303 beta support
external_components:
  - source: github://latonita/esphome-ltr303
    components: [ ltr_als_ps ]
    refresh: 0s

# Enable Home Assistant API
api:
  encryption:
    key: [removed]

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-18B9F8"
    password: [removed]

captive_portal:

# I2C Bus
i2c:
  sda: 4
  scl: 5
  scan: true
  id: bus_a
  frequency: 100kHz

# Sensors
sensor:
  - platform: ltr_als_ps
    address: 0x29
    auto_mode: true
# gain and time ignored in auto mode
    gain: 1x
    integration_time: 100ms
    glass_attenuation_factor: 1.0
    ambient_light:
      name: "Ambient light"
# Following sensors are not really of a lot of use, to be honest :)
    full_spectrum_counts:
      name: "Full spectrum counts"
    infrared_counts:
      name: "Infrared counts"
    actual_gain:
      name: "Actual gain"
[... 931 lines cut ...]
[12:22:04][D][ltr_als_ps:159]: Reading sensor data having gain = 48x, time = 100 ms
[12:22:04][D][ltr_als_ps:371]: Got sensor data: CH1 = 1087, CH0 = 1171
[12:22:04][D][ltr_als_ps:415]: Illuminance is good enough.
[12:22:04][D][ltr_als_ps:444]: Lux calculation: ratio 0.481399, gain 48.000000, int time 1.000000, inv_pfactor 1.000000
[12:22:04][D][sensor:094]: 'Ambient light': Sending state 60.10950 lx with 1 decimals of accuracy
[12:22:04][D][sensor:094]: 'Infrared counts': Sending state 1087.00000 # with 0 decimals of accuracy
[12:22:04][D][sensor:094]: 'Full spectrum counts': Sending state 1171.00000 # with 0 decimals of accuracy
[12:22:04][D][sensor:094]: 'Actual gain': Sending state 48.00000  with 0 decimals of accuracy
[12:22:04][D][sensor:094]: 'Actual integration time': Sending state 100.00000 ms with 0 decimals of accuracy
[12:23:02][D][ltr_als_ps:110]: Updating
[12:23:02][D][ltr_als_ps:112]: Initiating new data collection
[12:23:02][D][ltr_als_ps:177]: Reconfiguring sensitivity
[12:23:02][D][ltr_als_ps:359]: Data ready, reported gain is 48
[12:23:02][D][ltr_als_ps:159]: Reading sensor data having gain = 1x, time = 100 ms
[12:23:02][D][ltr_als_ps:371]: Got sensor data: CH1 = 1087, CH0 = 1174
[12:23:02][D][ltr_als_ps:415]: Illuminance is good enough.
[12:23:02][D][ltr_als_ps:444]: Lux calculation: ratio 0.480761, gain 1.000000, int time 1.000000, inv_pfactor 1.000000
[12:23:02][D][sensor:094]: 'Ambient light': Sending state 2898.09131 lx with 1 decimals of accuracy
[12:23:02][D][sensor:094]: 'Infrared counts': Sending state 1087.00000 # with 0 decimals of accuracy
[12:23:02][D][sensor:094]: 'Full spectrum counts': Sending state 1174.00000 # with 0 decimals of accuracy
[12:23:02][D][sensor:094]: 'Actual gain': Sending state 1.00000  with 0 decimals of accuracy
[12:23:02][D][sensor:094]: 'Actual integration time': Sending state 100.00000 ms with 0 decimals of accuracy
```

Thanks so much!

latonita, 2024-02-27:
Please check new version and report the log. Some double checks implemented.

From: abluesheep ***@***.***>
Sent: Tuesday, February 27, 2024 7:59 AM
To: esphome/feature-requests ***@***.***>
Cc: Anton Viktorov ***@***.***>; Mention ***@***.***>
Subject: Re: [esphome/feature-requests] Support for sensor: LTR-303 (Issue #2485)

Thanks for the quick edits @latonita<https://github.com/latonita>

Unfortunately the problem seems to be still occurring, as per the chart below.
Screenshot.2024-02-27.at.2.53.13.PM.png (view on web)<https://github.com/esphome/feature-requests/assets/1185630/c4953796-4f27-4122-b336-20c4bed4e576>

Config:

esphome:

  name: esphome-web-18b9f8

  friendly_name: Mini2

esp32:

  board: esp32-c3-devkitm-1

  framework:

    type: arduino

# Enable logging

logger:

  level: DEBUG

#section for LTR303 beta support

external_components:

  - source: github://latonita/esphome-ltr303

    components: [ ltr_als_ps ]

    refresh: 0s

# Enable Home Assistant API

api:

  encryption:

    key: [removed]

ota:

wifi:

  ssid: !secret wifi_ssid

  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails

  ap:

    ssid: "Esphome-Web-18B9F8"

    password: [removed]

captive_portal:
[... 1757 lines cut ...]
[12:20:03][D][ltr_als_ps:392]: Low illuminance. Increasing gain.

[12:20:03][D][ltr_als_ps:177]: Reconfiguring sensitivity

[12:20:04][D][ltr_als_ps:359]: Data ready, reported gain is 8

[12:20:04][D][ltr_als_ps:159]: Reading sensor data having gain = 8x, time = 100 ms

[12:20:04][D][ltr_als_ps:371]: Got sensor data: CH1 = 202, CH0 = 246

[12:20:04][D][ltr_als_ps:392]: Low illuminance. Increasing gain.

[12:20:04][D][ltr_als_ps:177]: Reconfiguring sensitivity

[12:20:04][D][ltr_als_ps:359]: Data ready, reported gain is 48

[12:20:04][D][ltr_als_ps:159]: Reading sensor data having gain = 48x, time = 100 ms

[12:20:04][D][ltr_als_ps:371]: Got sensor data: CH1 = 1330, CH0 = 1585

[12:20:04][D][ltr_als_ps:415]: Illuminance is good enough.

[12:20:04][D][ltr_als_ps:444]: Lux calculation: ratio 0.456261, gain 48.000000, int time 1.000000, inv_pfactor 1.000000

[12:

abluesheep, 2024-02-27:
Thanks @latonita , looks like it's working well now! I've attached the logs in case you need them for anything.

Cheers!
[logs_esphome-web-18b9f8_run (1).txt](https://github.com/esphome/feature-requests/files/14416733/logs_esphome-web-18b9f8_run.1.txt)

klaasjanhorlings, 2024-03-01:
I'm trying to run the latest version of your code on my own board and it gets stuck reconfiguring the gain while already at full gain. It does eventually report values, but there still seems to be something wrong with the code waiting for the gain configuration to complete.

Relevant config:
```
external_components:
  - source: github://latonita/esphome-ltr303
    components: [ ltr_als_ps ]
    refresh: 1h

sensor:
  - platform: ltr_als_ps
    address: 0x29
    auto_mode: true
    type: ALS
    ambient_light:
      name: "Ambient light"
    full_spectrum_counts:
      name: "Full spectrum counts"
    infrared_counts:
      name: "Infrared counts"
```

And logs:
```
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 1x, time = 100 ms
[W][ltr_als_ps:463]: Sensors blacked out
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 2x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 4x, time = 100 ms
[W][ltr_als_ps:483]: Impossible ch1/(ch0 + ch1) ratio
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 8x, time = 100 ms
[W][ltr_als_ps:483]: Impossible ch1/(ch0 + ch1) ratio
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 48x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 150 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 200 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 250 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 300 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 350 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 400 ms
[W][ltr_als_ps:408]: Too many sensitivity adjustments done. Apparently, sensor reconfiguration fails. Stopping.
[D][sensor:094]: 'Ambient light': Sending state 0.37259 lx with 1 decimals of accuracy
[D][sensor:094]: 'Infrared counts': Sending state 45.00000 # with 0 decimals of accuracy
[D][sensor:094]: 'Full spectrum counts': Sending state 54.00000 # with 0 decimals of accuracy
```

latonita, 2024-03-01:
Please set logging to verbose and share log file.
logger:
   verbose

BR, Anton

klaasjanhorlings, 2024-03-01:
Start:
```
[C][ltr_als_ps:085]:   Address: 0x29
[C][ltr_als_ps:086]:   Device type: ALS only
[C][ltr_als_ps:087]:   Automatic mode: ON
[C][ltr_als_ps:088]:   Gain: 1x
[C][ltr_als_ps:089]:   Integration time: 100 ms
[C][ltr_als_ps:090]:   Measurement repeat rate: 500 ms
[C][ltr_als_ps:091]:   Glass attenuation factor: 1.000000
[C][ltr_als_ps:092]:   Proximity gain: 16x
[C][ltr_als_ps:093]:   Proximity cooldown time: 5 s
[C][ltr_als_ps:094]:   Proximity high threshold: 65535
[C][ltr_als_ps:095]:   Proximity low threshold: 0
[C][ltr_als_ps:097]:   Update Interval: 60.0s
[C][ltr_als_ps:099]:   ALS calculated lux 'Ambient light'
[C][ltr_als_ps:099]:     Device Class: 'illuminance'
[C][ltr_als_ps:099]:     State Class: 'measurement'
[C][ltr_als_ps:099]:     Unit of Measurement: 'lx'
[C][ltr_als_ps:099]:     Accuracy Decimals: 1
[C][
[C][ltr_als_ps:100]:   CH1 Infrared counts 'Infrared counts'
[C][ltr_als_ps:100]:     Device Class: 'illuminance'
[C][ltr_als_ps:100]:     State Class: 'measurement'
[C][ltr_als_ps:100]:     Unit of Measurement: '#'
[C][ltr_als_ps:100]:     Accuracy Decimals: 00;35m[C][ltr_als_ps:100]:     Icon: 'mdi:brightness-5'
[C][ltr_als_ps:101]:   CH0 Visible+IR counts 'Full spectrum counts'
[C][ltr_als_ps:101]:     Device Class: 'illuminance'
[C][ltr_als_ps:101]:     State Class: 'measurement'
[C][ltr_als_ps:101]:     Unit of Measureme[C][ltr_als_ps:101]:     Accuracy Decimals: 0
[C][ltr_als_ps:101]:     Icon: 'mdi:brightness-7'
```

It does work when there is enough light:
```
[D][api.connection:1121]: Home Assistant 2024.2.5 (192.168.0.234): Connected successfully
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 1x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 2x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 4x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 8x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 48x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 100 ms
[D][ltr_als_ps:448]: Illuminance is good enough.
[D][sensor:094]: 'Ambient light': Sending state 32.74389 lx with 1 decimals of accuracy
[D][sensor:094]: 'Infrared counts': Sending state 2533.00000 # with 0 decimals of accuracy
[D][sensor:094]: 'Full spectrum counts': Sending state 1892.00000 # with 0 decimals of accuracy
```

But retries 96 gain with less light:
```
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 1x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 2x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 4x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 8x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 48x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 150 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 200 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 250 ms
[D][ltr_als_ps:448]: Illuminance is good enough.
[D][sensor:094]: 'Ambient light': Sending state 3.86947 lx with 1 decimals of accuracy
[D][sensor:094]: 'Infrared counts': Sending state 2516.00000 # with 0 decimals of accuracy
[D][sensor:094]: 'Full spectrum counts': Sending state 1064.00000 # with 0 decimals of accuracy
```

And fails with hardly any light (does seem to be related to the previous measurement:
```
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 1x, time = 100 ms
[W][ltr_als_ps:483]: Impossible ch1/(ch0 + ch1) ratio
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 2x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 4x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 8x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 48x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 100 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 150 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 200 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 250 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 300 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 350 ms
[D][ltr_als_ps:182]: Reconfiguring sensitivity: gain = 96x, time = 400 ms
[W][ltr_als_ps:408]: Too many sensitivity adjustments done. Apparently, sensor reconfiguration fails. Stopping.
```

latonita, 2024-03-04:
@klaasjanhorlings okay. so this is normal operation mode - when there is not enough sensitivity it increases gain and then int.time.
However with super low sensiitivity - counter of adjustments was showing error message too early) I increased it, so it shall properly get to gain 96 and time 400.

klaasjanhorlings, 2024-03-05:
Great, thanks for checking anyway. Wasn't sure what to expect exactly with very low light, but I guess this is as good an option as any. 
Thanks for your work on this, it was a very nice find and looks solid

closed 2024-05-30 by kbx81