esphome/feature-requests #2488: Allow compatible Arduino and esp-idf partition scheme for auto update

X-Ryl669, 2023-11-19:
**Describe the problem you have/What new integration you would like**

Since ESPHome invites to upgrade from Arduino framework to esp-idf framework, it would be a lot easier if it was possible to do so over the air. 

Currently, this doesn't work, because the partition scheme is different between the 2 frameworks, so updating to esp-idf means that the application will not *stick*. It'll work once, but after a reboot, it'll start the previous arduino's firmware.

Yet, it's possible to make it work (see technical discussion below). 

**Please describe your use case for this integration and alternatives you've tried:**

The use case is when a ESPHome project wants to switch from Arduino to ESP-IDF for the framework without having to reprogram the device over the serial link. It would be better if the change could be done via OTA.

**Additional context**

It's possible to do that via 2 possible path. The hacky path is like this:

Currently, the arduino partitioning put the OTA data at offset 0xE000 in flash, while esp-idf put this data at offset 0x9000, see the current code:
```py
def get_arduino_partition_csv(flash_size):
    app_partition_size = APP_PARTITION_SIZES[flash_size]
    eeprom_partition_size = 0x1000  # 4 KB
    spiffs_partition_size = 0xF000  # 60 KB

    app0_partition_start = 0x010000  # 64 KB
    app1_partition_start = app0_partition_start + app_partition_size
    eeprom_partition_start = app1_partition_start + app_partition_size
    spiffs_partition_start = eeprom_partition_start + eeprom_partition_size

    partition_csv = f"""\
nvs,      data, nvs,     0x9000, 0x5000,
otadata,  data, ota,     0xE000, 0x2000,
app0,     app,  ota_0,   0x{app0_partition_start:X}, 0x{app_partition_size:X},
app1,     app,  ota_1,   0x{app1_partition_start:X}, 0x{app_partition_size:X},
eeprom,   data, 0x99,    0x{eeprom_partition_start:X}, 0x{eeprom_partition_size:X},
spiffs,   data, spiffs,  0x{spiffs_partition_start:X}, 0x{spiffs_partition_size:X}
"""
    return partition_csv

def get_idf_partition_csv(flash_size):
    app_partition_size = APP_PARTITION_SIZES[flash_size]

    partition_csv = f"""\
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
app0,     app,  ota_0,   ,        0x{app_partition_size:X},
app1,     app,  ota_1,   ,        0x{app_partition_size:X},
nvs,      data, nvs,     ,        0x6D000,
"""
    return partition_csv 
```
I think the code was written this way trying to optimize the partition layout, since it moved and enlarged the nvs partition to the end of the flash. The `app_partition_size` is the same for both framework. 
Yet, this "gain" isn't a real gain, because esp-idf will align the app partition to multiplies of 0x10000 (see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#offset-size)).
So even if the first 2 partitions in idf scheme only "take" 0x3000 bytes, there is a padding after them of 0xD000 bytes that inserted by esp-idf tools.

In the end, if esp-idf partition was made this way:
```csv
padding, data, undefined,    ,       0x4000,
phy_init, data, phy,     ,        0x1000,
otadata,  data, ota,     ,        0x2000,
app0,     app,  ota_0,   ,        0x{app_partition_size:X},
app1,     app,  ota_1,   ,        0x{app_partition_size:X},
nvs,      data, nvs,     ,        0x6D000,
```
then when updating from arduino to esp-idf it would work OTA since the partition scheme that's written in arduino's bootloader and in the flash would be almost compatible. 

The padding partition could later be converted to a littlefs or fat (instead of currently being wasted as invisible padding). Even better, we could store compressed source YAML in there. 

Once arduino framework's OTA write the app0 or app1 to esp-idf framework, and the ESP32 reboots, it'll update the next OTA to app1 or app0 *and save the actual active partition* to otadata that's located at the same place as Arduino framework is expecting it. So it'll work for the next reboot (unlike currently).

Please notice that the arduino actual nvs partition would be overwritten by the phy partition in esp-idf but it doesn't seem to be an issue in esp-idf firmware, since it ignore completely the partition type before writing into. 

The second solution that's less hackish, would be to write (or have) a minimal firmware that's updating the partition layout in the flash and perform a new OTA when booted at the expected new partition scheme. That's possible but a lot harder to get right.

Bascht74, 2024-01-21:
Could ESPHome use the same saveboot way that Tasmota uses?
That way everybody gets more space for their firmware so maybe the partition layout could be more flexible to be pleasant to both Arduino and ESP-IDF.

... and it would be posible again to migration from Tasmota to ESPHome (right now it has similar side effects, an update is not working)...

X-Ryl669, 2024-02-17:
It's working as expected with this hack. You can now upgrade your framework OTA. See [here](https://github.com/Blackymas/NSPanel_HA_Blueprint/issues/1714#issuecomment-1950276075) for a tutorial