esphome/feature-requests #2489: Reaktivity System to set outputs

autoantwort, 2023-11-19:
**Describe the problem you have/What new integration you would like**
I want a reactivity system like in frontend web frameworks like vue. This could simplify the conditions when to turn something on or off. 
So instead of 
```yaml
switch:
  - platform: gpio
    id: relay
    name: "Power Switch"
    on_turn_on:
      - output.turn_on: has_smoke_power_output
    on_turn_off:
      - output.turn_off: has_smoke_power_output

output:
  - id: has_smoke_power_output
    platform: gpio
```
I want to say:
```yaml
switch:
  - platform: gpio
    id: relay
    name: "Power Switch"

output:
  - id: has_smoke_power_output
    platform: gpio
    value: relay
```

**Please describe your use case for this integration and alternatives you've tried:**
This simplifies the specification for outputs, especially when a output depends on more input then one. For example:
```yaml
sensor:
  - platform: dht
    id: temp_sensor
    temperature:
      name: "Temperature 2"
switch:
  - platform: gpio
    id: relay
    name: "Power Switch"

output:
  - id: power_if_warm
    platform: gpio
    value: relay && temp_sensor.temperature > 30
```
This would be hard to do with only imperative execution commands. 

Maybe we want to use `<id>.value` instead of only `<id>`, but since this would be always required this seems redundant.

iBobik, 2025-04-20:
Probably more practical will be to use lambdas, similarly like Home Assistant does:

```
output:
  - id: power_if_warm
    platform: gpio
    reactive_value: !lambda return id(relay).state && id(temp_sensor).state > 30
```

iBobik, 2025-04-20:
I would use it on many places in the LVGL:
```
     widgets:
        - label:
            text: Insert bills
            text_font: !lambda return id(bank_amount).state > 0 ? montserrat_24 : montserrat_48
        - label:
            hidden: !lambda return id(bank_amount).state == 0
            id: accepting_bank_label
            text_font: montserrat_48
            text: !lambda return id(bank_amount).state + " USD"
        - button:
            hidden: !lambda return id(bank_amount).state == 0
            on_click:
              then:
                - lvgl.page.show: withdraw
            widgets:
              - label:
                  text: Withdraw
                  align: center
```

clydebarrow, 2025-05-02:
This approach was considered when implementing LVGL but the problem is it requires the value to be polled, which adds CPU overhead even when nothing is happening. The event-driven approach avoids that overhead.

Another technique that can be used is to implicitly add listeners to dependencies, but this won't work with lambdas, and can get very complicated and risks feedback loops. Web frameworks like React and Vue can deal with this since they are only working with a DOM which is a tree - within ESPHome you can create arbitrary graphs representing dependencies which makes it quite hard.

I would suggest that if you have multiple state values that might need to be updated in the same way after different events, create a script that updates everything that might be affected, and call that script from the triggers. There will be some redundant updating, but it won't matter too much and it keeps all the update logic in one place.

iBobik, 2025-05-02:
@clydebarrow Thank you for explanation, I understand it now.

I have alternative idea what could give us similar DX but without the reactivity overhead - in widgets definition have everything templatable, but similarly to your suggested approach it will execute only when triggered (on page switch or manually on triggers like on_value on sensors).

By "everything templatable“ I mean I can have lambda (or easier templating engine like Jinja) in pages and widgets definition on all properties.

For example see my previous comment - it could be useable like this, just required to put automations like this on sensors:
```
number:
  - id: bank_amount
    on_value:
      - lvgl.widget.redraw: 
```

closed 2025-05-05 by jesserockz