esphome/feature-requests #2490: Voice Assistant: allow audio feedback tones on events

h3ndrik, 2023-11-21:
**Describe the problem you have/What new integration you would like**
It would be nice if the voice assistant was able to beep if the wake-word got detected (for example). Maybe also on other states.

**Please describe your use case for this integration and alternatives you've tried:**
Currently, I try to intercept the trigger `on_wake_word_detected` and tell it to play a sound (see below). But it doesn't work that way. Also esphome doesn't wait for the on_... to be finished, even if I add something like a delay. All the subsequent events fire milliseconds later. Lighting up LEDs works, though.

**Additional context**

```
i2s_audio:
  [...]

media_player:
  [...]

microphone:
  [...]

voice_assistant:
  on_wake_word_detected:
    - media_player.play_media:
        id: wm8978_audio
        media_url: 'http://192.168.1.21:8123/local/sounds/receive.wav'
```

swiergot, 2023-11-24:
That would be extremely useful. Today I have to look for the listen light so make sure I'm not talking to myself.

h3ndrik, 2023-11-29:
Additional info: You need to go to the ESPhome integration in HA and "configure" your device and allow it to execute services. I've tried the following:

```
  on_wake_word_detected:
    - lambda: ESP_LOGD("voice_assistant", "TRIGGER on_wake_word_detected");
    - homeassistant.service:
        service: media_player.play_media
        data:
         #entity_id: media_player.any_media_player_in_ha
          entity_id: media_player.taudio_taudio_i2saudio
          media_url: 'http://192.168.1.21:8123/local/sounds/receive.wav'
          media_content_type: music
          announce: "true"
```

```
[13:49:59][D][voice_assistant:240]: VAD detected speech
[13:49:59][D][voice_assistant:438]: State changed from WAITING_FOR_VAD to START_PIPELINE
[13:49:59][D][voice_assistant:444]: Desired state set to STREAMING_MICROPHONE
[13:49:59][D][voice_assistant:256]: Requesting start...
[13:49:59][D][voice_assistant:438]: State changed from START_PIPELINE to STARTING_PIPELINE
[13:49:59][D][voice_assistant:459]: Client started, streaming microphone
[13:49:59][D][voice_assistant:438]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[13:49:59][D][voice_assistant:444]: Desired state set to STREAMING_MICROPHONE
[13:49:59][D][voice_assistant:542]: Event Type: 1
[13:49:59][D][voice_assistant:545]: Assist Pipeline running
[13:49:59][D][voice_assistant:465]: TRIGGER on_start
[13:49:59][D][voice_assistant:542]: Event Type: 9
[13:50:02][D][voice_assistant:542]: Event Type: 10
[13:50:02][D][voice_assistant:551]: Wake word detected
[13:50:02][D][voice_assistant:489]: TRIGGER on_wake_word_detected
[13:50:02][D][voice_assistant:542]: Event Type: 3
[13:50:02][D][voice_assistant:556]: STT started
[13:50:02][D][voice_assistant:467]: TRIGGER on_listening
[13:50:02][D][light:036]: 'NeoPixel Light' Setting:
[13:50:02][D][light:051]:   Brightness: 50%
[13:50:02][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[13:50:02][D][light:109]:   Effect: 'Slow Pulse'
[13:50:02][D][media_player:059]: 'TAudio I2SAudio' - Setting
[13:50:02][D][media_player:063]:   Command: STOP
[13:50:04][D][voice_assistant:542]: Event Type: 11
[13:50:04][D][voice_assistant:680]: Starting STT by VAD
[13:50:04][D][voice_assistant:478]: TRIGGER on_stt_vad_start
[13:50:06][D][voice_assistant:542]: Event Type: 12
[13:50:06][D][voice_assistant:684]: STT by VAD end
[13:50:06][D][voice_assistant:480]: TRIGGER on_stt_vad_end
[13:50:06][D][light:036]: 'NeoPixel Light' Setting:
[13:50:06][D][light:051]:   Brightness: 50%
[13:50:06][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[13:50:06][D][light:109]:   Effect: 'Fast Pulse'
[13:50:07][D][voice_assistant:542]: Event Type: 4
[13:50:07][D][voice_assistant:438]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[13:50:07][D][voice_assistant:444]: Desired state set to AWAITING_RESPONSE
[13:50:07][D][voice_assistant:571]: Speech recognised as: "das Wohnzimmerlicht ein"
[13:50:07][D][voice_assistant:511]: TRIGGER on_stt_end
[13:50:07][D][voice_assistant:438]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[13:50:07][D][voice_assistant:542]: Event Type: 5
[13:50:07][D][voice_assistant:576]: Intent started
[13:50:07][D][voice_assistant:542]: TRIGGER on_intent_start
[13:50:07][D][voice_assistant:438]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[13:50:07][D][voice_assistant:542]: Event Type: 6
[13:50:07][D][voice_assistant:544]: TRIGGER on_intent_end
[13:50:07][D][voice_assistant:542]: Event Type: 7
[13:50:07][D][voice_assistant:599]: Response: "wohnzimmerlicht eingeschaltet"
[13:50:07][D][voice_assistant:546]: TRIGGER on_tts_start
[13:50:07][D][light:036]: 'NeoPixel Light' Setting:
[13:50:07][D][light:051]:   Brightness: 50%
[13:50:07][D][light:059]:   Red: 0%, Green: 100%, Blue: 0%
[13:50:07][D][light:109]:   Effect: 'None'
[13:50:07][D][voice_assistant:542]: Event Type: 8
[13:50:07][D][voice_assistant:617]: Response URL: "http://192.168.1.21:8123/api/tts_proxy/6d14c05c27e38b28db1144b87cf987e0ce862b8b_de-de_60afa496b6_tts.piper.raw"
[13:50:07][D][media_player:059]: 'TAudio I2SAudio' - Setting
[13:50:07][D][media_player:066]:   Media URL: http://192.168.1.21:8123/api/tts_proxy/6d14c05c27e38b28db1144b87cf987e0ce862b8b_de-de_60afa496b6_tts.piper.raw
[13:50:07][D][voice_assistant:438]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[13:50:07][D][voice_assistant:444]: Desired state set to STREAMING_RESPONSE
[13:50:07][D][voice_assistant:568]: TRIGGER on_tts_end
[13:50:07][W][component:214]: Component api took a long time for an operation (0.06 s).
[13:50:07][W][component:215]: Components should block for at most 20-30ms.
[13:50:08][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.53 s).
[13:50:08][W][component:215]: Components should block for at most 20-30ms.
[13:50:08][D][voice_assistant:542]: Event Type: 2
[13:50:08][D][voice_assistant:629]: Assist Pipeline ended
[13:50:08][D][voice_assistant:586]: TRIGGER on_end
[13:50:11][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.51 s).
[13:50:11][W][component:215]: Components should block for at most 20-30ms.
[13:50:11][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.50 s).
[13:50:11][W][component:215]: Components should block for at most 20-30ms.
[13:50:11][D][light:036]: 'NeoPixel Light' Setting:
[13:50:11][D][light:051]:   Brightness: 40%
[13:50:11][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[13:50:13][D][voice_assistant:438]: State changed from STREAMING_RESPONSE to IDLE
[13:50:13][D][voice_assistant:444]: Desired state set to IDLE
```

It runs my ESP_LOGD but the service gets lost.

demey, 2023-11-29:
@h3ndrik thanks a lot for the detailed answer.
It looks like that ESPHome software must be changed for this purpose.
Somewhere between TRIGGER on_wake_word_detected and TRIGGER on_listening must be added an confirmation event with possibility to output to  the speaker or an media player.

h3ndrik, 2023-11-29:
> [...] looks like that ESPHome software must be changed for this purpose [...]

Yeah, I think so, too. I don't have a good idea on how to implement it. Seems as of now the voice_assistant just uses Component::defer() to schedule the task and moves on with the state machine logic.

I'm not sure if waiting / blocking has downsides for other users. But maybe we should come up with a solution that works with all the triggers. I could image another (short) beep being useful once it stops listening. And I'd maybe like to replace the whole returned tts answer with a chime. But that's not necessarily a valid use-case.

demey, 2023-11-29:
@h3ndrik I'm not familiar with C++, but based on my experience in other programming languages, after reviewing this code:
https://github.com/esphome/esphome/blob/dev/esphome/components/voice_assistant/voice_assistant.cpp

I think it will be enough to add these lines after the line [538](https://github.com/esphome/esphome/blob/dev/esphome/components/voice_assistant/voice_assistant.cpp#L538)
As well we need to define url value to desired confirmation sound.
```
#ifdef USE_MEDIA_PLAYER
        if (this->media_player_ != nullptr) {
          this->media_player_->make_call().set_media_url(url).perform();
        }
#endif
```

h3ndrik, 2023-12-04:
> I think it will be enough to add these lines after the line [538](https://github.com/esphome/esphome/blob/dev/esphome/components/voice_assistant/voice_assistant.cpp#L538)

That perform() seems to be an async call. It shows up in the debug log but several other things happen within the same second and I think it gets overwritten or something like that. In any case nothing gets played in the end. Like with the other approach.

And I don't think this is the right place. `void VoiceAssistant::on_event(const api::VoiceAssistantEventResponse &msg)` seems to handle the events from Home Assistant. And strictly speaking playing the tone isn't something HA told us to do. It's something from within the trigger where the url would be specified, too.

Maybe we need for the triggered user-actions to finish, before the state-machine transitions to the next state. I believe I tried some 'wait_for' in the yaml. But the voice_assistant seems to like to listen to the home assistant events more than doing the wait I've told it to do.

demey, 2023-12-06:
@h3ndrik I'm agree that perform() seems to be an async call, but we need to set url for media player in any case.
I have analyzed code once again and have the code for testing )
How about these lines after the line 538?
```
#ifdef USE_MEDIA_PLAYER
      bool playing = false;
      if (this->media_player_ != nullptr) {
        this->media_player_->make_call().set_media_url("http://192.168.5.105:8123/local/sounds/confirm.mp3").perform();
        playing = (this->media_player_->state == media_player::MediaPlayerState::MEDIA_PLAYER_STATE_PLAYING);
      }
      if (playing) {
        this->set_timeout("playing", 2000, [this]() {
          this->cancel_timeout("speaker-timeout");
          this->set_state_(State::IDLE, State::IDLE);
        });
      }
#endif
```

kizovinh, 2023-12-06:
Can you guy create a pull request for it, I need this feature :)))

demey, 2023-12-06:
@kizovinh first of all, for the any pull request it MUST be working solution )
At this moment we still trying to find it.

kizovinh, 2023-12-06:
> @kizovinh first of all, for the any pull request it MUST be working solution ) At this moment we still trying to find it.

I knew that, I just want to inspire you guys that this feature is very useful, keep going 👍

kizovinh, 2023-12-06:
> I am having another idea for working around it, is using the "assist in progress" sensor to detected a pipeline start and play media stream. Let me try

Result are the same, the speaker will play only after pipeline end

demey, 2023-12-14:
Finally, I attached to ESP32 that using Arduino framework and media_player instead of speaker, small [rtttl buzzer](https://esphome.io/components/rtttl.html).
In configuration it looks like:
```
output:
  - platform: ledc
    id: buzzer_output
    pin: GPIO13
    frequency: 2000 Hz

rtttl:
  output: buzzer_output
  id: my_rtttl

voice_assistant:
  on_wake_word_detected:
    - rtttl.play: "two_short:d=4,o=5,b=100:16e6,16e6"
```

and it works like a charm.

kizovinh, 2023-12-14:
> Finally, I attached to ESP32 that using Arduino framework and media_player instead of speaker, small [rtttl buzzer](https://esphome.io/components/rtttl.html). In configuration it looks like:
>
> ``` [...]

Yep I think any general gpio will work, for me I use a servo sg90, when detect the wake word, I made the servo rotate and it produce a small sound which is hearable. But a real media player will still give better interaction

vilhalmer, 2023-12-30:
I was just going down the path of using `speaker.play` to hardcode a feedback sound in the yaml (to avoid waiting on #2429), but it has the same problem where the playback gets buffered until the assistant stops its loop.

nielsnl68, 2023-12-30:
Did you know the RTTTL component can now also play sounds over the speaker?

nielsnl68, 2023-12-30:
Did you know the RTTTL component can now also play sounds using the speaker component?

demey, 2023-12-30:
The problem is that the microphone continues to listen at the moment when it is necessary to play the confirmation sound.
I added couple of lines to voice_assistant.cpp to stop microphone during this period of time.
Overall behavior became better and the confirmation sound is no longer perceived as the beginning of a command.
The system works stable, but only if the confirmation sound is output through RTTTL component.
All attempts to output a confirmation sound through components: media_player, speaker or speaker as RTTTL very unstable.

I currently use two versions of the voice assistant using confirmation sounds. 
The first outputs it to an external media player (as STT), the second to a separate RTTTL component.

nielsnl68, 2023-12-30:
yeah, that is the issue indeed. I hope @jesserockz can find a way to auto switch between the two.

demey, 2023-12-30:
My changes in voice_assistant.cpp. Hope it can help:
```
    case api::enums::VOICE_ASSISTANT_WAKE_WORD_END: {
      ESP_LOGD(TAG, "Wake word detected");
// stop microphone for confirmation signal
      this->mic_->stop();
      this->set_state_(State::STOPPING_MICROPHONE);
      this->defer([this]() { this->wake_word_detected_trigger_->trigger(); });
      break;
    }
    case api::enums::VOICE_ASSISTANT_STT_START:
// delay to start microphone to finish confirmation signal
      this->set_timeout("pause", 800, [this]() {
        this->cancel_timeout("pause");
        this->set_state_(State::START_MICROPHONE);
      });
      ESP_LOGD(TAG, "STT started");
      this->defer([this]() { this->listening_trigger_->trigger(); });
      break;
```

h3ndrik, 2024-01-07:
One thing I found: There is a Mutex lock in the i2s_audio component. That is locked and unlocked both by the microphone and the media_player. Since the microphone is always running, I suppose the media_player waits for the microphone to be stopped anyways. And that probably happens once the pipeline gets teared down and started again. Maybe that's the underlying problem.

kizovinh, 2024-01-07:
I switch to the dev branch, and in the lastest release I think they lower the priority of media player over voice assistant. Media player play only when listening to wake word is turned off

Chrissi02, 2024-01-24:
Hi everyone,
I'm facing the same problem as all of you.
I'm currently using an M5Atom Echo for testing, as my ESP Wroom32 + accessories haven't arrived yet.
I would also like to receive a voice response such as "Yes, how can I help?" in addition to the flashing blue LED as soon as my "Hey Jarvis" is recognised.

Have a look at this link (https://www.esp-voice.com/espvoice/), I have no idea what to make of it, especially as 45 money is already a lot.
But the thing has a response after recognising the wake word....

mihaichrapan, 2024-02-02:
I could really use this, too!!!

janstadt, 2024-03-07:
@Chrissi02 have you looked at the esphome docs? You should be able to hook into one of the optional automation steps like `on_wake_word_detected` and have it say something: https://esphome.io/components/voice_assistant.html#configuration

h3ndrik, 2024-03-07:
Back when I last tried it the issue was that you can't do audio output in 'on_wake_word_detected' it will just not play anything.

janstadt, 2024-03-07:
Cripes i forgot your initial issue had `on_wake_word_detected`. My bad. What im doing right now is outputting to an external speaker via homeassistant but i know that not everyone has that functionality in their ecosystem. Did you try any of the other automation steps @h3ndrik ?

smoldersonline, 2024-03-08:
[Link to Discord](https://discord.com/channels/429907082951524364/1211899225915265064/1211899228251627530) - this may be helpful.

I am (very) new to ESPHome configuration, and still very much 'living & learning'. Wake word sound (with this code) works for me, however, (for some reason ...) I lost other functionality included in my config.

derhappy, 2024-03-08:
> [Link to Discord](https://discord.com/channels/429907082951524364/1211899225915265064/1211899228251627530) - this may be helpful.
>
> I am (very) new to ESPHome configuration, and still very much [...]

I‘m not allowed to see this conversation, could you post the relevant part here?

smoldersonline, 2024-03-08:
[image]

derhappy, 2024-03-08:
Thank you @smoldersonline, this looks like an interesting approach!
I'm currently experimenting with local wake word and there it seems to be even easier since wake word detection and voice assistant are separated. Once the media player is available for the esp-idf framework this might also be a viable option.

smoldersonline, 2024-03-08:
I'm using [this configuration](https://gist.github.com/MarkWattTech/4c248cee42b32d1c5f84091256cc3279). Including above code into this configuration will cause that the 'wake word switch' (from the MarkWattTech configuration) stops working. Unfortunately I don't know why this happens.

derhappy, 2024-03-08:
> I'm using [this configuration](https://gist.github.com/MarkWattTech/4c248cee42b32d1c5f84091256cc3279). Including above code into this configuration will cause that the 'wake word switch' (from the MarkWattTech configuration) stops working. Unfortunately I don't know why this happens.

Since you're disabling the wake work detection on_wake_word_detected, do you enable it again in on_end like in the screenshot above or is your coding in on_end still the same as in the MarkWattTech version?

smoldersonline, 2024-03-08:
Thanks a lot for looking into this!

My (current) on_end:

    - delay: 600ms
    - voice_assistant.start_continuous
    - lambda: id(va).set_use_wake_word(true);

Hence, as in above screenshot.

tetele, 2024-03-08:
> I'm using [this configuration](https://gist.github.com/MarkWattTech/4c248cee42b32d1c5f84091256cc3279). Including above code into this configuration will cause that the 'wake word switch' (from the MarkWattTech configuration) stops working. Unfortunately I don't know why this happens.

Mark Watt's config is a very old, buggy, uncredited version of [my config](https://github.com/tetele/onju-voice-satellite) ([initial location](https://gist.github.com/tetele/75815c53cfd2a02412ef153f9c2d24e2). I might be wrong about this, but i think he hasn't updated it since his main drive is getting Youtube views, not helping people :) Try the updates i added, they will help.

> But it feels very wrong to stop the voice_assistant from within the voice_assistant, do the stuff and then (re)start the pipeline. And it's probably not intended that this workaround works.

I think the root of the problem lies within the fact that the I2S audio component can't use both the input and output simultaneously, which is why the `voice_assistant` (which is listening for something) has to be stopped. I agree with you that it's a workaround which probably is not meant to work.

smoldersonline, 2024-03-08:
> Mark Watt's config is a very old, buggy, uncredited version of [my config](https://github.com/tetele/onju-voice-satellite) ([initial [...]

Thanks a lot for your feedback - I was not aware of this. Later today, I will certainly try this with your config.

smoldersonline, 2024-03-12:
> Mark Watt's config is a very old, buggy, uncredited version of [my config](https://github.com/tetele/onju-voice-satellite) ([initial [...]

@tetele - thank you much (!) for sharing your configuration. I have now installed your most recent version. After inclusion of the 'wake word sound' code (as detailed above), I'm still losing the 'wake word switch' from your config.  Unfortunately I don't know what is causing this. Not a very big deal, but for me it still would be helpful to have both.

smoldersonline, 2024-03-12:
You are right, my issue does not belong here. I'm sorry!

s00500, 2024-03-24:
Hey guys,

I have been playing with my own microWakeWord version in the last days and made a simple sound playback work.

Basically like this:
```
micro_wake_word:
  model: okay_nabu
  on_wake_word_detected:
    - logger.log: Local wakeword detected
    - speaker.play:
        data: !lambda return beep;
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
```

The beep is included using a included .h file and currently consists out of a few random bytes... I should probably still search for a better way to encode these... cant be to much data though as microWakeWord already uses a lot of memory...

```
esphome:
  name: "${name}"
  includes:
    - beep.h
```

In beep.h:
```
std::vector<uint8_t> beep = { 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8, 0x52, 0x49, 0x46, 0x46, 0x46, 0xD8, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45, 0x66, 0x6D, 0x74, 0x20, 0x10, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00, 0x44, 0xAC, 0x00, 0x00, 0x10, 0xB1, 0x02, 0x00, 0x04, 0x00, 0x10, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x1A, 0x00, 0x00, 0x00, 0x49, 0x4E, 0x46, 0x4F, 0x49, 0x53, 0x46, 0x54, 0x0E, 0x00, 0x00, 0x00, 0x4C, 0x61, 0x76, 0x66, 0x35, 0x39, 0x2E, 0x32, 0x37, 0x2E, 0x31, 0x30, 0x30, 0x00, 0x64, 0x61, 0x74, 0x61, 0x00, 0xD8 };
```

Eknem, 2024-03-29:
@s00500 Brilliant!!! Thank you!

Djelle, 2024-03-31:
I got something working. Though the sound is just a very short click. But the sound is at least played at the right time.

For me this is just a temporary solution until something better is added in "voice_assistant".

Snippets of relevant code:
```
esp32:
  board: esp32dev
  framework:
    type: arduino

speaker: # MAX98357A
  - platform: i2s_audio
    i2s_audio_id: i2s_out
    id: external_speaker
    dac_type: external
    i2s_dout_pin: GPIO14 # DIN
    mode: mono

voice_assistant:
  id: va
  microphone: external_microphone 
  speaker: external_speaker
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2
  on_wake_word_detected:
    - speaker.play:
        id: external_speaker
        data: [0x80,0x60,0x80,0xA0,0x80]
```

Djelle, 2024-04-05:
I had an idea of using two i2s_dout_pin's. One for speaker and one for media_player. And then feed them both to the same DIN pin on the I2C amplifier board. Though I can't test this, because voice_assistant seems to capture the I2S channels (2s_lrclk_pin and i2s_bclk_pin) and not let anything else use them. Which might be considered a bug that should be fixed.

Relevant code snippet:
```
voice_assistant:
  id: va
  microphone: external_microphone 
  #speaker: external_speaker
  use_wake_word: false
  noise_suppression_level: 2 # 0, 1, 2, 3, 4
  auto_gain: 31dBFS # 0dBFS - 31dBFS
  volume_multiplier: 2 # >0

i2s_audio:
  - id: i2s_in
    i2s_lrclk_pin: GPIO26 # WS
    i2s_bclk_pin: GPIO25 # SCK
  - id: i2s_out
    i2s_lrclk_pin: GPIO13 # LRC
    i2s_bclk_pin: GPIO12 # BCLK

media_player:
  - platform: i2s_audio
    i2s_audio_id: i2s_out
    name: "test"
    id: test
    dac_type: external
    i2s_dout_pin: GPIO27 # DIN
    mode: mono

speaker: # MAX98357A
  - platform: i2s_audio
    i2s_audio_id: i2s_out
    id: external_speaker
    dac_type: external
    i2s_dout_pin: GPIO14 # DIN
    mode: mono

- platform: gpio 
  id: pb_play
  pin:
    number: 21
    mode:
      input: true
      pulldown: true
  filters:
    - delayed_on: 10ms
  on_press:
    - homeassistant.service:
        service: tts.speak
        data:
          entity_id: tts.piper
          cache: 'true'
          media_player_entity_id: media_player.esphome_voice_satellite_1_test
          message: 'Volume'
```
Notice that I have tried with both i2s_dout_pin's connected. And with only one of them connected at a time. 

With the code above, where va.speaker is disabled. I can play through the homeassistant.service (media player) by pressing my button on pin 21.

But if I enable the va.speaker, then homeassistant.service runs with no errors. But there is no data send to 2s_lrclk_pin and/or i2s_bclk_pin (I assume). Though voice_assistant can still play sounds through the external_speaker.

I might be wrong. But I thought that I2S was ment to be shared. Like the speaker and microphone should be able to share it (Though only one can use it at a time). Not sure they actually can. As I have split spk and mic from the start as people seemed to have trouble in that regard.

Anyone have an idea if I am correct in my assumption about I2S?

h3ndrik, 2024-04-08:
> [...] I thought that I2S was meant to be shared. Like the speaker and microphone should be able to share it (Though only one can use it at a time). Not sure they actually can. As I have split spk and mic from the start as people seemed to have trouble in that regard.
> 
> Anyone have an idea if I am correct in my assumption about I2S?

As far as I know the I2S peripheral can do full duplex. You can simultaneously output sound and read from a microphone. The only restriction is RX and TX channels share a clock. So you have to use the same configuration for both (samplerate, ...) and then use an binary OR to enable MODE_TX and MODE_RX simultaneously. You can also implement mixing like some of the ESP32 audio libraries have implemented and output two or more sounds at the same time.

Issue here is that all of that isn't implemented. Instead the `i2s_audio` components use a Mutex to lock everything so just one component at a time can access I2S in esphome. But thats an artificial restriction and unnecessary.

Djelle, 2024-04-20:
I have made a script for my voice satelite, which include two short sounds: Key-click and a beep (more like a twing). You are welcome to copy and use those in your scripts. Or use my full code if you wish.
There is also a solution for these type of scripts locking up at times. Though there might be more than this one problem coursing lockups.
https://github.com/Djelle/ESPHomeVoiceSattelite

Pippisbruffi, 2024-12-21:
Feel like this issue should get more attention. I think audio feedback for wakeword detected is essential on a voice assist! In case you do not have a display/status led on your build or if you simply don't want to look at the device to know if it got the wakeword. Right now I'm using the workaround: -when wakeword detected - stop assist - play sound -  start assist, but a built in solution will be a lot better/faster...

demey, 2026-04-25:
Hi @jftkcs 
Thanks a lot for the useful information.
I'm trying to use your version for just one word or sound, and almost everything works as expected, 
but right after downloading the new firmware, my device's screen became just white, without any images, a blank screen.
The software installed from [here](https://www.home-assistant.io/voice_control/s3_box_voice_assistant) brought the image back to the screen, so it's 100% a software problem. 
Can you suggest something about this?

jftkcs, 2026-04-26:
Hmm, how are you flashing - ESPHome app in homeassistant? Can you share logs?

jftkcs, 2026-04-26:
Just tested again and not having issues. 
Also - updated that branch to use AEC to remove the mute limitation and put another copy of the file for a single chime in the directory. AEC subtracts the chime from the audio stream so that you can start talking before the chime is finished.

demey, 2026-04-26:
> Hmm, how are you flashing - ESPHome app in homeassistant? Streaming wake detection or mww? Can you share logs?

I tried two methods. The first is ESPHome Device Builder as a constructor (2026.4.1) in HA, and this method gives a blank screen after flashing using ESPHome Web. The device works with voice commands, but the screen is blank.
The build log can be found here:
[logs_esp32-s3-box-3_compile.txt](https://github.com/user-attachments/files/27098326/logs_esp32-s3-box-3_compile.txt)
And the device log after flashing:

[logs.txt](https://github.com/user-attachments/files/27098351/logs.txt)

which contains lines:
[13:47:23][E][display:017]: Could not allocate buffer for display!
[13:47:23][E][component:284]: display was marked as failed
[13:47:23][E][component:405]: display set Error flag: unspecified
[13:47:23][E][component:224]:   display is marked FAILED: unspecified

After installing the software from https://www.home-assistant.io/voice_control/s3_box_voice_assistant, I noticed that it had ESPHome version 2025.11.3 installed. So I tried using the same version of ESPHome Device Builder to build the firmware using your YAML configuration. I got the error on validation:
[logs_esp32-s3-box-3_validate_2025.11.3.txt](https://github.com/user-attachments/files/27098194/logs_esp32-s3-box-3_validate_2025.11.3.txt)

jftkcs, 2026-04-26:
Oh haha that change/break in esphome just happened last week and I hadn't updated in a bit (https://github.com/esphome/esphome/issues/15784, https://github.com/esphome/wake-word-voice-assistants/pull/202).

I was holding back for a bit due to dependencies but looks like it's all fixed. Thanks for your help! Updated the repo.