esphome/feature-requests #2491: Disable Outputs for PID Climate Component

Bidthedog, 2023-11-22:
**Describe the problem you have/What new integration you would like**

I'm using the Climate PID component (https://esphome.io/components/climate/pid) to adjust the speed of a fan using the cool_output. I also have a speed fan (https://esphome.io/components/fan/speed) component. Both point to the same output (LEDC) which is wired to a PWM fan.

Both components work as expected independently, but when both components are enabled (i.e. present in the YAML), they conflict - as you might expect. The climate PID component eventually wins in all arguments (I assume because it is constantly sending signals to the output, whereas the fan component does not seem to be).

What I need to be able to do, is switch between which component controls the output. Turning the climate control OFF doesn't make any difference, I think because it continues to send on_state events, and (I expect) sends 0 signals to the output?

I thought I might be able to proxy the output to another component (like a templated number with an on_value event that sets the output level based on what it receives), and have both the fan and PID components optionally send to that (or have it control which signal it reacts to based on other settings), but it doesn't seem possible.

**Please describe your use case for this integration and alternatives you've tried:**

In this example, I am using two identical controllers to control two different banks of fans based on two different input temperatures. The fan banks are located at the top and bottom of a enclosed cabinet, which generates varying levels of heat depending what device inside the cabinet is on.

* For certain devices (ones that get particularly hot) I want to be able to enable the climate controller and have the PID loop handle the fans.
* For other devices I want to be able to disable the climate controller, and override the fan speeds to different preset speeds, depending on other sensor readings outside of the cabinet (sensor inputs & automations are set up in Home Assistant, not as part of this ESP board).
* I want to be able to turn the fans off completely when no devices are on (I can already achieve this by setting the PID Climate mode to "OFF" AND setting the speed fan override to "OFF").

**Additional context**

Interestingly, I have observed the following behaviour (which I find odd, but I'm no expert):

* When the PID Climate component mode is set to "OFF", it continues to emit on_state events, and when I enable debug, it still makes a lot of log noise.
* When I completely disable the entity in Home Assistant, on_state continues to be called (and the fan output seems to continue to receive signals from the PID Climate control).
* When I enable my speed fan override, I set the PID Climate component mode to "OFF". The fan spins up to the set speed, then spins back down again. This also happens if I have the PID Climate component mode set to "OFF" and I change the Fan Speed component speed.
* This behaviour doesn't seem to be that consistent. If I restart the ESP board with the PIC Climate component set to "OFF", I don't see any on_state calls until I set it to "COOL" again. If I then change it back to "OFF", I continue to see the on_state calls.

I appreciate that HASS doesn't directly have any control over what the PID controller is doing, but I don't seem to be able to disable the PID Controller completely inside ESPHome, or at least stop it emitting a signal to the PWM output.

nagyrobi, 2023-11-22:
Try https://esphome.io/components/output/template

patrickcollins12, 2024-01-27:
Here is a solution using @nagyrobi  's suggestion of using template output

Bidthedog, 2024-01-31:
Thanks Patrick - good to know you're active on here - it was your blog that I originally used as inspiration for my bits and pieces.

I'll take a look - been waylaid with real life (aren't we all), but good to know there is a solution to this.