esphome/feature-requests #2493: Support for VEML7700 ambient light sensor

joshuaboniface, 2023-11-24:
**Describe the problem you have/What new integration you would like**
I would like a *proper* integration for the VEML7700 light sensor. At least 4 people, including myself (https://community.home-assistant.io/t/veml7700-with-esphome/469545/) have requested this.

**Please describe your use case for this integration and alternatives you've tried:**
#218 provides two "workarounds", neither of which are compatible.

The first is to use the BH1750 component, which simply does not work. Nor does the TSL2591 component.

The second is to use a custom component which directly leverages the Adafruit Arduino libraries. This is both deprecated as per the notice on the [custom components article](https://esphome.io/custom/custom_component.html), and incompatible with non-Arduino frameworks, namely ESP-IDF, which can break other components. This is also the "workaround" referenced in the above forum post.

**Additional context**
#218 is over 4 years old, and its "workarounds" are no longer valid.
#1915 was erroneously closed as a "duplicate" of the above despite the clearly changing situation re: custom components, and since it has been aggressively locked, there is no opportunity to provide this feedback there. Thus, I am opening a new feature request to ensure this request is properly tracked.

lboue, 2023-12-13:
I'd also like to be able to use my [Adafruit Right Angle VEML7700 Lux Sensor - I2C Light Sensor - STEMMA QT / Qwiic](https://www.adafruit.com/product/5378) module.

Kilill, 2023-12-18:
I second this as well

B48D81EFCC, 2023-12-26:
Would love to use the VEML7700 with esphome.

latonita, 2023-12-30:
I can take a look into this when my sensor arrives in a week or so

latonita, 2024-01-07:
@B48D81EFCC @Kilill @lboue @joshuaboniface 
please try draft version I've just made https://github.com/latonita/esphome-veml7700

gain = one from [X1, X2, X1/4, X1/8]
integration time = one of [25, 50, 100, 200, 400, 800]
glass_attenuation_factor not yet working

plans to do AUTO gain/time mode, but not sure yet
```
esphome:
  name: veml7700-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# esp8266:
#   board: nodemcuv2

logger:
  level: DEBUG

external_components:
  - source: github://latonita/esphome-veml7700
    components: [ veml7700 ]

i2c:
  sda: GPIO25
  scl: GPIO32
  scan: true
  frequency: 5kHz

sensor:
  - platform: veml7700
    address: 0x10
    update_interval: 30s
    gain: X1
    integration_time: 100ms
    glass_attenuation_factor: 1.0
    ambient_light:
      name: "Ambient light"
    white:
      name: "White"

```

lboue, 2024-01-07:
> please try draft version I've just made https://github.com/latonita/esphome-veml7700

Thanks for your work, it's great! I've just done a test:
- compilation works with ESPHome 2023.12.5 and platformio/espressif32@5.4.0
- the sensor also works correctly

[image]

B48D81EFCC, 2024-01-10:
Great, thanks a lot.
I am going to use/test the sensor on the weekend.

latonita, 2024-01-10:
please use 
```
external_components:
  - source: github://pr#6067
    components: [veml7700]
```

B48D81EFCC, 2024-01-13:
Hi @latonita,

I just gave it a quick try, generally it is working fine. Thanks a lot.
I discovered two things:

- The example snippet in the documentation is not 100% correct, since name on the root level is not allowed (and not needed).
If you just copy & past the example you get the following error:
`
platform: veml7700  
[name] is an invalid option for [sensor.veml7700]. Please check the indentation. 
`

- I always get a warning message:
`
[16:48:39][W][component:204]: Component veml7700.sensor took a long time for an operation (0.08 s).  
[16:48:39][W][component:205]: Components should block for at most 20-30ms. 
`

My first guess was, that the reason is the default integration time of 100ms.
I tried to set the integration time to 25m by setting `auto_mode: false` and `integration_time: 25ms` but I still got the very same warning.

latonita, 2024-01-14:
@B48D81EFCC thanks for testing. I will fix documentation!
As for timing, it's definitely not possible to do measurements in a given time in auto mode, I will check the fixed settings if it can be improved.
Thanks for your findings.

Kilill, 2024-01-14:
Tested with a ESP32 C3 Super mini board (these are now my new favorites when it comes to ESP-Home), works like charm right out of the box! 
Thanks a lot!

latonita, 2024-01-22:
@Carpetner 

> I get a strange warning when compiling; and even more strangely, same warning appears sometimes between the logs:

thats okay, its a "note", we will try to suppress this, but its just an information

> Auto-mode does not work for me, it is constantly re-adjusting (again under intense light conditions). Maybe the the code could wait some time at the lowest gain and int-time for things to settle? My response times are also quite long, between 70 and 120 ms. I have another VEML in the mail and will do some more testing...

I would be happy to see full logs, to understand whats happening and how it can be improved. I hope there is no some kind of indefinite loop which is definitely a bug! And for high bright conditions  it should work really fast if you leave default gain/time unchanged.
What happening is 
1) to do a change we need to turn off device first
2) wait for int.time period
3) turn it on
4) to make sure we get stable and good samples we need to wait at least 2, better 3 int.time periods
5) we get good stable reading
6) too dim/bright -> adjust and go to 1)
In case we get data too quick - it might be more "noisy" and not reliable.

However, we definitely not in a hurry with this measurements, I don't believe we need anything faster than 60s update ;))

Thanks in advance for full logs.

latonita, 2024-01-22:
Are you using version from PR? github://pr#6067

latonita, 2024-01-24:
@Carpetner i would appretiate if you can upload some logs to find out where is the delay.
One of the reason I found on my setup - publishing state is taking long some time, so i had to split publishing in few batches.
maybe i will need to split more.
thanks

latonita, 2024-01-24:
@Carpetner looks like it takes a while to send data, i will split data publishing in even smaller pieces

B48D81EFCC, 2024-02-24:
HI @latonita,

it looks like your PR passed all checks and will be merged into the esphome codebase soon, am I correct?

latonita, 2024-02-24:
@B48D81EFCC Yes, its correct. Its in the long queue to be reviewed and merged in.

rici44, 2024-05-30:
During building I got errror
In file included from src/esphome/components/veml7700/veml7700.cpp:1:
src/esphome/components/veml7700/veml7700.h:70:10: note: offset of packed bit-field 'esphome::veml7700::ConfigurationRegister::<unnamed struct>::ALS_IT' has changed in GCC 4.4
   70 |   struct {
      |          ^

And after boot sensor was not found
[23:15:29][I][i2c.arduino:096]: Results from i2c bus scan:
[23:15:29][I][i2c.arduino:098]: Found no i2c devices!
[23:15:29][E][veml7700:112]: Communication with I2C VEML7700/6030 failed!
[23:15:29][E][component:082]:   Component veml7700.sensor is marked FAILED

Hillhousetour, 2025-02-27:
[image]

I need help about the i2c adress... 

saids to me that Communication with I2C VEML7700/6030 failed!

B48D81EFCC, 2025-02-27:
Hi,

it seems the wiring is wrong or the sensor is broken. You can see in the logs that the I2C bus scan did not find any device.

Can you use an alternative pin for GPIO05? 
At the beginning of the logs you can see a warning regarding GPIO05. The warning says the pin should not used for I/O.

saschachina, 2025-06-14:
**Thank you very much guys**, the VEML7700 works perfectly for me with an ESP32.
Here is my code:

```
esphome:
  name: lichtsensor_aussen
  friendly_name: Lichtsensor Aussen

esp32:
  board: esp32dev
  framework:
    type: arduino

i2c:
  sda: GPIO25
  scl: GPIO32
  scan: true
  frequency: 10kHz

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXX"

ota:
  - platform: esphome
    password: "XXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid_ext
  password: !secret wifi_password_ext

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lichtsensor-Aussen"
    password: "XXXXXXXX"

captive_portal:

sensor:
  - platform: veml7700
    address: 0x10
    update_interval: 60s
    auto_mode: True
    gain: 1/8x
    integration_time: 100ms
    lux_compensation: True
    glass_attenuation_factor: 1.0

    ambient_light: 
      name: "Umgebungslicht"
    actual_gain:
      name: "Aktueller Verstärkungswert"
    full_spectrum:
      name: "Vollspektrum"
    infrared:
      name: "Nah IR Spektrum"
    actual_integration_time:
      name: "Integrationszeit"

  - platform: wifi_signal
    name: "WLAN Signal dB"
    id: wifi_signal_lichtsensor_aussen
    update_interval: 60s

  - platform: copy
    source_id: wifi_signal_lichtsensor_aussen
    name: "WLAN Signal Prozent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"

button:
  - platform: restart
    name: "Neustart"
```

[image]