esphome/feature-requests #2494: Less updates for the sensors

pstimpel, 2023-11-24:
**Describe the problem you have/What new integration you would like**
On every update of ESPHome HA asks for updating all sensor devices as well

**Please describe your use case for this integration and alternatives you've tried:**
Even if there are updates affecting the ESPHome itself only, HA asks to update all devices. This is just inconvenient. Therefore, I would like to see a change so that updates not affecting the sensors are not asking for updating sensors. 

**Additional context**

ssieb, 2023-11-24:
How do you expect it to know what devices would be affected?

pstimpel, 2023-11-24:
If I understood the changelog right, the last update was not affecting sensors, as example.

ssieb, 2023-11-24:
Ok, but how is HA supposed to know what is affected and what is on your device?

pstimpel, 2023-11-24:
If an update is meant to do change inside ESPHome-server component itself, but without doing anything on the sensor side, why do sensors need updates at all?

ssieb, 2023-11-24:
You haven't answered the question.  How is HA supposed to know what changes there were and what it would apply to?

pstimpel, 2023-11-24:
This question has to be answered by the developer who wants to integrate this feature.

ssieb, 2023-11-24:
The answer is that there's no way to do that without making a lot of extra work for very little benefit.

pstimpel, 2023-11-25:
Then that is the answer to it.