esphome/feature-requests #2497: Ability to choose an arbitrary STT service

jerome83136, 2023-11-25:
**Describe the problem you have/What new integration you would like**

I would like to use an arbitrary URL for reaching the STT service after the wakeword is triggered
My goal is to decouple from home assistant for STT and the rest of the pipeline.

**Please describe your use case for this integration and alternatives you've tried:**

As far as I know, esphome reaches open assistant STT service by default, after the wakeword is triggered. I would like being able to customize this to reach the STT service of my choice.

**Additional context**

I think I only miss the variable that needs to be changed in esphome yaml, for pointing to the STT service URL
I assume esphome is sending some kind of wav file to the STT service containing the pronounced voice command. (Maybe I'm wrong?)
If this is true; I would like being able to choose the URL of the STT service that will receive the wav file from esphome

jerome83136, 2023-11-27:
Hello,
Can I have a feedback on this request please ?
Maybe pointing me to some relevant documentation for achieving what I want ?
Or at least to tell me that it is not feasible?
Thank you for your help
Best regards

nagyrobi, 2023-11-28:
Please create a thread on this, on our Discord support channel.

jerome83136, 2023-11-30:
Closing after feedback received on Discord

This is not feasible this way because voice assistant for esp is meant for working with home assistant

closed 2023-11-30 by jerome83136