esphome/feature-requests #2498: Separate Camera functionality from esp32_camera to make it reusable by other components

rrooggiieerr, 2023-11-25:
**Describe the problem you have/What new integration you would like**

CameraImage, CameraImageReader and maybe other functionality are now part of the esp32_camera component. Separate the camera functionality to a reusable component so other component can reuse the functionality.

**Please describe your use case for this integration and alternatives you've tried:**

I'd like to implement an IR 32x32 pixel sensor and make it as a camera available.

**Additional context**

rrooggiieerr, 2023-11-25:
This integration (not mine) is actually copying the camera functionality from the esp32_camera component, thus there is a need for a reusable camera component.
https://github.com/anton-malakhov/esphome_webcam/

ssieb, 2024-02-02:
This is a valid feature request.  At one time, I actually attempted it for the same reason, but it's a lot of work and I didn't get it finished.

rrooggiieerr, 2026-09-08:
It looks like this is now implemented:

https://esphome.io/components/camera/