esphome/feature-requests #2499: Allow custom pin schema for Dallas component

ptr727, 2023-11-27:
**Describe the problem you have/What new integration you would like**

I would like to be able to use advanced `pin` configuration for the `dallas` component, including using `ignore_strapping_warning` and `pullup`.

**Please describe your use case for this integration and alternatives you've tried:**

I am using an ESP with the only external GPIO usable for Dallas DS18B20 being GPIO15.  
GPIO15 is a bootstrap pin, and I get warnings about not using `ignore_strapping_warning`.  
But the `dallas` component does not support advanced `pin` configuration.  
I would also like to enable the internal pullup so that I don't need to use an external 4.7K resistor.

**Additional context**

Current config:

```yaml
# Use expansion slot Pin6 GPIO15 for Dallas
# https://esphome.io/components/sensor/dallas.html
dallas:
  # TODO: How to add ignore_strapping_warning: true
  # WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
  - pin: GPIO15
    update_interval: 5s
```

Desired config e.g.:
```yaml
# Use expansion slot Pin6 GPIO15 for Dallas
# https://esphome.io/components/sensor/dallas.html
dallas:
  - pin: 
       number: GPIO15
       pullup: true
       ignore_strapping_warning: true
    update_interval: 5s
```

ptr727, 2024-07-04:
@ssieb could this be considered for 1-wire please?

ssieb, 2024-07-04:
The component already uses the internal pullup, but that's quite weak, so might not work.

ptr727, 2024-07-04:
> The component already uses the internal pullup, but that's quite weak, so might not work.

Ok, thx, getting rid of the strapping warning would be nice.