esphome/feature-requests #2500: Support for m5dial touch driver

dgaust, 2023-11-27:
**Describe the problem you have/What new integration you would like**
The m5dial uses a GC9A01 screen with a FT3267 touch driver. I have been able to integrate the display using an external component, but not the touchscreen.

**Please describe your use case for this integration and alternatives you've tried:**
I have tried external components with used for other devices with GC9A01 based screens (like the ones with CST816S series touchscreens), but have not had any luck registering touch events.

**Additional context**
Details of the m5dial are here, including pinouts and other details

https://docs.m5stack.com/en/core/M5Dial

dgaust, 2023-12-01:
I think? this might be the relevant driver file for the touchscreen

https://github.com/Xinyuan-LilyGO/T-RGB/blob/b5f004cfe723dd54a0fa63f04eaf4668e20ac77f/examples/factory/ft3267.cpp
https://github.com/Xinyuan-LilyGO/T-RGB/blob/b5f004cfe723dd54a0fa63f04eaf4668e20ac77f/examples/factory/ft3267.h

lboue, 2023-12-13:
I'd like to do exactly the same.

lboue, 2024-03-09:
Any new about FT3267 touch driver ?
- [HAL_TouchPanel.cpp](https://github.com/sidianneng/scooter_dashboard/blob/e0a45bde65f3733847d02ce47ffabe27a3341f71/Software/X-Track/USER/HAL/HAL_TouchPanel.cpp#L4) from [scooter_dashboard](https://github.com/sidianneng/scooter_dashboard)
- Arduino [FT3267_Lib](https://github.com/io4nn1s/Arduino/tree/main/Libraries/FT3267_Lib)

lboue, 2024-03-09:
I found an [ESPHome config file](https://github.com/dgaust/esphome_files/blob/main/m4_dial_v2.yaml) for M5Dial 1.28" Round Touch Screen

```yaml
external_components:
  - source: github://dgaust/esphome@gc9a01
    components: [ ft3267, gc9a01 ]
    refresh: 0s
```

It works !
```
[09:48:03][D][ft3267.touchscreen:132]: Read 0 status, id: 0, pos 113/73
[09:48:03][D][ft3267.touchscreen:132]: Read 0 status, id: 0, pos 130/79
[09:48:05][D][ft3267.touchscreen:132]: Read 0 status, id: 0, pos 129/44
[09:48:05][D][ft3267.touchscreen:132]: Read 0 status, id: 0, pos 117/11
```

dgaust, 2024-03-09:
Yeah, that's mine, I ended up getting a custom component working for the touchscreen, based off a similar touchscreen component

lboue, 2024-03-09:
> Yeah, that's mine, I ended up writing a custom component that got the touchscreen working based off a similar touchscreen component 

You should have put a comment here to share 😉

dgaust, 2024-03-09:
Probably, if someone with more experience than me with esphome components wants to the repository is here.... It works, but no idea how it complies with esphome's requirements

https://github.com/dgaust/esphome/tree/gc9a01/esphome/components/ft3267

More discussion on how it all comes together is over

https://community.home-assistant.io/t/m5stack-dial-esp32-s3-smart-rotary-knob/623518