esphome/feature-requests #2502: ld2410 dynamic throttle

clinta, 2023-11-28:
**Describe the problem you have/What new integration you would like**
Trying to fine-tune the ld2410, but the `throttle` configuration causes it to miss short spikes that need to be taken into account when tuning the thresholds. It would be nice if the throttle value could be dynamically set so that it could be turned down when using engineering mode and tuning the sensor, then turned back up when complete without having to recompile.

**Please describe your use case for this integration and alternatives you've tried:**
Other thoughts are replacing the throttle function that simply skips values with a running max so that every throttle period the max value for each gate is sent rather than the latest.

**Additional context**
Thanks @regevbr for the excellent ld2410 integration.

regevbr, 2023-11-28:
@clinta this is a nice addition sadly I don't have the time. You can open a PR I will gladly review it. Should be pretty easy - add a number sensor that when changed, can control the throttle (and should be set to the initial value on boot). Be aware though that if the device is restarted, the throttle will revert back to the default value that is in the compiled config.

Unfocused, 2024-02-12:
If your goal is tuning thresholds/filters, then another alternative is to use the [copy component](https://esphome.io/components/copy). Remove any filters from the original sensor, and set `disabled_by_default` to `true`. Then add a copy sensor (using the original sensor as its source) and add your filters to the copy. Then you have one sensor that outputs filtered values, and another that outputs raw values that you can enable only when needed. It provides a way of comparing the two (filtered vs raw) and seeing how the filters are affecting the reported values.