esphome/feature-requests #2505: Allow configuration of the dashboard base path

JamieMagee, 2023-11-30:
**Describe the problem you have/What new integration you would like**

Support serving the dashboard under a different base path. For example instead of `/`, it could be served under `/esphome`.

**Please describe your use case for this integration and alternatives you've tried:**

This allows reverse proxying via nginx, caddy, etc. using a different path, instead of a different subdomain i.e. `esphome.example.com` vs `example.com/esphome`.

**Additional context**

ssieb, 2023-11-30:
That's up to the proxy to define the path.

JamieMagee, 2023-11-30:
It has to be supported and configured by both the proxy and the application. For example, here's a configuration I enabled for Caddy with the [`reverse_proxy` directive][1]

```
rpi.tailnet-0b15.ts.net {
  reverse_proxy /esphome* http://[::1]:6052
}
```

But when I `curl https://rpi.tailnet-0b15.ts.net/esphome` I get a 404, and in the esphome logs I see

```
Nov 27 19:58:32 rpi esphome[939]: 2023-11-27 19:58:32,815 INFO Starting dashboard web server on http://localhost:6052 and configuration dir /var/lib/esphome...
Nov 28 13:36:56 rpi esphome[939]: 2023-11-28 13:36:56,627 WARNING 404 GET /esphome (::1) 1.80ms
```

This issue is so common with reverse proxies, that it's documented in Caddy's wiki: [The "subfolder problem", OR, "why can't I reverse proxy my app into a subfolder?"](https://caddy.community/t/the-subfolder-problem-or-why-cant-i-reverse-proxy-my-app-into-a-subfolder/8575). Here's the key excerpt:

> ## Fixing the problem
>
> [...] either the HTTP application must correct its URLs to add the subfolder you want, or you need to configure Caddy to let the app have the web root it wants.

[1]: https://caddyserver.com/docs/caddyfile/directives/reverse_proxy

JamieMagee, 2023-11-30:
Updating my Caddy configuration to be

```
handle_path /esphome* {
  reverse_proxy http://[::1]:6052
}
```

which strips the `/esphome` prefix, does allow me to load the `index.html`:

<details>
<summary>index.html</summary>

```html
<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="UTF-8" />
<meta name="viewport" content="width=device-width, initial-scale=1.0" />
<title>Dashboard - ESPHome</title>
<link
rel="shortcut icon"
href="./static/images/favicon.ico?hash=2655a073"
/>
<link
rel="stylesheet"
href="./static/fonts/material-icons/material-icons.css?hash=ead4c276"
/>
<link
rel="preload"
href="./static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838"
as="font"
crossorigin="anonymous"
/>
<link
rel="stylesheet"
href="./static/css/esphome-2.css?hash=c222cb3b"
/>
</head>
<body>

<esphome-main
version="2023.11.4"
docsLink="https://www.esphome.io/"

></esphome-main>

<script
src="./static/js/esphome/index-6rBAfUo5.js"
type="module"
></script>

</body>
</html>
```

</details>

However, the links are returned as absolute URLs i.e. `https://rpi.tailnet-0b15.ts.net/static/js/esphome/index-6rBAfUo5.js`, which returns a 404.

Proxies are able to rewrite the URL, but not the returned content.

randybb, 2023-11-30:
Haven't done that for more than 10 years, but there was always a way how to do it.
Check HA how it is handling forwarding to addons.  And since it uses websockets, it needs a bit more love with headers.

JamieMagee, 2023-11-30:
I appreciate your help, but unfortunately it's not possible without support for setting a base path from esphome.

For example, here's [the documentation for setting up a reverse proxy on Radarr][1]. The key excerpt about configuring Radarr correctly:

> These examples assumes the default port of 7878 and that you set a `baseurl` of `radarr`

The proxy cannot do it without support from the application.

[1]: https://wiki.servarr.com/radarr/installation/reverse-proxy

ssieb, 2023-12-01:
Then that's a limitation of the proxy that you're using.  Apache can do it.

diepeterpan, 2024-05-25:
> Updating my Caddy configuration to be
> 
> ```
> handle_path /esphome* {
>   reverse_proxy http://[::1]:6052
> }
> ```
> 
> which strips the `/esphome` prefix, does allow me to load the `index.html`:
> 
> index.html
> However, the links are returned as absolute URLs i.e. `https://rpi.tailnet-0b15.ts.net/static/js/esphome/index-6rBAfUo5.js`, which returns a 404.
> 
> Proxies are able to rewrite the URL, but not the returned content.

It should return relative URLs then we don't need a base. Relative URLs don't start with HTTP or a / ;  in the above example a relative URL in content would be `static/js/esphome/index-6rBAfUo5.js`.  I have seen so many open-source project's which is not use relative URLs.   Our company will never touch these projects without them supporting relative paths.