esphome/feature-requests #2507: support commit SHA as legal value in ref in external_components/source

stleusc, 2023-12-01:
Support commit SHA as legal value in ref in external_components/source

There is no alternative.
Use case is a commit to the external component repo that has a bug e.g.
Sometimes selecting specific commit is helpful.

LelandSindt, 2024-03-20:
I had started a thread in discord before finding this issue.. figured I would re-post here. 
---- 
https://esphome.io/components/external_components.html

I see that `ref:` calls out `branch` and `tag` I had hoped that `commit` would work. (undocumented) but no such luck. 

```
INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome-web-003948.yaml...
INFO Cloning https://github.com/ssieb/esphome_components@c1d639faefe8bbd37bea626051b910fb7d657da1
Failed config

external_components: [source /config/esphome-web-003948.yaml:29]
  - 
    Remote branch c1d639faefe8bbd37bea626051b910fb7d657da1 not found in upstream origin.
    source: 
      type: git
      url: https://github.com/ssieb/esphome_components
      ref: c1d639faefe8bbd37bea626051b910fb7d657da1
    components: 
      - serial_csv
```

I try to make a habit of pinning my dependencies in case breaking changes are introduced in the future. In this case there are no tags to pin to available. (which is the choice of the developer to provide or not)

LelandSindt, 2024-03-21:
I _think_ I could extent this pattern 

https://github.com/esphome/esphome/commit/73accf747f64daf32c25a127b8410d87335b615d

something like....

`fetch_commit = ref.startswith("commit/")`

:thinking:

LelandSindt, 2024-03-23:
https://github.com/esphome/esphome/compare/dev...LelandSindt:esphome:commit :thinking:

closed 2024-04-02 by jesserockz