esphome/feature-requests #2510: IRK Sensor/Onboarding for Private BLE

margau, 2023-12-03:
**Describe the problem you have/What new integration you would like**
Let esphome provide the IRK (Identity Resolving Key) for devices to be used in the Home Assistant [Private BLE](https://www.home-assistant.io/integrations/private_ble_device) integration.

**Please describe your use case for this integration and alternatives you've tried:**

Private BLE needs the IRK of a device, which apparently can be determined by peering a (e.g, android) device using bluetooth. With private BLE, esphome often is already present as bluetooth proxy, so the "onboarding" of a new device might be useful here also.

espresense as alternative firmware already does this (https://www.home-assistant.io/integrations/private_ble_device/#espresense), but needing another esp just to read out the IRK, while the remaining parts of the presence tracking are done by Private BLE/esphome bluetooth proxy (instead of espresense) is not straightforward.

margau, 2023-12-03:
This is somewhat related to https://github.com/esphome/feature-requests/issues/2285, but not identical. 2285 is about handling IRK inside the tracking itself, while this is about initially determining the IRK (mainly for the existing private ble integration)

agittins, 2024-03-21:
It  looks a lot like someone has already written a component to do this: https://github.com/dgrnbrg/appdaemon-configs/tree/a5780a16e15b8b10f6dd1cdad20250adc13a9451/custom_components/irk_enrollment

It looks like it will require some updating to [get it working with current esphome](https://github.com/dgrnbrg/appdaemon-configs/issues/4), but it looks like all the heavy lifting has been done already.

It would be great to have a component to do this, especially if it could be turned on and off as required (assuming it might have an impact on proxy performance). Likewise if it could be hooked into the Private BLE Device integration, to give a seamless enrolment into HA.

dgrnbrg, 2024-04-05:
I updated my component so that it builds again, but I haven't tested it. I'd be happy to help fixing/testing when I can, but I have a very large amount of other ESPHome development I'm doing, and I won't be able to focus on the entire lifecycle of this until late summer or fall.

myhomeiot, 2024-05-04:
You can use small [fix](https://github.com/dgrnbrg/appdaemon-configs/pull/7)

dgrnbrg, 2024-05-05:
I've merged @myhomeiot's fix. Thank you!

mano3m, 2025-12-30:
It seems similar functionality is made by @Foleychris here: [Foleychris/esphome-esp32-ble-irk](https://github.com/Foleychris/esphome-esp32-ble-irk) and @DerekSeaman discusses [DerekSeaman/irk-capture](https://github.com/DerekSeaman/irk-capture) here: [https://community.home-assistant.io/t/new-esphome-bluetooth-irk-package/964447](https://community.home-assistant.io/t/new-esphome-bluetooth-irk-package/964447)

DerekSeaman, 2025-12-31:
@mano3m My ESPHome project uses the NimBLE package, which does not allow concurrent BLE proxy functionality along with IRK capturing. They are mutually exclusive. Also, I don't think you'd want a permanently advertised BLE device that random people can pair to. Capturing an IRK should be a very infrequent process.

mano3m, 2025-12-31:
> @mano3m<https://github.com/mano3m> I'm about to push a big update (v1.5) to my IRK capture component later today. Better Android compatibility, and tons of under the hood changes from my prior public version.
>
> v1.5.0 on my Github repo is now live on the main branch. I added keyboard emulation, which seems to work great with Android phones without any needed workarounds. Android watches are still problematic, but I may try and work on a solution for that.
>
> IRK Capture for ESPHome<https://github.com/DerekSeaman/irk-capture>

I'll try it out. Yesterday only an old Samsung tablet could find the device (and worked perfectly after that).

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

mano3m, 2026-01-02:
My Galaxy Watch7 indeed cannot find the heart monitor nor keyboard. I can search for bluetooth audio devices like headhpones, however. Could you try this?

DerekSeaman, 2026-01-02:
> My Galaxy Watch7 indeed cannot find the heart monitor nor keyboard. I can search for bluetooth audio devices like headhpones, however. Could you try this?

Unfortunately, "classic Bluetooth" audio is an older protocol, and not the same as Bluetooth Low Energy (BLE). BLE is what can require the IRK exchange for added security. Even if I could emulate a classic Bluetooth audio device, and could pair, it would not result in capturing an IRK. Also, the NimBLE Bluetooth stack that I'm using does NOT support the classic Bluetooth audio stack.

mano3m, 2026-01-02:
simulating a phone is probably the same? You could try it the other way around and have the watch function as a heart monitor: [reddit:how_to_broadcast_heart_rate_hr_over_bluetooth_low](https://www.reddit.com/r/GalaxyWatch/comments/1fdfp4r/how_to_broadcast_heart_rate_hr_over_bluetooth_low/)

Also a generic problem: https://github.com/home-assistant/android/issues/5673

DerekSeaman, 2026-01-02:
> Yes, I did it! An app called Gear Tracker II can be installed on the watch and has the option to add an external BLE hart monitor. That one sees your device immediately which gives the IRK when pairing!

Perfect! I've updated my Git README instructions to add a dedicated Android watch section and I added a link to that app.

DerekSeaman, 2026-01-11:
@howels I revised the installation instructions and now have three distinct options. Option #1 now points everything to Github so no local filesystem modifications are needed.