esphome/feature-requests #2514: S0 output generator

ivo7B4, 2023-12-10:
**S0 signal generator output from ESPHome, as provided by some power meters.**

**To use the own solar produced energy efficiently, HeatPumps provide some interfaces to provide them the current additionaly produced solary energy. Some HeatPumps have an S0 input such that with an S0 output signal, this information can be provided to the HeatPump. This is much better as the SG (SmartGrid) value only as SG can only be set to one specific threashold but does not provide modulation information to the HeatPump.
An alternativ is to use an additional power meter with an S0 signal and this power meter connected not for the currently used energy, but for the energy supplied back to the grid. And to connect this S0 signal to the HeatPump. As I have this information from the inverter of the solar system in HomeAssistant, I clearly would prefer to generate such a signal myself with ESPHome (with providing this value from HomeAssistant to ESPHome). In addition, this also would provide more flexibility, e.g. to keep an offset for some energy reserve in case another device switches on or if there is some variability with the current solar production etc.
There are various projects and also ESPHome functionality to read out the S0 signals from power meters to get the current electricty usage. But I haven't seen any project or library support to generate such an S0 signal. Besides the software for this, it also would be great to have a hardware schematic for this project. As far as I have the knowledge, I'm happy to contribute myself.**

**Besides that I think this would be a great feature and would optimize the usage costs of electricity, I mainly see the environmental benefit when the energy is used at the time it is produced. Thus, please vote and support this feature request. Thank you!**

latonita, 2023-12-29:
@ivo7B4 what is the input you have? which component?
Its possible to write simple script to generate pulses based on 1 minute energy consumption for example. I dont believe its important to have better resolution.
1. input - energy consumed/produced for last X seconds period
2. output - proper amount of pulses during next X seconds period
3. rinse, repeat

then implementations details - you might fire up all pulses at once (flash x times, need to provide minimum acceptable flash length) or spread it evenly through the period (use pwm)

latonita, 2023-12-29:
@ivo7B4 
Here is an example I made for blinking the led.
I used `interval` as a trigger but you need to use your solar device component and use "on_value:" trigger to get used/produced energy

```
esphome:
  name: s0-8266

esp8266:
  board: nodemcuv2

logger:
  level: INFO

globals:
  - id: impulses_to_make
    type: int
    restore_value: no
    initial_value: '0'

switch:
  - platform: gpio
    id: s0
    pin: GPIO02
    inverted: true
  - platform: template
    id: s0_generator
    optimistic: true
    on_turn_on:
      - while:
          condition: 
            lambda: return id(impulses_to_make) > 0;
          then:
            - switch.turn_on: s0
            - delay: 30ms
            - switch.turn_off: s0
            - delay: 60ms
            - lambda: id(impulses_to_make)--;
      - switch.turn_off: s0_generator

interval:
  - interval: 15s
    then:
      - lambda: |- 
          // check how many energy used since last check
          // and calculate how many flashes to add to the queue
          // bigger intervals - less errors, but lower update rate for external device
          // shorter intervals - more accumulated errors
          //
          // assume we would like to pretend we are power meter with 1000 imp/kwh
          // it means 1 Wh = 1 impulse
          // if delta energy used is 25 Wh, then we need to do 25Wh * 1 = 25 impulses
          id(impulses_to_make) += 25; 
          ESP_LOGI("main","We need to flash %d times",id(impulses_to_make));
          if (!id(s0_generator).state) 
            id(s0_generator).turn_on();

 ```

ivo7B4, 2023-12-30:
> @ivo7B4 what is the input you have? which component? Its possible to write simple script to generate pulses based on 1 minute energy consumption for example. I dont believe its important to have [...]

@latonita thank you for your questions and your example!
As this S0 output won't be used to count the energy, but mainly to inform e.g. a heatpump (or other devices that get from an S0 signal the currently supplied energy to the grid), I think it's easier to take the current power to the grid (and more responsive to e.g. changes, as it's based on the latest power information) and to calculate the S0 low duration based on the assumption that the power to the grid stays on average the same until the next update, which should be sufficient if this update is done frequently enough.
I've meanwhile started to implement an embedded SW implementation directly on an ESP32 (with MQTT interface to HomeAssistant), where I provide this power information whenever I have an update and then calculate the corresponding S0 Signal.
I'd still prefer an ESPhome solution due to the additonal functionalities, such as directly OTA updates etc, but I'm not sure whether I can achieve these dynamic low period durations with ESPhome.
P.S.: I also think an ESP32 is better as the S0 Signal is specified up to max. 20mA. And depending on the Optocoupler, you need a similar current on the GPIO. And the ESP32 has higher max. current as an ESP8266, as far as I know.

ivo7B4, 2023-12-30:
> @ivo7B4 Here is an example I made for blinking the led. You can use optocoupler to switch on/off your s0 input of the heatpump. I used `interval` as a trigger but you need to use your solar device component and use "on_value:" trigger to get used/produced energy
> 
> as per IEC standard for S0 signal - pulse ON shall be at least 30ms + pulse OFF state for at least 30ms as well
> 
> ```
> esphome:
>   name: s0-8266
> 
> esp8266:
>   board: nodemcuv2
> 
> logger:
>   level: INFO
> 
> globals:
>   - id: impulses_to_make
>     type: int
>     restore_value: no
>     initial_value: '0'
> 
> switch:
>   - platform: gpio
>     id: s0
>     pin: GPIO02
>     inverted: true
>   - platform: template
>     id: s0_generator
>     optimistic: true
>     on_turn_on:
>       - while:
>           condition: 
>             lambda: return id(impulses_to_make) > 0;
>           then:
>             - switch.turn_on: s0
>             - delay: 40ms
>             - switch.turn_off: s0
>             - delay: 40ms
>             - lambda: id(impulses_to_make)--;
>       - switch.turn_off: s0_generator
> 
> interval:
>   - interval: 15s
>     then:
>       - lambda: |- 
>           // check how many energy used since last check
>           // and calculate how many flashes to add to the queue
>           // bigger intervals - less errors, but lower update rate for external device
>           // shorter intervals - more accumulated errors
>           //
>           // assume we would like to pretend we are power meter with 1000 imp/kwh
>           // it means 1 Wh = 1 impulse
>           // if delta energy used is 25 Wh, then we need to do 25Wh * 1 = 25 impulses
>           id(impulses_to_make) += 25; 
>           ESP_LOGI("main","We need to flash %d times",id(impulses_to_make));
>           if (!id(s0_generator).state) 
>             id(s0_generator).turn_on();
> ```

@latonita , thank you for your example!
Please see my previous comments. And based on this example, I see that I need to get a better knowledge about the lambda logic/functionality in ESPhome.

latonita, 2024-01-31:
@ivo7B4 Yesterday I needed to make an emulator of electricity meter pulsing its light. This is something I came up.
You set  current "Power" and then it generates impulses like normal device will do - with proper delay.
It doesnt take real consumption but just makes proper signal frequency.

```
esphome:
  name: power-meter-emulator
  on_loop:
    then:
      - lambda: |-
          static uint32_t last_time = 0;
         // 1600 imp/kWh =
          uint32_t pulse_width_ms = 1000*(3600*1000/1600)/(id(req_power).state);

          if ((pulse_width_ms) > 0) {
            uint32_t now = millis();
            if (now - last_time > (pulse_width_ms)) {
                last_time = now;
                id(out_power_pulse).turn_on();
                delay(30);
                id(out_power_pulse).turn_off();
                ESP_LOGD("loop", "pulse_width_ms: %d", (pulse_width_ms));
            }
          }
esp32:
  board: esp32dev
  framework:
#    type: esp-idf
    type: arduino

# esp8266:
#   board: nodemcuv2

logger:
  level: DEBUG

globals:
  - id: total_pulses
    type: int
    restore_value: false
    initial_value: '0'
  - id: pulse_width_ms
    type: int
    restore_value: false
    initial_value: '0'

number:
  - platform: template
    id: req_power
    name: "Requested power"
    optimistic: true
    min_value: 0
    max_value: 2000
    step: 50
    initial_value: 0

output:
  - platform: gpio
    id: out_power_pulse
    pin: GPIO18
    inverted: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 5min
  power_save_mode: NONE
  fast_connect: true
api:
  password: !secret api_password

ota:
  password: !secret ota_password

web_server:
  port: 80
  auth:
    username: !secret web_user
    password: !secret web_password

```

drRobertDev, 2025-01-28:
Is work good? because i'm interest to do the same

ivo7B4, 2025-01-29:
> Is work good? because i'm interest to do the same

I have a solution. But I decided to implement an own software program for an ESP32 (I started with an ESP32 WIFI, and then changed to a ESP32 with POE LAN) as that provided much more flexiblity to me (at least as I don't know ESPhome that well and as I have more the software developer background). 
And as I had added many smaller protection logics in software, e.g. that the S0 signals don't stay at a higher value in case that the communication might be interrupted for some reason (e.g. network issue).

In addition, the hardware part was more effort as I did a small PCB as I didn't want to connect just some breadboard to the heatpump and also, as I did calculation and a small simulation to design the S0 hardware, and also did various tests. But when you also would like to do this, I assume that you are aware that you need to have GPIO, resistors, diode, ..., to make such an S0 interface (and not only the corresponding software).

drRobertDev, 2025-01-29:
I understand your points. Well i think to start with esphome and use 4N25 for output S0, then will see what issue can happen.

ivo7B4, 2025-01-29:
You also need to consider the right resistor values. Please consider that most schematics that at least I found in the internet are for 5V GPIOs, but (most? all?) ESP32 boards have 3.3V output.
You thus need to adjust the resistors for a 3.3V output and also check that the output current stays in safe limits (but most ESP32 boards have a stronger output than e.g. Arduino boards, at least as far as I know).

ivo7B4, 2025-11-16:
@metalMajor , which schematic are you looking for?
The PCB (which I did with the Fritzing software) or the LTSpice model (which I did to check on the components, e.g. the resistor value) or just a drawing or ...?
And FYI, I did the software with the ArduinoIDE and also have some additional logic in HA as I transfer the data from the inverter to HA to that embedded device which then generates the S0 pulses.
I haven't uploaded this information. What do you reommend for sharing? Especially as all would be without any warranty. I am rather careful here as this gets connected to more expensive devices (heatpumps). I did many additional checks, e.g. also used a PowerSupply and also double checked with an Oscilloscope etc. after I had soldered everything together. But mistakes can still happen, e.g. there might still be a mistake that I did or could be a mistake while soldering, ...
Maybe this link is of some interest for you: https://forum.iobroker.net/topic/58860/s0-signal-mit-wemos-d1-mini-erzeugen/78
(I did two versions: One for an ESP32 WIFI board first, then I decided to use the Olimex ESP32 board with LAN and POE)

metalMajor, 2025-11-22:
So some more background: 

1/ I have no battery, so I want to use as much of my solar production immediately, because buying electricity is much more expensive than what you get when you return the solar production to the net.
2/ My Nibe heatpump can read the state of my SMA inverter over modbus TCP and it shows the current solar production. That works. 
3/ You can then set some settings that it should only heat the internal boiler when there is solar production so you can make use of the information for scheduling. However, it does not know the other usage in my house, so it may as wel schedule while other stuff is using electricity too and so it can not schedule correctly for my use case. 
4/ So I mailed Nibe and they say, the only way to give that information is via S0, and they added that "the heatpump does not use that info for scheduling". It's only acts as a sort of recorder of information, that is it. So yeah, I will not bother then.

And the thing is: in the latest software update of the Nibe, next to the settings for the modbus TCP inverter, there is a new button "Search for smart meter". But there is no information to be found in that. I mailed Nibe again, but now they did not answer yet. So I let it search, but it did not find my P1 dongle on the network. But probably they mean some other smart meter attached to the solar panels. So I thought, let's check if I can fake an energy meter to my SMA inverter and yes, there are "emeter.py" scripts. So I managed to fake an SMA Energy Meter to my SMA inverter, but the "Search for smart meter" did not find it, although the SMA inverter showed the exact values the my fake SMA energy meter produces. 

So yeah, without knowing more information about Nibe's "search for smart meter" function I'm stuck with my investigation.

Aat this point I'm just nerding out haha: I am making a Homey app that converts my P1 dongle values to my fake SMA energy meter, just for fun. Then at least my inverter has all information.

Thanks for your message!

drRobertDev, 2025-11-22:
The thing should be quite simple: you just need to use the four S1, S2, S3, S4 contacts available on the board to limit the maximum output power, without doing anything odd with the S0 contact.

The problem I’ve encountered with the EBLA (three-phase) is that when setting a current limit in Amps, the heat pump, once it reaches the maximum configured power, doesn’t slow down — instead, it shuts off completely and then restarts shortly after, looping continuously.
For example, if I set 3 Amps, it starts from 0, ramps up to 3, then shuts down for 2–3 minutes and restarts, repeating this cycle. Yet in the manual it clearly states that the pump should reduce its power.

I also tried using kW, but it behaves the same way. And besides that, the displayed values are completely wrong — only the Amp reading shows the correct value.