esphome/feature-requests #2516: Fingerprint sensor HLK-FPM383C (or GROW R557) LED control

mrbluebrett, 2023-12-13:
**Describe the problem you have/What new integration you would like**

I have a fingerprint sensor HLK-FPM383C (also labeled in some websites as Grow R557), that is working good with fingerprint grow integration. However the aura LED control, even it prints command set succesful, does not light the LED ring.

**Please describe your use case for this integration and alternatives you've tried:**

The LED ring of this sensor is not working.

**Additional context**

I attach the documentation I got from the seller, even if I don't know if this is the correct one.
[FPM383C用户手册V1.2.zh-CN.en.docx](https://github.com/esphome/feature-requests/files/13659421/FPM383C.V1.2.zh-CN.en.docx)

krangchen, 2024-03-20:
Hey,
I actually have the same problem as I thought I have just a flat version of the same 503/507 sensor just in flat housing... However, it works partially for me. I can get get green and red colors using e.g.
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: GREEN
        count: 4
       
However, I had to use a different git library,
external_components:
  - source: github://alexborro/esphome@Sleep_mode_rebase
    refresh: 0s
    components: [ fingerprint_grow ]
for it to work at all... see here:
https://github.com/esphome/issues/issues/5507
With the current library and HASS-core 2024.2.1 I had similar behavior and thought it was a lib problem first...

Hope this helps to some extend. Still needs official fixing/inclusion of the corresponding library.

mrbluebrett, 2024-03-20:
I will take a look at it and try it tomorrow.
Thanks!