esphome/feature-requests #2519: Adafruit I2C Stemma QT Rotary Encoder Breakout

lboue, 2023-12-15:
**Adding support for Adafruit I2C Stemma QT Rotary Encoder Breakout**

Adding support for Adafruit I2C Stemma QT Rotary Encoder Breakout

**Please describe your use case for this integration and alternatives you've tried:**

I would use the encoder and NeoPixel on board.

**Additional context**

The onboard microcontroller is programmed with our seesaw firmware and will track all pulses and pins for you and then save the incremental value for querying at any time over I2C. 

I2C address: 0x36

ssieb, 2023-12-15:
If it's running seesaw, you could try my seesaw component: https://github.com/ssieb/custom_components/tree/master/components/seesaw
You will need to set the i2c address.

lboue, 2023-12-18:
> If it's running seesaw, you could try my seesaw component: https://github.com/ssieb/custom_components/tree/master/components/seesaw You will need to set the i2c address.

Thank you. I'll try to adapt it to my needs.

ssieb, 2023-12-18:
I was thinking it would work right away.  You can find me on discord if you need help using it.

lboue, 2023-12-18:
> I was thinking it would work right away. You can find me on discord if you need help using it.

https://github.com/ssieb/custom_components/blob/master/components/seesaw/__init__.py#L31
`).extend(i2c.i2c_device_schema(0x49))

For the encoder module, the I2C address is 0x36.
`

ssieb, 2023-12-18:
You really should ask about this on discord.  For any i2c device, you specify the address using `address: 0x36`.  Put that on the top-level `seesaw:` component.  That's why I said you would need to set it.  Don't continue this here.

lboue, 2023-12-30:
The problem I'm facing is that the [seesaw](https://github.com/ssieb/custom_components/tree/master/components/seesaw) component only supports a single encoder, whereas it is possible to have a chain of 8 encoders.

> Using the three on-board address jumpers, you can connect up to 8 of these rotary encoders to a single I2C port. The first will be at address 0x36, the last at 0x3D when all three jumpers are soldered.

lboue, 2024-04-08:
Here is a working config for [Adafruit I2C Quad Rotary Encoder Breakout with NeoPixel - STEMMA QT / Qwiic, Product ID: 5752
]( https://www.adafruit.com/product/5752):

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: [ seesaw ]

seesaw:
  address: 0x49
  # The default I2C address is 0x49. Using the three onboard address jumpers, you can connect up to 8 of these encoders on a single I2C port. The first one will be at address 0x49, the last one at 0x51 when all three jumpers are cut open.
  # https://learn.adafruit.com/adafruit-i2c-quad-rotary-encoder-breakout
  
sensor:
  - platform: seesaw
    id: encoder0
    type: encoder
    name: "Seesaw encoder #0"

  - platform: seesaw
    id: encoder1
    type: encoder
    name: "Seesaw encoder #1"

  - platform: seesaw
    id: encoder2
    type: encoder
    name: "Seesaw encoder #2"

  - platform: seesaw
    id: encoder3
    type: encoder
    name: "Seesaw encoder #3"

binary_sensor:
  - platform: seesaw
    id: button0
    name: "Seesaw encoder button #0"

  - platform: seesaw
    id: button1
    name: "Seesaw encoder button #1"

  - platform: seesaw
    id: button2
    name: "Seesaw encoder button #2"

  - platform: seesaw
    id: button3
    name: "Seesaw encoder button #3"

light:
  - platform: seesaw
    id: pixel0
    name: "Seesaw neopixel LED#0"
    
  - platform: seesaw
    id: pixel1
    name: "Seesaw neopixel LED#1"
    
  - platform: seesaw
    id: pixel2
    name: "Seesaw neopixel LED#2"
    
  - platform: seesaw
    id: pixel3
    name: "Seesaw neopixel LED#3"
```