esphome/feature-requests #2522: Ability to specify more than one active page for touchscreen sensors

BrantleyHobbs, 2023-12-16:
**Describe the problem you have/What new integration you would like**

I'd like to be able to create a touchscreen button that is reused on more than one page (but not on all pages).  Perhaps the "page_id" attribute in the YAML would be able to accept a space or comma-separated list of pages to operate on?

**Please describe your use case for this integration and alternatives you've tried:**

There are situations where it would be extremely convenient for the touchscreen binary_sensor platform to have the ability to specify more than one page_id for the sensor to be active on.  For example, a "return" button could be drawn on the multiple "sub" pages which would perform the same act of returning the user to the main page in the UI.  The return button should **not** be drawn on the main UI page itself, so simply allowing it to exist on all pages is not a good solution.

**Additional context**

A proposed configuration example for this feature could be (proposed syntax change in parenthesis):
```
binary_sensor:
  - platform: touchscreen
    name: Return Button
    id: return_button
    page_id: details (settings credits eyecandy)
    ....
```

Zoogara, 2023-12-21:
page_id: is already implemented. At the bottom of the configuration variables on the touchscreen page.

It's per sensor.  So you can define a set of touch areas for one page then a different one for the next etc.

BrantleyHobbs, 2023-12-21:
Yes, I see page_id is there, but it only accepts a single page ID for its
"argument".  I would like to be able to specify multiple pages for this
binary sensor to be active on, and I don't see a way of doing that without
defining the sensor multiple times.

pixelwave, 2024-06-18:
Would love that too!