esphome/feature-requests #2525: Daly BMS - integration with the energy dashboard

bogdanr, 2023-12-18:
It would be nice if the daly BMS integration would provide the sensors for energy going into the battery and energy commoing out of the battery to interface with the energy dashboard.

B48D81EFCC, 2023-12-26:
Hi @bogdanr,

is this really related to esphome or more related to Home Assistant?
I guess you are referring to the Energy Dashboard included in Home Assistant.

clydebarrow, 2023-12-27:
It does - it provides battery voltage and current, the polarity of of the current indicates charge or discharge, and you would multiply voltage by current to get power, and integrate that to get energy. So all you need to do is create template entities in HA to represent charge and discharge power. How exactly to do that would be a question for an HA channel, not here.

You could also create template sensors in ESPHome, but that's a question for the #general-support Discord channel, not a feature request.

closed 2023-12-27 by clydebarrow