esphome/feature-requests #2528: The way to ignore yaml file in configuration directory

asergunov, 2023-12-21:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I'm using cspell in my project which allows to have a cspell.yaml in directories to configure cspell settings for that directory. I really like to have yaml, not json for consistency. But esphome dashboard shows this cspell file as device configuration. I'd like to have a way to make esphome ignore that file.

ptr727, 2026-10-14:
I have the same problem when using `codecov.yml`, we really need a way to ignore specific yml files that are not device config but live in the repo root.

ssieb, 2026-10-14:
You can put them in a subdirectory.

ptr727, 2026-10-14:
> You can put them in a subdirectory.

I use shared github actions tooling across a fleet of repos that expects codecov in root, it would be easier if this repo supported ignoring non-esphome files, e.g. an ignore file.

randybb, 2026-10-15:
codecov accepts `.codecov.yml`