esphome/feature-requests #2529: Delayed initialization I2C bus

tiimsvk, 2023-12-21:
**Describe the problem you have/What new integration you would like**
Is it possible to integrate in the code an option where the time delay of initialization of the i2c bus would be set?

```
i2c:
  sda: "$i2c_sda_pin"
  scl: "$i2c_scl_pin"
  scan: True
  delay: 300ms # delay after boot
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

tiimsvk, 2023-12-31:
SOLUTION:
[https://community.home-assistant.io/t/delayed-initialization-i2c-bus/659101/5](url)

nagyrobi, 2023-12-31:
Correct link: https://community.home-assistant.io/t/delayed-initialization-i2c-bus/659101/5

zagi988, 2024-03-14:
That's not a solution, that's a workaround. And it would be nice if priorities aren't hardcoded since there is a feature called: "on_boot"

miroslavpetrov, 2024-07-03:
Adafruit Feather ESP32v2 has a separate LDO(EN Pin of the LDO is connected to GPIO2 on the ESP32) that provides power to the QT connector. So before I2C bus initialisation,  GPIO2 must be pulled high. Can the esphome team suggest a more elegant fix to this issue?

half-duplex, 2025-11-30:
I'm also having this issue with i2c on the grove connector of an M5StickC Plus. It would be nice to have a way to make the bus work without infuriating workarounds requiring manual maintenance.