esphome/feature-requests #2533: display driver ST7796UI 480x320 MCU8080 8Bit

pantileionut, 2023-12-25:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

hello can support for ST7796UI 480x320 MCU 8080 8 Bit. it's the improved version for WT32-sc01, with the last update from 12.2023, touchscreen support also came.
WT32-sc01 Plus /ESP32-S3

latonita, 2023-12-27:
What about ili9xxx component?
https://esphome.io/components/display/ili9xxx

pantileionut, 2023-12-28:
ili9xxx are drivers for spi connections.... ST7796UI has parallel connection 8 bit MCU 8080

xNUTx, 2024-02-17:
I will second this request. I think this may be the way to go in the future as it in general is faster then SPI driven displays.

patmtp35, 2024-03-22:
one more help about having sc01-plus not working

dkalliv, 2024-07-28:
Hi there, I have added support for the WT32-SC01 Plus module to TFT_eSPI and created a pull request (3423). While it is being reviewed you can give it a spin using my [fork](https://github.com/dkalliv/TFT_eSPI).