esphome/feature-requests #2534: ESPHome Add-ons container NanoCpus setting.

tufand, 2023-12-26:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

EspHome Add-ons container crash every time on building, we solved only set the container cpus limit
I think required cpus limit parameters every starting home assistant on boot

ssieb, 2023-12-26:
There is a setting to limit the number of CPUs.  The crashing is most likely due to running out of RAM.

tufand, 2023-12-27:
CPU's over 100% is build process, but change container maximum CPU usage is one running normally.
[image]

Every restart recreate add-ons esphome container on home assistant.. 
if settings parameters maximum cpu problem solve

tufand, 2023-12-27:
[image]
I already set it in esphome, but container cpu limit 1 solved the problem.
The problem is that the container CPU limit is lifted every time it restarts.