esphome/feature-requests #2535: Add support for MH Z19

bendamat, 2023-12-26:
**Describe the problem you have/What new integration you would like**

I'd like to get support for the MH Z19 CO2 sensor from Gravity.

**Please describe your use case for this integration and alternatives you've tried:**

Use case: measure CO2 in ambiant air.
I've looked at other available sensors (especially [T6613/15](https://esphome.io/components/sensor/t6615) which seems similar, but it uses UART, whereas Gravity's one uses PWM)

**Additional context**

bendamat, 2023-12-27:
Hi, I got it to work using a `pulse_width` sensor and a template where I calculate the ppm value based on the pwm value. I derived the formula from the [DFRobot wiki page describing how to use this sensor](https://wiki.dfrobot.com/Gravity__Infrared_CO2_Sensor_For_Arduino_SKU__SEN0219).

Here's a snipped of my sensors:
```
  - platform: pulse_width
    pin: 33
    id: co2pwm
  - platform: template
    name: "CO2 in ppm"
    unit_of_measurement: "ppm"
    lambda: |-
      if(id(co2pwm).state < 0.08) {
        return 0.0;
      }
      else {
        return (id(co2pwm).state * 5000) - 10;
      }
```

closed 2023-12-27 by bendamat

RemBrandNL, 2023-12-28:
Can you explain in a bit more detail how you got this working / connected this to your ESP? I have this same sensor and would like to use it with a D2 mini. By the way, it actually does support UART as well. Still at the start of my journey and lack skills to figure this out by myself ;-) Cheers,
Remco

[image]

bendamat, 2023-12-29:
Hi, the sensor itself supports three ways of reading the values: analog, UART, and PWM. I have the [Gravity version of the sensor board](https://wiki.dfrobot.com/Gravity__Infrared_CO2_Sensor_For_Arduino_SKU__SEN0219), where only the pins for PWM are broken out to a 3 pin connector. That's why I had to use PWM. Technically the other pins are there, but I'd need to desolder the board to get access to all pins.
Also I've read [an article that compared the accuracy and reliability of the three different ways to read the sensor](https://diyi0t.com/mh-z14a-co2-meter-arduino-esp8266-esp32/) and PWM seems to be most reliable. The article also explains the different pins in more details that what you have above.
Hope that helps!