esphome/feature-requests #2537: Tilt in Time Based Cover

schu-r, 2023-12-27:
**Describe the problem you have/What new integration you would like**
I have four Time Based covers. All of them work just with Relais and 24V. I can easy run them up and down. But I want these covers to close but not to make it that dark in the room. So i could give them a little push up so the turn and everything is bright without the Sun heating my room. The Turn takes about 1 Second. So if you move the cover more down first it makes it more dark and then moves down.  

**Please describe your use case for this integration and alternatives you've tried:**

I need the covers closed so that the sun wont heat up the room up to 30°C but on the other Hand I dont want to have to switch on the light because it is dark in the room. Just open the covers a bit, so that the sund wont go in directly.

**Additional context**

There was an issue #664 which was marked as 'alternative-offered' but i dont want to do this for all covers in the house. This is messy. There got to be a cleaner solution.

ssieb, 2023-12-28:
Duplicate of #664.
I reopened that one since it has more discussion, including the "workaround".

closed 2023-12-28 by ssieb

klaudiusz223, 2024-01-06:
Hi @schu-r,
Please try my component which I have been using for almost a year
In fact, it's ready to PR
https://github.com/klaudiusz223/esphome/tree/tilt-cover

Documentation for the component
https://github.com/klaudiusz223/esphome-docs/blob/tilt-cover/components/cover/time_based_tilt.rst

Information on how to use my component is available in this thread
https://github.com/esphome/feature-requests/issues/2087#issuecomment-1421419734