esphome/feature-requests #2542: Allow specifying `vars:` for Remote/Git Packages

barndawgie, 2024-01-02:
**Describe the problem you have/What new integration you would like**
We would like to create configurations in which we instantiate multiple copies of our components, but with different configuration. We are trying to do this using remote packages, but hitting an issue where `vars:` is not supported for remote packages as they are for `!include` packages.

E.g. we would like to create config like the following:
```yaml
econet:
  url: https://github.com/bitfliq/esphome-econet
  ref: zones-initial
  file: econet_hvac_base.yaml
zone1:
  url: https://github.com/bitfliq/esphome-econet
  ref: zones-initial
  file: econet_hvac_zone_template.yaml
  vars:
    zoneid: "1"
    zonename: "Zone1"
    requestmod: 4
    thermostat_address: 0x380
zone2:
  url: https://github.com/bitfliq/esphome-econet
  ref: zones-initial
  file: econet_hvac_zone_template.yaml
  vars:
    zoneid: "2"
    zonename: "Zone2"
    requestmod: 5
    thermostat_address: 0x381
```

**Please describe your use case for this integration and alternatives you've tried:**
Our ESPHome component (https://github.com/esphome-econet/esphome-econet) can be used to control a thermostat. For homes with multi-zone HVAC systems, we'd want to instantiate multiple thermostats on a single ESPHome controller.

**Additional context**
Some additional details around this ask were discussed in ESPHome discord: https://discord.com/channels/429907082951524364/1191494270624993420

KG3RK3N, 2024-01-30:
I want to use configurations from my own git repo on the NAS, that feature would makes it easier for multiple copies of components in combination with the home assistant esphome addon. Because the addon don't support multiple configuration files on a easy way.

barndawgie, 2024-02-04:
This would really benefit some of our configurations over at https://github.com/esphome-econet/esphome-econet as for multi-zone HVAC systems you will want to create a component for each zone/thermostat and currently we end up copying lots of yaml configurations when we'd rather have a single YAML that we can import repeatedly for each zone.

mariodivece, 2024-02-16:
This would highly reduce unnecessary package copies and make them much more extensible and maintainable.

pszafer, 2024-10-15:
I made PR which allows to include vars in remote package files. You can test it.
https://github.com/esphome/esphome/pull/7606

closed 2025-02-26 by jesserockz