esphome/feature-requests #2546: GPIO Viewer ...

ChristophCaina, 2024-01-04:
**Describe the problem you have/What new integration you would like**
I just read about a new Arduino library for ESP32 boards, which would allow you to see the behave of an IO Pin in a Webviewer.
The Library puts a small webinterface on the ESP and you can then connect the website and monitor your IO pins...

I wonder, if it would be possible to implement this into ESPHome, to allow some debugging ...?

link to the repository:
https://github.com/thelastoutpostworkshop/gpio_viewer

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jurgen2005, 2024-01-06:
This would be great and fun!

erazor666, 2024-01-06:
Looks like I also requested the same feature. Looks neat!

trantoriana, 2024-01-07:
Looks like the GPIOViewer source needs to be rearranged according to this:

* https://esphome.io/guides/contributing#contributing-to-esphome

It looks like  the author does not know ESPHome, perhaps someone of the core ESPHome team can help or give some tips to @thelastoutpostworkshop so this has any change of success?

nagyrobi, 2024-01-07:
That library is heavily dependant on Arduino libs, that means it won't work on esp-idf version of ESPHome.

erazor666, 2024-01-08:
https://github.com/thelastoutpostworkshop/gpio_viewer/issues/50

Joo01, 2024-03-06:
Has anyone managed to get GPIOViewer to work in ESPHome in the meantime?

RFDarter, 2024-06-27:
I like the idea, but since this library is heavlily based on the arduino frsmework and esphome like to move away frpm that wewould need.