esphome/feature-requests #2547: Add restore_mode DISABLED  option in Light component

ruben0909, 2024-01-05:
**Describe the problem you have/What new integration you would like**

Add restore_mode DISABLED  option in Light component

**Please describe your use case for this integration and alternatives you've tried:**

I have an ESP32 with ESPHOME, it works well but it restarts every now and then. I have some Light components configured on it that send RF codes. My problem is that every time it restarts, it tries to recover the state, but since those lights can also be manually turned on, it would be preferable if it didn’t try to recover any state.
ALTERNATIVES I HAVE TRIED 
Use restore_from_flash to avoid restoring it but it is not for ESP32 so it does not work 

**Additional context**
The SWITCH component does have this functionality as described in the restore_mode section of https://esphome.io/components/switch/index.html

I could try to replicate the functionality of the Switch in the light component but my C++ skills are between scarce and none.

Thank you, and sorry for my english

t3chguy, 2024-01-17:
I had a crack at this on a fork for precisely the same purpose: an RF light.

You can try it by adding

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/t3chguy/esphome/
      ref: LIGHT_RESTORE_DISABLED
    components: [ light ]
    refresh: 0s
```

and `restore_mode: DISABLED` to your light definition

ssieb, 2024-01-17:
So what state will the light have when you reboot?

t3chguy, 2024-01-17:
The programmatic/reported state will be false, but it will not write/trigger false, akin (but not exactly the same as) the switch(_output) component

jesserockz, 2024-01-17:
I guess the ideal solution for this is a `template` light that has `assumed_state` similar to template switch and template cover and then doesnt really care for the actual state.

t3chguy, 2024-01-17:
Will look into that tomorrow

t3chguy, 2024-01-18:
Have gone ahead with `assumed_state`