esphome/feature-requests #2549: Support for power monitor INA228/229

taHC81, 2024-01-06:
**Describe the problem you have/What new integration you would like**

Would it be possible to add INA228 (I2C) and INA229 (SPI) power monitoring chips? There're already some INAs supported, but these 2 are often used for battery management.

**Please describe your use case for this integration and alternatives you've tried:**

Battery voltage/current/power/SoC monitoring.

**Additional context**

taHC81, 2024-01-06:
https://www.ti.com/document-viewer/ina228/datasheet

latonita, 2024-01-14:
@taHC81 I can take a look into this - I have no device, so I will need your help to test/debug

taHC81, 2024-01-14:
> @taHC81 I can take a look into this - I have no device, so I will need your help to test/debug

With a pleasure, I have 3 devices ;)

latonita, 2024-01-14:
@taHC81  try this one https://github.com/latonita/esphome-ina228
if it works at all

taHC81, 2024-01-15:
Tried that, had to comment out TAG read part in a component setup (https://github.com/taHC81/esphome-ina228/blob/f05410bf26457f0036598b09aa1ad7a122a0b3a5/components/ina228/ina228.cpp#L59), then  it started working. Unfortunately, waiting for a bigger shunt (75mV / 400A), so can't provide some real numbers.

https://pastebin.com/B3pvi3eV

```
[08:48:22][D][sensor:094]: 'INA228 Shunt Voltage': Sending state 0.24000 V with 2 decimals of accuracy
[08:48:22][D][sensor:094]: 'INA228 Bus Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[08:48:22][D][sensor:094]: 'INA228 Current': Sending state 0.00000 A with 3 decimals of accuracy
[08:48:22][D][sensor:094]: 'INA228 Power': Sending state 0.00000 W with 2 decimals of accuracy
[08:48:24][D][sensor:094]: 'INA228 Shunt Voltage': Sending state 0.24000 V with 2 decimals of accuracy
[08:48:24][D][sensor:094]: 'INA228 Bus Voltage': Sending state 0.05000 V with 2 decimals of accuracy
[08:48:24][D][sensor:094]: 'INA228 Current': Sending state 0.00078 A with 3 decimals of accuracy
[08:48:24][D][sensor:094]: 'INA228 Power': Sending state 0.00000 W with 2 decimals of accuracy
[08:48:26][D][sensor:094]: 'INA228 Shunt Voltage': Sending state 0.24000 V with 2 decimals of accuracy
[08:48:26][D][sensor:094]: 'INA228 Bus Voltage': Sending state 0.05000 V with 2 decimals of accuracy
[08:48:26][D][sensor:094]: 'INA228 Current': Sending state 0.00078 A with 3 decimals of accuracy
[08:48:26][D][sensor:094]: 'INA228 Power': Sending state 0.00000 W with 2 decimals of accuracy
[08:48:28][D][sensor:094]: 'INA228 Shunt Voltage': Sending state 0.24000 V with 2 decimals of accuracy
```

latonita, 2024-01-15:
good, i actually supposed this might fail with some of the clones and forgot to make it just an informational/warning message

please check whether data is good - i hope byte order is okay

btw, do you own SPI versions of this device? then I can split this driver to base/i2c/spi

taHC81, 2024-01-15:
No, just I2C INA228, not INA229. But they should be completely same (registers) except the interface.

Die temp is not measured and energy/charge are "bit" off as well :)
[image]

latonita, 2024-01-15:
what are the 'expected' values, approximately?

latonita, 2024-01-15:
@taHC81 can you please send me output of these lines

  ESP_LOGI(TAG, "New Rshunt=%f ohm, max current=%.3f", r_shunt, max_current);
  ESP_LOGI(TAG, "New CURRENT_LSB=%f, SHUNT_CAL=%u", this->current_lsb_, this->shunt_cal_);

taHC81, 2024-01-15:
Not reported at all, I set debug level VERBOSE. Maybe you can make updates on your repo...

taHC81, 2024-01-15:
> @taHC81 i updated the code, added debug output. please enable DEBUG level logging and put output to the pastebin. thanks make sure you compile new version, sometimes it caches old one.
> 
> ```
> logger:
>   level: DEBUG
> ```

Here it is - https://pastebin.com/E4rkNWKF

latonita, 2024-01-15:
try one more time please + share logs.
thanks

taHC81, 2024-01-15:
Charge and energy look better, voltage still zero when I connect 3.3V

```
[13:41:39][D][ina228:031]: read_s20_4_ 0x04, ret= 0, raw 0x00A00000
[13:41:39][D][ina228:116]: read_volt_shunt_ is TRUE, 0x00000000
[13:41:39][D][sensor:094]: 'INA228 Shunt Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[13:41:39][D][ina228:031]: read_s20_4_ 0x05, ret= 0, raw 0x00100000
[13:41:39][D][ina228:125]: read_voltage_ is TRUE, 0x00000000
[13:41:39][D][sensor:094]: 'INA228 Bus Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[13:41:39][D][ina228:031]: read_s20_4_ 0x05, ret= 0, raw 0x00100000
[13:41:39][D][ina228:143]: read_voltage_ is TRUE, 0x00000000
[13:41:39][D][sensor:094]: 'INA228 Current': Sending state 0.00000 A with 3 decimals of accuracy
[13:41:39][D][ina228:152]: read_power_1 ret= 0, 0x00000000
[13:41:39][D][ina228:155]: read_power_2 ret= 0, 0x00000000
[13:41:39][D][sensor:094]: 'INA228 Power': Sending state 0.00000 W with 2 decimals of accuracy
[13:41:39][D][ina228:164]: read_energy_1 ret= 0, 0xFFFFA00000
[13:41:39][D][ina228:167]: read_energy_2 ret= 0, 0xA0FFFF
[13:41:39][D][sensor:094]: 'INA228 Energy': Sending state 206079.98438 J with 2 decimals of accuracy
[13:41:39][D][ina228:176]: read_charge_1 is TRUE, 0xFFFFA00000
[13:41:39][D][ina228:181]: read_charge_2 is TRUE, 0xA0FFFF
[13:41:39][D][sensor:094]: 'INA228 Charge': Sending state 4024.99951 C with 2 decimals of accuracy
```

latonita, 2024-01-15:
please share full log when it works for few minutes. thanks.

latonita, 2024-01-15:
one more try plz :)

taHC81, 2024-01-15:
> one more try plz :)

Unfortunately, still the same. https://pastebin.com/ZNrRwfgi

latonita, 2024-01-15:
maybe initialization is not enough, i will check other libs how do they initialize the device

taHC81, 2024-01-15:
Looks better now! Shunt will arrive tomorrow, so I'll provide another data.
[image]

latonita, 2024-01-15:
@taHC81 another minor update 
- charge shall be reading properly
- number of digits after decimal point increased

taHC81, 2024-01-16:
Even better, Anton! Thanks a lot! After I get the shunt, will do some preliminary tests.
[image]

taHC81, 2024-01-16:
**1A current +-**
```
[ina228:029]: read_s20_4_ 0x07, ret= 0, raw 0x00D00CFD
[ina228:140]: read_voltage_ is TRUE, 0xFFEFD0CD
[sensor:094]: 'INA228 Current': Sending state -101.15233 A with 5 decimals of accuracy
```

**1A current -+**
```
[ina228:029]: read_s20_4_ 0x07, ret= 0, raw 0x0060EF02
[ina228:140]: read_voltage_ is TRUE, 0x00002EF6
[sensor:094]: 'INA228 Current': Sending state 1.14651 A with 5 decimals of accuracy
```

Going to tweak shunt value to calibrate it.

taHC81, 2024-01-16:
After calibration:

```
[ina228:029]: read_s20_4_ 0x07, ret= 0, raw 0x00308F02
[ina228:140]: read_voltage_ is TRUE, 0x000028F3
[sensor:094]: 'INA228 Current': Sending state 0.99974 A with 5 decimals of accuracy

[ina228:029]: read_s20_4_ 0x07, ret= 0, raw 0x00606FFD
[ina228:140]: read_voltage_ is TRUE, 0xFFEFD6F6
[sensor:094]: 'INA228 Current': Sending state -101.00193 A with 5 decimals of accuracy
```

latonita, 2024-01-17:
hi. some updates posted.
- shunt voltage is shown in mV
- adc_range can be selected. option 0 or 1.
- current readings - its important to add some load between ground and Rshunt, and know whats approximate current shall be there.
just as it is here: 
[image]

I would be happy to see full logs +if possible, your manual multimeter readings of Vbus, Vshunt, Rshunt and current 

thanks!

taHC81, 2024-01-17:
Hello, here's the log - https://pastebin.com/ydERZ675

Please note the adc_range: 0 returns correct data, while I'd expect it from "1". I have 2x 10R filter resistors and 100nF cap.

10V, 99mA, 0.99W reported by a bench power supply. Charge and Energy are cleared only on a power cycle.

latonita, 2024-01-17:
1) so you say in adc_range 0 data is good, but in adc_range 1 it is wrong. maybe 4 times smaller/bigger?
2) I added some code with debug output to check whether wring to registers works okay
3) looks like some defaults values contradict datasheets... need to double check new logs
with your electrical values i expect v shunt to be 0.00113 ohm* 99ma = 0.11187 mV
which is exactly 4 times bigger than the reading

try and i need full log again plz.  and i will need to recheck all the nanos and millis :)

taHC81, 2024-01-18:
Thanks, now with adc_range = 1 - https://pastebin.com/L63Y12N3

Maybe with adc_range =1 we just get higher precision...

taHC81, 2024-01-18:
Strange behavior - on a first power on it measure half current and power, after restart (re-upload same code) starts measuring correctly.

https://pastebin.com/1nmGVV5i

latonita, 2024-01-18:
@taHC81 thanks for your tests 

not clear about todays results - as for yesterdays - unfortunately old software got cached.

would you please somehow reset caches and build from scratch.
i would be happy to see logs for adc_range = 0 and adc_range = 1.
i hope this time its getting closer and closer to what we all need )))

latonita, 2024-01-18:
one more thing. i see you are connecting to devices via network
is it possible to get connected via serial?  its missing setup() part of logs which are written before network is initialized
for adc 0 and adc 1
thanks

taHC81, 2024-01-19:
I'm playing also with some other ESP32's, and C3 has integrated USB interface, it drops connection during restart... Here it is, logger via serial port.

adc: 0 - https://pastebin.com/e3U7LB2B
adc: 1 - https://pastebin.com/MKm264kE

latonita, 2024-01-21:
thanks! i found an issue with device reset. you may try new one.
meanwhile i will start reworking this component to split into i2c and spi + i think its easy to add support for ina238/239 and ina237

latonita, 2024-01-24:
please check updated version. supporting ina228, ina238, ina237
note, that it has a new name

```
external_components:
  source: github://latonita/esphome-ina228
  components: [ina2xx_base, ina2xx_i2c]
```

I would appreciate full log from serial.
as well as your real electrical params and readings, thanks

latonita, 2024-01-24:
> Hello, some strange numbers this time - https://pastebin.com/nLj6vwgt , still same 10V/0.1A/1W.

Ouch, stupid copy-paste-typo. fixed, please check and send logs
thanks

latonita, 2024-01-24:
also here is documentation i prepared. I would be happy if you can take a look. maybe some mistakes, unclear things, etc.
https://deploy-preview-3571--esphome.netlify.app/components/sensor/ina2xx

taHC81, 2024-01-25:
> > Hello, some strange numbers this time - https://pastebin.com/nLj6vwgt , still same 10V/0.1A/1W.
> 
> Ouch, stupid copy-paste-typo. fixed, please check and send logs thanks

Yep, now it's ok. https://pastebin.com/HxM5ijH9

taHC81, 2024-01-25:
> also here is documentation i prepared. I would be happy if you can take a look. maybe some mistakes, unclear things, etc. https://deploy-preview-3571--esphome.netlify.app/components/sensor/ina2xx

Looks good, no obvious errors. Would you consider to add more human-friendly sensors like power/energy in mAh/Ah/Wh/kWh?

latonita, 2024-01-25:
> Looks good, no obvious errors. Would you consider to add more human-friendly sensors like power/energy in mAh/Ah/Wh/kWh?

Good idea. I have added Ah and Wh. 
energy and charge in Ah and Wh now, as its closer to real life humans :)
two new sensors are energy_joules and charge_coulombs.
docs updates as well

taHC81, 2024-01-25:
Probably you have to forgot to push the changes, still see the Joules and Couloumbs. I did the esphome cleanup...

```
[13:49:32][D][ina2xx:052]: Initiating new data collection
[13:49:32][D][ina2xx:501]: read_unsigned_ reg=0x04, ret=OK, len=3, val=0x5900
[13:49:32][D][ina2xx:341]: read_shunt_voltage_mv_ ret=OK, shunt_cal=4440, reading_lsb=1424.000000
[13:49:32][D][sensor:094]: 'INA228 Shunt Voltage': Sending state 0.11125 mV with 5 decimals of accuracy
[13:49:32][D][ina2xx:501]: read_unsigned_ reg=0x05, ret=OK, len=3, val=0xC6A30
[13:49:32][D][ina2xx:362]: read_bus_voltage_ ret=OK, reading_lsb=50851.000000
[13:49:32][D][sensor:094]: 'INA228 Bus Voltage': Sending state 9.93184 V with 5 decimals of accuracy
[13:49:32][D][ina2xx:510]: read_unsigned_16_ 0x06, ret= OK, val=0x0BB5
[13:49:32][D][ina2xx:382]: read_die_temp_c_ ret=OK, reading_lsb=2997.000000
[13:49:32][D][sensor:094]: 'INA228 DIE Temperature': Sending state 23.41406 °C with 5 decimals of accuracy
[13:49:32][D][ina2xx:501]: read_unsigned_ reg=0x07, ret=OK, len=3, val=0x5340
[13:49:32][D][ina2xx:398]: read_current_a_ ret=OK. current_lsb=0.000076. reading_lsb=1332.000000
[13:49:32][D][sensor:094]: 'INA228 Current': Sending state 0.10162 A with 8 decimals of accuracy
[13:49:32][D][ina2xx:501]: read_unsigned_ reg=0x08, ret=OK, len=3, val=0x1004
[13:49:32][D][ina2xx:412]: read_power_w_ ret=OK, reading_lsb=4100
[13:49:32][D][sensor:094]: 'INA228 Power': Sending state 1.00098 W with 6 decimals of accuracy
[13:49:32][D][ina2xx:510]: read_unsigned_16_ 0x0B, ret= OK, val=0x0003
[13:49:32][D][ina2xx:468]: read_diagnostics_and_act_ ret=OK, 0x0003
[13:49:32][D][ina2xx:501]: read_unsigned_ reg=0x09, ret=OK, len=5, val=0x4B
[13:49:32][D][ina2xx:432]: read_energy_j_ ret=OK, reading_lsb=0x4B, current_lsb=0.000076, overflow_cnt=0
[13:49:32][D][sensor:094]: 'INA228 Energy': Sending state 0.29297 J with 8 decimals of accuracy
[13:49:32][D][ina2xx:501]: read_unsigned_ reg=0x0A, ret=OK, len=5, val=0x1AA
[13:49:32][D][ina2xx:454]: read_charge_c_ ret=1, curr_charge=426.000000 + 39-bit overflow_cnt=0
[13:49:32][D][sensor:094]: 'INA228 Charge': Sending state 0.03250 C with 8 decimals of accuracy
```

latonita, 2024-01-25:
@taHC81 my bad, i pushed it to main PR only, but not into latonita ina228 repository

latonita, 2024-01-27:
@taHC81 can you please test negative current.

taHC81, 2024-01-27:
> @taHC81 can you please test negative current.
> 
> Basically Charge/discharge cycle
> 
> P. S. Vbus shall stay almost positive as per datasheet

Yeah, Vbus is the voltage across the battery terminals. I need to manage some test load. Will run test most probably in upcoming days.

taHC81, 2024-01-27:
https://pastebin.com/ujj7pnXz

Here's how it looks like when I've connected charger and current flowed in an opposite direction. Crazy numbers :)

latonita, 2024-01-27:
@taHC81 what are expected negative values?

taHC81, 2024-01-27:
Load (from a battery) is 0.286A, when I turn on power supply, it draws 0.317A, it can be translated to a charging with 0.031A. Maybe a bit more, as the Vbus is a bit higher (15.58V vs. 15.85V).

Let me discharge that battery a bit, to have a bigger difference.

latonita, 2024-01-27:
@taHC81  i pushed the fix. please try

taHC81, 2024-01-27:
No funny numbers anymore, but I just realized it shows positive power & current when I connect the load, and it should be opposite - load = minus power, charging = plus power. My smart battery is acting strangely, it doesn't accept any charge at the moment, will try with non-smart battpack on Monday. Sunday = family.

https://pastebin.com/Mp8WA0yR

latonita, 2024-01-27:
@taHC81 good to hear.
as for +- .... don't shoot the messenger. i'm just getting data from the device :) 
you might have `on_value` triggers in particular sensor to switch polarity if you need 
have a nice sunday. meanwhile i'll remove debug logs and make PR ready for review

taHC81, 2024-01-28:
Had some spare time, so did a try. Counters reset works fine, energy/charge working fine, but charge overflows, see the video and logs - https://pastebin.com/sW96Ww8s

https://www.youtube.com/watch?v=W_YwNdLt76I

latonita, 2024-01-28:
@taHC81 try get new ver, shall be proper negative charge

taHC81, 2024-01-28:
Fixed, good! ;) But I'd recommend to swap the polarity of the current as a default.

latonita, 2024-01-28:
@taHC81 this is just what i get from the device. i think it might be changed in particular sensor adding
```
  charge:
     name: xx
     filter:
        multiply : -1.0
```

latonita, 2024-01-28:
and one ugly thing which is coming - is overflowing charge register. 
i know its a big number (for your settings it is 11.5 kAh) but...
funny is that device counts charge as signed value, but at the same time overflow bit doesnt say is it + overflow or -overflow. :)))
I will ask TI forum

audito67, 2024-01-28:
Greetings.
I also have ina228 for monitoring the battery of a solar station.
At the moment I do not use any processing using HA tools; all data is displayed in its original form. There are fluctuations in the shunt voltage and shunt current in the near-zero zone, but the Ah meter increases the readings. After about an hour of operation of the device, the Ah reading is 0.00003181 Ah, while no real current flows through the shunt resistor.
 charge:
     name: xx
     filter:
        multiply : -1.0
If you use the solution you suggested, the register values will continue to increase even when the battery is low.

taHC81, 2024-01-28:
> Greetings. I also have ina228 for monitoring the battery of a solar station. At the moment I do not use any processing using HA tools; all data is displayed in its original form. There are [...]

It's a common noise, you have either live with that or filter out values under f.i. 0.005A and integrate energy by yourself. Even the sub-sized PV plant uses 100+ Ah batteries, so considering 1e-5 Ah values is a bit odd to me.

audito67, 2024-01-28:
I understand perfectly well that this is an insignificant value and can be neglected, the point is that this noise accumulates in the register, which logically should not happen.
And filtering can be done using the chip, as mentioned in the datasheet.

latonita, 2024-01-28:
I may change integration time and turn on averaging
For now it's 50us, changing to even max 4000us should not harm anyone (4 ms)
I will take a look later today

audito67, 2024-01-28:
> Было немного свободного времени, поэтому попробовал. Сброс счетчиков работает нормально, энергия / заряд работает нормально, но переполнение заряда, см. Видео и журналы - https://pastebin.com/sW96Ww8s
> 
> https://www.youtube.com/watch?v=W_YwNdLt76I

The question is a little off topic, sorry.
How did you implement resetting data in registers on your HA?
If possible, a piece of Yaml code on a third-party resource.

latonita, 2024-01-28:
BTW, as per datasheet energy/charge error is 1%

taHC81, 2024-01-28:
> The question is a little off topic, sorry. How did you implement resetting data in registers on your HA? If possible, a piece of Yaml code on a third-party resource.

```
button:
  - platform: template
    name: "Reset counters"
    on_press:
      - lambda: "id(ina228_sensor).reset_energy_counters();"
```

audito67, 2024-01-28:
> I may change integration time and turn on averaging For now it's 50us, changing to even max 4000us should not harm anyone (4 ms) I will take a look later today

The integration time seemed to be 1052 µs, and the averaging was “1”, or am I mistaken?
By and large, whether 1ms or 4ms is not a big difference, greater accuracy is achieved (according to the datasheet) by averaging the measurements.

[31 more replies]

closed 2024-05-16 by jesserockz