esphome/feature-requests #2550: Add support for the HC8 CO2 sensor

ClarkAlmazan, 2024-01-06:
**Describe the problem you have/What new integration you would like**

I would like to see the HC8 CO2 sensor integrated. This sensor is almost the same as the MHZ19, but uses a different protocol. It's been reported previously on https://github.com/esphome/issues/issues/4534.

**Please describe your use case for this integration and alternatives you've tried:**

This is one of the NDIR-based CO2 sensors that are available for a very cheap price. Currently, it's already supported in [Tasmota](https://github.com/arendst/Tasmota/pull/19714/files). I've tested their code for reading the CO2 values on an Arduino UNO R3 and I can confirm that it works.

I'm unaware of other functions besides reading the values.

**Additional context**

[Blogpost](https://spezifisch.codeberg.page/posts/2022-08-23/co2-sensor-reverse-engineering/) from someone who found this sensor.
[Cached datasheet](https://webcache.googleusercontent.com/search?q=cache%3Ahttp%3A%2F%2Fwww.icmkw.com%2Ffile-download-158219-left.html)

fanfaron2004, 2024-02-07:
hello ...  i use the PWM output ...

I measure the pulsation time (x), and we can calculate the CO2 value: ppm = (1000x-2)*5

here is an ESPhome example

sensor:
  - platform: pulse_width
    pin: GPIO19
    name: "Co2"
    device_class: carbon_dioxide
    state_class: measurement
    filters:
      - lambda: !lambda |-
          if (x < 0.02) return {};
          return round(((1000*x)-2)*5);
    unit_of_measurement: "ppm"
    update_interval: 60s

fanfaron2004, 2024-02-07:
[image]
[image]

ClarkAlmazan, 2024-02-08:
Hi @fanfaron2004, this looks kinda neat, definitely much simpler than what I had in mind. Never thought about using PWM, as I've never used it before. For this method, I'm assuming you only need 5V, GND, and the PWM pins connected, right?

For my own use, I made an esphome component, but honestly it's not very good so I don't think I want to make a pull request. I started off copying the mhz19 code, then just changed it to run like how the tasmota implementation did. It works but not without issues... You can see it here: https://github.com/ClarkAlmazan/esphome/commit/f751d198f07954478eb8664255f5dd9d7d0158a0

Racer0412, 2024-02-16:
Hey @fanfaron2004.

Could you please share the code in a different format so it's easier to copy?
That'd be nice. My  ESPHome is throwing errors and due to my limited experience I'm not able to fix it on my own :/

Thanks a lot!

IZZE2000, 2024-03-23:
FYI i asked Bing (GPT4) to fix the indentation and it worked out nicely!
sensor:
  platform: pulse_width
  pin: GPIO19
  name: "Co2"
  device_class: carbon_dioxide
  state_class: measurement
  filters:
    lambda: !lambda |-
      if (x < 0.02) return {};
      return round(((1000*x)-2)*5);
  unit_of_measurement: "ppm"
  update_interval: 60s

alexschultze, 2024-05-22:
+1 For implementing this "C08" type of sensor protocol.

vladjucov, 2025-04-14:
@spezifisch Thank you for your code! It's super what you did. 

But I have a question: why can I have almost 24 hours the same response from the sensor? 
```
[16:02:27][D][uart_debug:114]: <<< 42 4D 08 6A 08 75 17 70 01 01 28 00 00 00 00 E2 00 06 16
[16:02:27][D][co2hc8:088]: HC8 Received CO₂=6000ppm Mode=Cont.
[16:02:27][D][sensor:094]: 'CO2 Value': Sending state 6000.00000 ppm with 0 decimals of accuracy
[16:02:37][D][uart_debug:114]: <<< 42 4D 08 6A 08 75 17 70 01 01 28 00 00 00 00 E2 00 06 16
[16:02:37][D][co2hc8:088]: HC8 Received CO₂=6000ppm Mode=Cont.
[16:02:37][D][sensor:094]: 'CO2 Value': Sending state 6000.00000 ppm with 0 decimals of accuracy
[16:02:47][D][uart_debug:114]: <<< 42 4D 08 68 08 74 17 70 01 70 01 28 00 00 00 00 E2 00 06 13
[16:02:47][D][co2hc8:088]: HC8 Received CO₂=6000ppm Mode=Cont.
[16:02:47][D][sensor:094]: 'CO2 Value': Sending state 6000.00000 ppm with 0 decimals of accuracy
```
maybe I need to do some kind of calibration?