esphome/feature-requests #2551: Add support for Lite-On LTR-553 proximity sensor for CoreS3

lboue, 2024-01-06:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

[CoreS3](https://docs.m5stack.com/en/core/CoreS3 )is the third generation of the M5Stack Core series. Camera GC0308 is integrated at the bottom of the screen, accompanied by a proximity sensor [LTR-553ALS-WA ](https://optoelectronics.liteon.com/upload/download/DS86-2014-0007/LTR-553ALS-01_DS_V1.pdf)for enhanced functionality.

**Additional context**

Datasheet: [LTR-553ALS-WA](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/K128%20CoreS3/LTR-553ALS-WA.PDF)

latonita, 2024-01-25:
@lboue what functionality required? Will you be able to test regularly?

lboue, 2024-01-25:
The primary function would be to detect presence.
If a person is present, the screen lights up
If no presence, then screen backlight off

This saves battery power.

latonita, 2024-01-25:
@lboue I will take a look. I have just made LTR-303, looks similar from ALS perspective,need to add Proximity functions and interrupts.

lboue, 2024-01-25:
Thanks.
I did a test with the Arduino library and it worked fine.
Effectively it
LTR-553

https://github.com/chenhonglin/LTR507-LTR553-LTR556_Arduino/blob/master/ltr_5xx.h#L64
```
  #define LTR5XX_PS_MEASUREMENT_RATE_10MS  0x08
  #define LTR5XX_PS_MEASUREMENT_RATE_50MS  0x00
  #define LTR5XX_PS_MEASUREMENT_RATE_70MS  0x01
  #define LTR5XX_PS_MEASUREMENT_RATE_100MS 0x02 //default
  #define LTR5XX_PS_MEASUREMENT_RATE_200MS 0x03
  #define LTR5XX_PS_MEASUREMENT_RATE_500MS 0x04
  #define LTR5XX_PS_MEASUREMENT_RATE_1000MS 0x05
  #define LTR5XX_PS_MEASUREMENT_RATE_2000MS 0x06
```

I2C address is 0x23.

https://github.com/chenhonglin/LTR507-LTR553-LTR556_Arduino/blob/master/ltr_5xx.cpp#L48C1-L55C43

```
// LTR5XX Registers address
  #define LTR5XX_PS_THRES_UP_0_REG    0x90
  #define LTR5XX_PS_THRES_UP_1_REG    0x91
  #define LTR5XX_PS_THRES_LOW_0_REG   0x92
  #define LTR5XX_PS_THRES_LOW_1_REG   0x93
  #define LTR5XX_ALS_THRES_UP_0_REG   0x97
  #define LTR5XX_ALS_THRES_UP_1_REG   0x98
  #define LTR5XX_ALS_THRES_LOW_0_REG  0x99
  #define LTR5XX_ALS_THRES_LOW_1_REG  0x9A
```

latonita, 2024-01-25:
hm. checking https://docs.m5stack.com/en/core/CoreS3  - looks like it is connected to i2c bus only, interrupt pin is not connected to anything. is it?
if not - then hardware interrupts are not an option, we can just measure distance in continuous mode, check in the loop() and trigger some event.

lboue, 2024-01-25:
> hm. checking https://docs.m5stack.com/en/core/CoreS3  - looks like it is connected to i2c bus only, interrupt pin is not connected to anything. is it?
> 

- Internal SCL: GPIO11
- Internal SDA: GPIO12

https://github.com/m5stack/M5CoreS3/blob/63249119cb22bbd7a936de72b970053b0739b24b/examples/Basic/ltr553/ltr553.ino#L31C1-L33C69
```
    device_init_base_para.ps_led_pulse_freq   = LTR5XX_LED_PULSE_FREQ_40KHZ;
    device_init_base_para.ps_measurement_rate = LTR5XX_PS_MEASUREMENT_RATE_50MS;
    device_init_base_para.als_gain            = LTR5XX_ALS_GAIN_48X;
```

> if not - okay too, just means hardware interrupts can't be used, so we can just measure distance in continuous mode in loop(), and trigger some event if its closer than some particular distance
> 
> 
> 
>

lboue, 2024-01-25:
> @lboue would you please check if https://github.com/esphome/esphome/pull/6076 is working on your setup.
> 
> it only doing ALS part, not proximity. For me it looks like 553 is similar to 303/329 + proximity sensor added.
> 
> You will find there example config and link to documentation (through deep linking to esphome-docs PR :) )
> 
> to load component you will need to use
> 
> ```
> 
> external_components:
> 
>    - source: github://pr#6076
> 
>      components: [ltr303]
> 
> ```
> 
> 
> 
> please include log (through [pastebin](https://pastebin.com/))

I this that test last week and the devices goes into boot loop caused by the i2c component. I will post the debug log here.

lboue, 2024-01-25:
This time it worked with github://pr#6076 as a source for external_components:

```yaml
esphome:
  name: cores3-ltr
  friendly_name: CoreS3 LTR

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: arduino

psram:
  mode: octal
  speed: 80MHz

# Enable logging
logger:
  level: DEBUG

external_components:
  #- source: github://latonita/esphome@ltr303
  - source: github://pr#6076
    components: [ ltr303 ]

i2c:
  scl: 11
  sda: 12
  scan: true

sensor:
  - platform: ltr303
    address: 0x23 #0x29
    # auto mode - if true, then gain and integration time are not used
    auto_mode: true
    gain: 1x
    integration_time: 100ms
    ambient_light:
      name: "Ambient light"      
```

**Log file**

[ltr553.log](https://github.com/esphome/feature-requests/files/14055897/ltr553.log)

lboue, 2024-01-26:
Here are the values I get in HA:

[image]

latonita, 2024-01-26:
@lboue i pushed test version to my personal github repo
check it, its separate branch called `alsps`
https://github.com/latonita/esphome-ltr303/tree/alsps

esphome yaml shall have, other config values - check in readme
```
  - source: github://latonita/esphome-ltr303@alsps
```
also turn auto mode to off, to have less trashy log

1 check if it works at all
2 report readings for fully opened and fully closed sensor
3 if you can measure several distances and report counts - would be cool. like ten different values when hand is from 1cm to 20cm far away

also please check logs, you might disable other sensors to have less trash in log

latonita, 2024-01-26:
good. I see its looking not too far.
default integration time is 50ms, i can add ability to change it to get more saturation for the data, but not sure its needed.

What I may add to help you working with the device:
1) add parameter for threshold (counts)
2) if its higher than that - trigger event. you can write any lambda or call any action
3) also you may not need trigger, you can use trigger from 'sensor' called `on_value`, but this is tied to `update_interval`, and may not be very useful for *waking up*.
4) for the trigger i would add `cooldown` time to not call the trigger too often.

I'll make draft changes and let you know

ideas? comments?

lboue, 2024-01-26:
> I'll make draft changes and let you know
> 
> ideas? comments?

Sounds like a great start. For the moment, no further proposals. 
Always available for a new test.

latonita, 2024-01-26:
@lboue new version is there.

new options
```
#for getting triggered when close to the sensor. default is 65535, so it will never trigger
proximity_high_threshold: 100
#for getting triggered when far from the sensor. default is 0, so it will never trigger
proximity_low_threshold: 0
#no triggering happens for some after one of triggers fired
proximity_cooldown: 5 s
on_high_threshold:
  then:
    lambda: .... or some action instead of lambda to wake up your screen

on_low_threshold:
  then:
     .....

```

latonita, 2024-01-26:
> I will compare results with the arduino lib demo.

great. i only set PS_MEAS_RATE_50MS (instead of defaul 100ms) , other parameters default after device reset

lboue, 2024-01-27:
I'm going to do a test today. I had a bit of trouble yesterday.

lboue, 2024-01-28:
> > I will compare results with the arduino lib demo.
> 
> great. i only set PS_MEAS_RATE_50MS (instead of defaul 100ms) , other parameters default after device reset

Here are the result with the M5Stack arduino lib to compare

### PS sensor = clear

```
ps value = 0
als value = 1239
ps value = 0
als value = 1446
ps value = 0
als value = 1446
ps value = 0
als value = 1446
ps value = 0
als value = 1446
ps value = 0
als value = 1482
ps value = 0
als value = 1482
ps value = 0
als value = 1482
ps value = 0
als value = 1464
```

### PS sensor distance = 5cm 
```
ps value = 16
als value = 152
ps value = 15
als value = 152
ps value = 16
als value = 152
ps value = 14
als value = 156
ps value = 16
als value = 156
ps value = 14
als value = 156
ps value = 12
als value = 156

### PS sensor distance = 10cm 
```
ps value = 2
als value = 264
ps value = 0
als value = 264
ps value = 3
als value = 264
ps value = 2
als value = 274
ps value = 0
als value = 274
ps value = 5
als value = 274
ps value = 3
als value = 274
ps value = 2
als value = 279
```

lboue, 2024-01-28:
> new options
> 
> ```
> #for getting triggered when close to the sensor. default is 65535, so it will never trigger
> proximity_high_threshold: 100
> #for getting triggered when far from the sensor. default is 0, so it will never trigger
> proximity_low_threshold: 0
> #no triggering happens for some after one of triggers fired
> proximity_cooldown: 5 s
> on_high_threshold:
>   then:
>     lambda: .... or some action instead of lambda to wake up your screen
> 
> on_low_threshold:
>   then:
>      .....
> ```
> 
> please try, i cant test it on my side, need to update my ltr emulator , it takes time :)

@latonita
I have a display issue with the screen in my example file. For the test, I use a GPIO output to test the triggering.
I've just done the test and the trigger activates correctly at the threshold. That's great!

```
D][ltr303:093]: Initiating new data collection
[D][ltr303:157]: Reconfiguring sensitivity
[D][ltr303:338]: Data ready, reported gain is 1
[D][ltr303:139]: Reading sensor data having gain = 1x, time = 100 ms
[D][ltr303:350]: Got sensor data: CH1 = 65, CH0 = 106
[D][ltr303:371]: Low illuminance. Increasing gain.
[D][ltr303:157]: Reconfiguring sensitivity
[D][ltr303:338]: Data ready, reported gain is 2
[D][ltr303:139]: Reading sensor data having gain = 2x, time = 100 ms
[D][ltr303:350]: Got sensor data: CH1 = 134, CH0 = 218
[D][ltr303:371]: Low illuminance. Increasing gain.
[D][ltr303:157]: Reconfiguring sensitivity
[D][ltr303:338]: Data ready, reported gain is 4
[D][ltr303:139]: Reading sensor data having gain = 4x, time = 100 ms
[D][ltr303:350]: Got sensor data: CH1 = 251, CH0 = 396
[D][ltr303:371]: Low illuminance. Increasing gain.
[D][ltr303:157]: Reconfiguring sensitivity
[D][ltr303:338]: Data ready, reported gain is 8
[D][ltr303:139]: Reading sensor data having gain = 8x, time = 100 ms
[D][ltr303:350]: Got sensor data: CH1 = 514, CH0 = 816
[D][ltr303:371]: Low illuminance. Increasing gain.
[D][ltr303:157]: Reconfiguring sensitivity
[D][ltr303:338]: Data ready, reported gain is 48
[D][ltr303:139]: Reading sensor data having gain = 48x, time = 100 ms
[D][ltr303:350]: Got sensor data: CH1 = 3029, CH0 = 4886
[D][ltr303:394]: Illuminance is good enough.
[D][ltr303:423]: Lux calculation: ratio 0.382691, gain 48.000000, int time 1.000000, inv_pfactor 1.000000
[D][sensor:094]: 'Proximity counts': Sending state 0.00000 # with 0 decimals of accuracy
[D][sensor:094]: 'Ambient light': Sending state 250.39586 lx with 1 decimals of accuracy
[D][sensor:094]: 'Infrared counts': Sending state 3029.00000 # with 0 decimals of accuracy
[D][sensor:094]: 'Full spectrum counts': Sending state 4886.00000 # with 0 decimals of accuracy
[D][sensor:094]: 'Actual gain': Sending state 48.00000  with 0 decimals of accuracy
[D][sensor:094]: 'Actual integration time': Sending state 100.00000 ms with 0 decimals of accuracy
[D][ltr303:202]: Proximity high threshold triggered. Value = 102, Trigger level = 100
[D][switch:012]: 'CoreS3 PortB Out pin' Turning ON.
[D][switch:055]: 'CoreS3 PortB Out pin': Sending state ON
[D][ltr303:091]: Updating
[D][ltr303:093]: Initiating new data collection
[D][ltr303:157]: Reconfiguring sensitivity
[D][ltr303:338]: Data ready, reported gain is 1
[D][ltr303:139]: Reading sensor data having gain = 1x, time = 100 ms
```

On the other hand, I haven't yet managed to activate deactivation after 5 seconds as set.

latonita, 2024-01-28:
Okay, I'm not yet done with doxumentation
Cool down time is just period when nothing triggers.
Just to avoid thousands of triggers.

lboue, 2024-05-24:
Any news on this request?

latonita, 2024-05-24:
it is in the queue for approval

closed 2024-05-30 by kbx81