esphome/feature-requests #2554: Tuya climate more flexible

diogopms, 2024-01-10:
**Describe the problem you have/What new integration you would like**

Tuya thermostats with internal temperature sensors require calibration to show the right temperature. Currently, it's not possible to use `offset` any value because it's connected to `current_temperature_datapoint` and not to the temperature sensor itself.

**Please describe your use case for this integration and alternatives you've tried:**

```yaml
# Not OK - Temperature: 20ºC
climate:
  - platform: tuya
    name: "Thermostat"
    icon: mdi:heating-coil
    switch_datapoint: 1
    target_temperature_datapoint: 2
    current_temperature_datapoint: 3
    temperature_multiplier: 0.5
    eco_datapoint: 5
    eco_temperature: 16 °C
    visual:
      min_temperature: 10 °C
      max_temperature: 30 °C
      temperature_step: 0.5 °C

# OK - Temperature: 18ºC
sensor:
  - platform: tuya
    name: "Temperature"
    device_class: "temperature"
    sensor_datapoint: 3
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    filters:
      - multiply: 0.5
      - offset: -3 # Because of the offset
```

[image]

**Additional context**

Bl00d-B0b, 2024-02-13:
Hello,

make it simple:
    filters:
      - lambda: return x / 2 + 3;

Ed

ssieb, 2024-02-13:
If you can find a datapoint to set the offset, then sure.  But there's no point in adjusting it on the esphome side.  All the control is inside the Tuya MCU.

Bl00d-B0b, 2024-02-13:
If you want to calibrate internal sensor, then You need to use device functionality, like:
[image]

Ed

closed 2024-08-28 by diogopms